git-kick/ioBroker.e3dc-rscp #28: donation for your great job...

kkeintzel, 2021-12-03:
Where can i "donate" something for your great job?

Perhaps you can take a preferential look for the wallbox communication.

regards
Karl K.

git-kick, 2021-12-03:
Hi Karl, glad to read that you like the adapter.
Yes, I can have a look at the wallbox data - my issue is it will be hard to test for me because I have no data-connection between wallbox and E3/DC. So a welcome donation would be if you could test upcoming betas w.r.t. wallbox connectivity.
Otherwise, you can buy me a (good Bavarian) beer via [PayPal](https://www.paypal.me/ukick) :-) 
Cheers, Uli

git-kick, 2021-12-04:
Noch ein Hinweis, ich weiß nicht ob du es schon gesehen hast: es gibt ein eigenes [Forum](https://forum.iobroker.net/topic/49217/test-adapter-e3dc-rscp-v0-0-x-github) zum Adaptertest.
Dort stehen immer die aktuellsten Infos und wir können bei Bedarf auch one-to-one chatten.
Ich schließe deshalb dieses Ticket.

closed 2021-12-04 by git-kick

git-kick, 2021-12-05:
Ich habe das Lesen der Wallbox-Daten in den Adapter eingebaut, allerdings wie angekündigt ohne Testmöglichkeit.
https://forum.iobroker.net/topic/49217/test-adapter-e3dc-rscp-v0-0-x-github/176

Bitte melde dich via Forum, wenn es Fragen oder Probleme gibt!