git-kick/ioBroker.e3dc-rscp #58: Save HISTORY_DATA as JSON file

git-kick, 2022-01-29:
Source: https://forum.iobroker.net/topic/49217/test-adapter-e3dc-rscp-v0-0-x-github/253

Add a configuration setting to save HISTORY_DATA results not only in the object tree, but also in JSON files.

This was suggested by @arnod for using HISTORY_DATA in vis diagrams, but requirement was not confirmed so far.

> Bevor du dir die Arbeit machst, müsste man sich vorher Gedanken machen, ob es überhaupt Sinn ergibt, das anzuzeigen und wenn ja, was von diesen Werten.
> Das meiste kann man bereits jetzt mit dem Adapter History logen und über den Adapter Flot oder echarts in einem Diagramm anzeigen.
> Das ist auch einfacher als hier das Rad noch mal neu zu erfinden.
> Das einzige, was jetzt für mich Sinn ergeben hätte, wäre die Autarkie gewesen, aber da habe ich die Werte mit meinen Aufzeichnungen aus dem letzten Jahr verglichen und habe festgestellt, dass diese auch nicht genau sind. Liegt natürlich auch daran das bei einem TIME_INTERVAL= 2592000 s alle Monate mit 31 Tage das Ergebnis schon mal verfälschen. Die Summe übers Jahr würde in etwa schon stimmen.
> 
> Wenn jemand das in einem Diagramm haben will, kann ich gerne eine JSON Beispieldatei zusammenstellen, um das in Vis in einem Materialdesign JSON Chart anzeigen zu können. Es müssten wahrscheinlich zwei oder drei Diagramme werden, um das noch übersichtlich darstellen zu können.
> Anleitung dazu kann man hier finden:
> [line-history-chart](https://github.com/Scrounger/ioBroker.vis-materialdesign#line-history-chart)

ESDN83, 2022-10-20:
In addition just saying updating the last History records as number in the Object would be nice in order to use it with Influx and Grafana.

Workaround is possible running your own schedule right now already.. but unnecessary...

git-kick, 2023-02-18:
Requirement was not confirmed for 1 year.
Also, a direct processing of time series data (e.g. using TICK-stack with Influxdb) is far more efficient.
Therefore, I close this issue.

closed 2023-02-18 by git-kick