git-kick/ioBroker.e3dc-rscp #63: Use standard password encryption method

git-kick, 2022-01-30:
@Apollon77 wrote:
> If you like you can remove all the encrypt/decrypt stuff and use the automatic solution ... see best practices in Readme of this repo for infos. YOu just need to add an "encrptedNative" and "protectedNative" sections to io-package and add a propoer admin globalDependency.

see https://github.com/ioBroker/ioBroker.repositories/pull/1589

git-kick, 2022-02-01:
@Apollon77, I'm afraid I need your support again, because I cannot figure out the "automatic solution":  in `io-package.json`, I can add `encryptedNative` and `protectedNative` as described, but...
- what do you mean by "add a proper admin globalDependency"? Which file is affected here?
- are `myencrypt()` and `mydecrypt()` definitions in `index_m.html` and `decrypt()` in `main.js` still neccessary?

In [README](https://github.com/ioBroker/ioBroker.repositories/blob/master/README.md), I found this hint:
> if you need to store passwords please encrypt them in Admin! You can check e.g. Apollon77/iobroker.meross for example code in index_m.html and main.js

In `ioBroker.meross`, I found the xxxNative declarations in `io-package.json`, but `index_m.html` and `main.js` contain the well-known `myencrypt()`/`mydecrypt()` and `decrypt()` definitions, similar to what I'm doing in e3dc-rscp V1.0.0 
I browsed some of your adapters, but found none besides `meross` having the xxxNative declarations in `io-package.json`.

So I'm kind of lost, could you give me a hint, perhaps an adapter example where _only_ the automatic solution is implemented?

Apollon77, 2022-02-02:
Hi,

Dependency wise it is admin >=4.0.9

The myen/decrypt is not longer needed then., And yes merioss is not longer "tsta eof the art" - I will adjust taht soon

git-kick, 2022-02-27:
@Apollon77 
> Dependency wise it is admin >=4.0.9

I'm a bit lost... in `iobroker-config.json`, I can see `"adminVersion": "admin5"`. Is this good enough? Or do I have to add a new dependency `"admin": ">=4.0.9"`under `io-package.json/"common"/"dependencies"`?
> 
> The myen/decrypt is not longer needed then., And yes merioss is not longer "tsta eof the art" - I will adjust taht soon

How do I decrypt then when I access `this.config.myPassword`? Could you point me to a working sample, please?

git-kick, 2024-09-09:
With v1.3.0, we migrated to admin5 json-config and passwords are protectedNative and encryptedNative.

closed 2024-09-09 by git-kick