git-kick/ioBroker.e3dc-rscp #67: Device Discovery

git-kick, 2022-01-31:
Add a discovery interface for E3/DC devices, supposedly using uPnP.

> You do not "need to" have a discovery ... If your dapter integrates devices that can be discovered then it would be great to add to discovery adapter.
> 
> So IF it is possible then you need to do a PR for https://github.com/ioBroker/ioBroker.discovery/tree/master/lib/adapters and addyour file there to detect a device.
> In fact discovery will scan all IPs in the network but can also do mdns or udp

Source: https://github.com/ioBroker/ioBroker.repositories/pull/1589

git-kick, 2022-02-01:
e3dc-rscp discovery is up and running.
But the discovery `e3dc-rscp.js` file is large because it duplicates all the polling interval defaults.
Requested advice by @apollon77

Apollon77, 2022-02-01:
Do you have a code link?

git-kick, 2022-02-01:
Yes: https://github.com/git-kick/ioBroker.discovery/blob/master/lib/adapters/e3dc-rscp.js

git-kick, 2022-02-14:
[PR#213](https://github.com/ioBroker/ioBroker.discovery/pull/213) is waiting for approval

git-kick, 2022-02-21:
[PR#213](https://github.com/ioBroker/ioBroker.discovery/pull/213) is merged into master, but `discovery@2.7.5` still does not find E3/DC.
Wait for `discovery>2.7.5`

Apollon77, 2022-02-21:
No, wait for 2.8.0 please ...see Discovery changelog https://github.com/ioBroker/ioBroker.discovery#280-2022-02-20

git-kick, 2022-03-21:
Tested o.k. with `discovery@3.0.2`

closed 2022-03-21 by git-kick