git-kick/ioBroker.e3dc-rscp #89: Emergency Power Level or Value

mccrossen, 2022-03-10:
Hi,
i miss the configured ermegency power level or value.

git-kick, 2022-03-10:
Thank you for your meaningful input.
nischram lists 11 tags which are new to me (and not mentioned in E3/DC docs):
```
TAG_SE_PARAM_EP_RESERVE_W      0x1B040033
TAG_SE_REQ_SET_EP_RESERVE      0x1B000010
TAG_SE_PARAM_INDEX             0x1B040000
TAG_SE_PARAM_TIME_LAST_FULL    0x1B040024
TAG_SE_PARAM_TIME_LAST_EMPTY   0x1B040025
TAG_SE_REQ_EP_RESERVE          0x1B000009
TAG_SE_PARAM_LAST_SOC          0x1B040026
TAG_SE_PARAM_EP_RESERVE        0x1B040023
TAG_SE_PARAM_EP_RESERVE_MAX_W  0x1B040034
TAG_SE_EP_RESERVE              0x1B800009
TAG_SE_PARAM_INT               0x1B040011
```
I will try to add the new namespace "SE" to the adapter.
@mccrossen, do you know what "SE" stands for, is it "**S**et **E**mergency power"?

mccrossen, 2022-03-10:
> @mccrossen, do you know what "SE" stands for, is it "**S**et **E**mergency power"?

I think it is a naming convention by Nischram and not the name given by E3DC.

git-kick, 2022-03-10:
Ah ok, then I will think about renaming the above tags to `TAG_EP_...`

git-kick, 2022-03-19:
With v1.0.3, the values mentioned above will be displayed under `EP.PARAM_0`
(it is a numbered subtree because the S10 returns a container with an index.)

I tried setting `EP_RESERVE`, but the S10 returned a weird container with a random index and value -1.
So I removed the setter code.

If anybody can tell me what the S10 expects for setting the relevant values, I can give it another try (please open a new issue if so).

Your can preview the fix using
 https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%2389

git-kick, 2022-03-21:
(von @arnod)

„ I tried setting EP_RESERVE, but the S10 returned a weird container with a random index and value -1.
So I removed the setter code.“

Aus der Historie der Anlagen habe ich die Vermutung, dass EP_Reserve nicht eine Watthzahl ist sondern 0-50%.
Denn so war es am Anfang auch angezeigt und musste auch so eingegeben werden und wurde erst später als Wh angezeigt.
Ich könnte mir gut vorstellen, dass es im Hintergrund weiter die % sind.

Also 0-50 % von Gesamtbatteriekapa.
Was hattest Du denn als Eingabewerte ausprobiert?

git-kick, 2022-03-21:
Ich habe (in der Annahme, dass es um Wh geht), ca. 1000 eingegeben. Ich werde es nochmal testen.

Matis-11, 2022-03-21:
Schade, war aber ein Versuch wert, danke.
Es muß aber irgendsowas sein, denn die Werte die jetzt auf dem P10 zum anklicken angeboten werden, sind 0%, 10%, 25%, 50% vom Max-Bat-Kapa.
Wenn ich am P10 einen Wert einstelle, wird der in den Objekten aber nicht ausgegeben, RESERVE und RESERVE_W bleiben 0.
LAST_SOC ist auch 0 und die beiden TIME Werte zeigen 1970. Also außer dem RESERVE_MAX_W ist kein Wert korrekt gefüllt.
Die Inselbetrieb Flags sind korrekt.

closed 2022-03-21 by git-kick

Matis-11, 2022-03-21:
[image]

... das ist doch wirklich verflixt.