git-kick/ioBroker.e3dc-rscp #92: Additional PVI tags?

trapezoid677, 2022-03-15:
Thank you for this great project!

I have a S10 where the maximum string voltage for the 2 strings is limited in order to improve the compatibility with installed Solaredge optimizers. But I can't find corresponding tags in your PVI implementation. Is it possible that the following tags are missing?

PVI_DC_MAX_VOLTAGE  = 0x028DC005
PVI_DC_MIN_VOLTAGE  = 0x028DC006
PVI_DC_MAX_CURRENT  = 0x028DC007
PVI_DC_MIN_CURRENT  = 0x028DC008
PVI_REQ_DC_MAX_VOLTAGE   = 0x020DC005
PVI_REQ_DC_MIN_VOLTAGE   = 0x020DC006
PVI_REQ_DC_MAX_CURRENT  = 0x020DC007
PVI_REQ_DC_MIN_CURRENT  = 0x020DC008

I'm using v1.0.2

git-kick, 2022-03-16:
@trapezoid677, thank you for the clear ticket.
The tags you mentioned are simply missing in the adapter's request.

I will add the tags in the upcoming version (release v1.0.3, probably next weekend, if everything works smoothly...)

git-kick, 2022-03-17:
The four REQ_ tags are implemented in branch
https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%2392
My S10 always returns `RSCP_ERR_UNKNOWN_TAG` for both MIN_ tags, and no value at all for both MAX_ tags.

@trapezoid677, please test using the branch mentioned above.
I want to keep the logs as "clean" as possible, so I will not add the tags unless there is significant benefit.

git-kick, 2022-03-19:
OK, thank you for testing.
Unfortunately, it's often trial & error because there is no up-to-date and model-specific doc.
I will not include the patch in v1.0.3

closed 2022-03-19 by git-kick