git-kick/ioBroker.e3dc-rscp #93: Connection Error

bushmills1608, 2022-03-18:
The Connection is not stable.
[image]
V1.02

git-kick, 2022-03-18:
Looks like a site-specific issue.
If you think it is worth, then please find steps to reproduce the problem and fill the bug report form, including a complete description of your installation.

bushmills1608, 2022-03-19:
Sehr gerne, 
leider weiß ich nicht wo ich den bug report form ausfüllen kann.

git-kick, 2022-03-19:
Wenn man den Button "New Issue" wählt, gibt es bei "Bug Report" einen Button "Get Started". Damit öffnet sich ein neuer Issue mit den entsprechenden Abschnitten.

bushmills1608, 2022-03-20:
ich finde dummerweise keinen Button "New Issue"

git-kick, 2022-03-20:
Hmm, hab ichs nicht richtig beschrieben? Also bitte erstmal hierher:
https://github.com/git-kick/ioBroker.e3dc-rscp/issues
Das sollte dan oben so aussehen:
[image]
Da ist rechts in grün "New Issue", wenn du den drückst kommt:
[image]
... und mit "Get started" gehts los.

git-kick, 2022-03-21:
Ersetzt durch [Issue#97](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/97)

closed 2022-03-21 by git-kick