git-kick/ioBroker.e3dc-rscp #95: Nach Laden der Konfiguration in der Instanz ist das Feld Speichern ausgegraut

ArnoD15, 2022-03-20:
**Describe the bug**  
Wenn die Instanz e3dc-rscp gelöscht wird und neu hinzugefügt wird, kann eine zuvor gespeicherte Konfiguration zwar geladen werden, aber der Button "Speichern" und "Speichern und schließen" bleiben ausgegraut.
Die geladene Konfiguration kann nur gespeichert werden, wenn ein Wert geändert wird, dann gespeichert wird und im Anschluss dieser Wert noch mal in den ursprünglichen Wert zurückgeändert wird und dann gespeichert wird.

**To Reproduce**  
Steps to reproduce the behavior:
1. Speichere Konfiguration der Instanz
2. Lösche Instanz
3. Instanz hinzufügen
4. Konfiguration laden
5. Dann kann nicht gespeichert werden.

**Expected behavior**  
Beiden Button müssten Speichern und Speichern und schließen müssten aktiv sein.

**Versions:**  
 - Adapter version: 1.0.3
 - JS-Controller version: 4.0.21
 - Node version: 14.19.0 
 - Operating system: linux

git-kick, 2022-03-20:
@ArnoD15, danke für das gut beschriebene Ticket - in der Tat hab ich mich mit dem Speichern der Konfiguration noch nie beschäftigt, das muss ich mir erstmal ansehen.

git-kick, 2022-03-21:
Ja, da war einmal zu oft `onChange(false)` in meinem (abgekupferten) `index_m.html`.
Sollte jetzt gehen, hier zu testen:
https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue-95

Kommt mit Release v1.0.4

closed 2022-03-21 by git-kick