git-kick/ioBroker.e3dc-rscp #96: Nach Adapterupdate auf 1.0.3 wird der zweite Batteriekreis BAT_1 nicht mehr erkannt bzw. abgefragt.

ArnoD15, 2022-03-20:
**Describe the bug**  
Mit dem neuen Update 1.0.3 wird bei meinem E3DC der zweite Batteriekreis nicht mehr erkannt.
Unter den Objekten wird nur noch BAT_0 angezeigt und BAT_1 fehlt.

**To Reproduce**  
Steps to reproduce the behavior:
1. Adapter auf 1.0.3 updaten
2. Instanz starten
3. unter den Objekten fehlt BAT_1 was in der vorherigen Version noch funktionierte.

**Screenshots & Logfiles**  
`
2022-03-20 10:54:35.074 - info: host.iobroker "system.adapter.e3dc-rscp.0" enabled
2022-03-20 10:54:35.229 - info: host.iobroker instance system.adapter.e3dc-rscp.0 started with pid 393
2022-03-20 10:54:36.646 - debug: e3dc-rscp.0 (393) Redis Objects: Use Redis connection: 0.0.0.0:9001
2022-03-20 10:54:36.701 - debug: e3dc-rscp.0 (393) Objects client ready ... initialize now
2022-03-20 10:54:36.706 - debug: e3dc-rscp.0 (393) Objects create System PubSub Client
2022-03-20 10:54:36.707 - debug: e3dc-rscp.0 (393) Objects create User PubSub Client
2022-03-20 10:54:36.785 - debug: e3dc-rscp.0 (393) Objects client initialize lua scripts
2022-03-20 10:54:36.790 - debug: e3dc-rscp.0 (393) Objects connected to redis: 0.0.0.0:9001
2022-03-20 10:54:36.793 - silly: e3dc-rscp.0 (393) redis psubscribe cfg.o.system.user.*
2022-03-20 10:54:36.827 - silly: e3dc-rscp.0 (393) redis psubscribe cfg.o.enum.*
2022-03-20 10:54:36.832 - silly: e3dc-rscp.0 (393) objectDB connected
2022-03-20 10:54:36.834 - debug: e3dc-rscp.0 (393) Redis States: Use Redis connection: 0.0.0.0:9000
2022-03-20 10:54:36.956 - debug: e3dc-rscp.0 (393) States create System PubSub Client
2022-03-20 10:54:36.957 - debug: e3dc-rscp.0 (393) States create User PubSub Client
2022-03-20 10:54:37.282 - debug: e3dc-rscp.0 (393) States connected to redis: 0.0.0.0:9000
2022-03-20 10:54:37.283 - silly: e3dc-rscp.0 (393) statesDB connected
2022-03-20 10:54:40.866 - info: e3dc-rscp.0 (393) starting. Version 1.0.3 in /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v14.19.0, js-controller: 4.0.21
2022-03-20 10:54:40.925 - silly: e3dc-rscp.0 (393) States system redis pmessage system.adapter.e3dc-rscp.0.logLevel/system.adapter.e3dc-rscp.0.logLevel:{"val":"silly","ack":true,"ts":1647770080899,"q":0,"from":"system.adapter.e3dc-rscp.0","lc":1647770080899}
2022-03-20 10:54:40.966 - debug: e3dc-rscp.0 (393) config.*: (xxx.xxx.xxx.xxx, 5033, xxxxxxx.xxxxxx@xxxxxx.xx, 2, 1, 1, 9)
2022-03-20 10:54:40.997 - debug: e3dc-rscp.0 (393) Probing for BAT units - 0..0.
2022-03-20 10:54:41.002 - debug: e3dc-rscp.0 (393) Probing for PVI units - 0..2.
2022-03-20 10:54:41.009 - debug: e3dc-rscp.0 (393) Sending request TAG_RSCP_REQ_AUTHENTICATION
2022-03-20 10:54:41.010 - silly: e3dc-rscp.0 (393) OUT: magic: >E3DC< is OK - ctrl: >0011< is OK - Version 1, with CRC - seconds: 1647770080 - nseconds: 0 - length: 58
TAG_RSCP_REQ_AUTHENTICATION - type: 0x0E - Container - length: 51
TAG_RSCP_AUTHENTICATION_USER - type: 0x0D - CString - length: 26 value: xxxxxxx.xxxxxx@xxxxxx.xx
TAG_RSCP_AUTHENTICATION_PASSWORD - type: 0x0D - CString - length: 11 value: xxxxxxxxxxxxxxxxxxxxxxxx
CRC32
2022-03-20 10:54:41.028 - silly: e3dc-rscp.0 (393) Successfully written data to socket
2022-03-20 10:54:41.030 - info: e3dc-rscp.0 (393) Connection to E3/DC is established
2022-03-20 10:54:41.031 - debug: e3dc-rscp.0 (393) Sending request TAG_BAT_REQ_DATA
2022-03-20 10:54:41.031 - silly: e3dc-rscp.0 (393) OUT: magic: >E3DC< is OK - ctrl: >0011< is OK - Version 1, with CRC - seconds: 1647770081 - nseconds: 0 - length: 23
TAG_BAT_REQ_DATA - type: 0x0E - Container - length: 16
TAG_BAT_INDEX - type: 0x05 - UInt16 - length: 2 value: 0
TAG_BAT_REQ_INFO - type: 0x00 - None - length: 0
CRC32
2022-03-20 10:54:41.062 - silly: e3dc-rscp.0 (393) Successfully written data to socket
2022-03-20 10:54:41.088 - silly: e3dc-rscp.0 (393) Received response
2022-03-20 10:54:41.088 - silly: e3dc-rscp.0 (393) TAG_RSCP_AUTHENTICATION
2022-03-20 10:54:41.089 - silly: e3dc-rscp.0 (393) IN: magic: >E3DC< is OK - ctrl: >0011< is OK - Version 1, with CRC - seconds: 1647770081 - nseconds: 523159000 - length: 8
TAG_RSCP_AUTHENTICATION - type: 0x03 - UChar8 - length: 1 value: 0x0A
no CRC32
2022-03-20 10:54:41.091 - silly: e3dc-rscp.0 (393) processTree: path = , tree = [ {TAG_RSCP_AUTHENTICATION(UChar8) = 10}, ]
2022-03-20 10:54:41.094 - debug: e3dc-rscp.0 (393) Sending request TAG_PVI_REQ_DATA
2022-03-20 10:54:41.094 - silly: e3dc-rscp.0 (393) OUT: magic: >E3DC< is OK - ctrl: >0011< is OK - Version 1, with CRC - seconds: 1647770081 - nseconds: 0 - length: 38
TAG_PVI_REQ_DATA - type: 0x0E - Container - length: 31
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 0
TAG_PVI_REQ_AC_MAX_PHASE_COUNT - type: 0x00 - None - length: 0
TAG_PVI_REQ_TEMPERATURE_COUNT - type: 0x03 - UChar8 - length: 1 value: 0x00
TAG_PVI_REQ_DC_MAX_STRING_COUNT - type: 0x00 - None - length: 0
CRC32
2022-03-20 10:54:41.096 - silly: e3dc-rscp.0 (393) Successfully written data to socket
2022-03-20 10:54:41.114 - silly: e3dc-rscp.0 (393) Received response
2022-03-20 10:54:41.119 - silly: e3dc-rscp.0 (393) TAG_BAT_DATA
2022-03-20 10:54:41.119 - silly: e3dc-rscp.0 (393) IN: magic: >E3DC< is OK - ctrl: >0011< is OK - Version 1, with CRC - seconds: 1647770081 - nseconds: 524792000 - length: 89
TAG_BAT_DATA - type: 0x0E - Container - length: 82
TAG_BAT_INDEX - type: 0x05 - UInt16 - length: 2 value: 0
TAG_BAT_INFO - type: 0x0E - Container - length: 66
TAG_BAT_RSOC - type: 0x0A - Float32 - length: 4 value: 75.70814514160156
[... 23 lines cut ...]
no CRC32
2022-03-20 10:54:43.031 - silly: e3dc-rscp.0 (393) processTree: path = , tree = [ {TAG_BAT_DATA(Container) = [ {TAG_BAT_INDEX(UInt16) = 0}, ]}, ]
2022-03-20 10:54:43.031 - silly: e3dc-rscp.0 (393) processTree: path = , tree = [ {TAG_BAT_INDEX(UInt16) = 0}, ]
2022-03-20 10:54:43.032 - debug: e3dc-rscp.0 (393) Sending request TAG_PVI_REQ_DATA
2022-03-20 10:54:43.032 - silly: e3dc-rscp.0 (393) OUT: magic: >E3DC< is OK - ctrl: >0011< is OK - Version 1, with CRC - seconds: 1647770083 - nseconds: 0 - length: 191

TAG_BAT_REQ_DATA - type: 0x0E - Container - length: 9
TAG_BAT_INDEX - type: 0x05 - UInt16 - length: 2 value: 0
CRC32
2022-03-20 10:54:47.025 - silly: e3dc-rscp.0 (393) Successfully written data to socket
2022-03-20 10:54:47.039 - silly: e3dc-rscp.0 (393) Received response
2022-03-20 10:54:47.039 - silly: e3dc-rscp.0 (393) TAG_BAT_DATA
2022-03-20 10:54:47.040 - silly: e3dc-rscp.0 (393) IN: magic: >E3DC< is OK - ctrl: >0011< is OK - Version 1, with CRC - seconds: 1647770087 - nseconds: 466334000 - length: 16
TAG_BAT_DATA - type: 0x0E - Container - length: 9
TAG_BAT_INDEX - type: 0x05 - UInt16 - length: 2 value: 0
no CRC32
2022-03-20 10:54:47.040 - silly: e3dc-rscp.0 (393) processTree: path = , tree = [ {TAG_BAT_DATA(Container) = [ {TAG_BAT_INDEX(UInt16) = 0}, ]}, ]
2022-03-20 10:54:47.040 - silly: e3dc-rscp.0 (393) processTree: path = , tree = [ {TAG_BAT_INDEX(UInt16) = 0}, ]
'

**Versions:**  
 - Adapter version: 1.0.3
 - JS-Controller version: 4.0.21
 - Node version: 14.19.0
 - Operating system: linux

Matis-11, 2022-03-20:
Bei mir Bat_1 noch einwandfrei angezeigt.
Hast du den Baum mal gelöscht und den Adapter neu gestartet?

ArnoD15, 2022-03-20:
Ich hatte vor dem Update die Instanz gelöscht und somit eigentlich alle Objekte.
Habe jetzt noch mal die Instanz gestoppt und den Objektbaum komplett gelöscht, jedoch immer noch das gleiche Problem, dass BAT_1 nicht mehr angelegt wird. Kann jetzt nur noch mal den Adapter löschen und neu installieren.

git-kick, 2022-03-20:
Hui, das ist schräg, da werde ich mir mal das Log von @ArnoD15 ganze genau aunsehen müssen.
Ich habe eigentlich bei der Verarbeitung der Setter-Tags (also Werte ändern) einiges geändert - woher diese Regression kommt, ist mir auf den ersten Blick gar nicht klar.
Als (hoffentlich temporärer) Work-around bleibt wohl nur downgrade auf v1.0.2... sorry.

git-kick, 2022-03-21:
Ich denke der Grund ist schon gefunden: im Zuge von [Issue#85](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/85#issuecomment-1054268729) habe ich `batProbes` auf 1 verringert, auf Basis eines Kommentars in der E3/DC Tagliste. Anscheinend ist der aber falsch oder veraltet. Also setze ich `batProbes` wieder auf 2:
https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%2396
@ArnoD15, bitte um kurzen Check.

Ein Release v1.0.4 mit der Korrektur folgt.

git-kick, 2022-03-21:
Was ich aber weiter nicht verstehe: warum sieht @Matis-11 auch mit v1.0.3 den BAT_1 Ast?
Sieht so aus, als gäbe es bei E3/DC zwei verschiedene Arten, mehrere Batterien am Interface abzubilden.
Es ist schon ein Kreuz mit der (mangelnden) Dokumentation :-/

Matis-11, 2022-03-21:
Ich hab noch die 2021_08 SW auf der P10. Bei mir der Update immer noch nicht erfolgt.
Vielleicht spielt das eine Rolle?

git-kick, 2022-03-21:
OK danke, dann ist die Erkenntnis wenigstens eindeutig: in der offiziellen Tag-Liste steht fälschlicherweise der Kommentar, dass der Batterie-Index _immer_ 0 ist. Der Adapter muss also auch den BAT_INDEX = 1 abfragen, was er bis v1.0.2 tat und auch ab v1.0.4 wieder tun soll.

closed 2022-03-21 by git-kick