git-kick/ioBroker.e3dc-rscp #97: connection error

bushmills1608, 2022-03-20:
Ich bekomme diese Fehler im Protokoll gezeigt.

2022-03-20 20:43:50.929 | info | Connection to E3/DC is established
e3dc-rscp.0 | 2022-03-20 20:43:50.925 | info | Connection to E3/DC is established
e3dc-rscp.0 | 2022-03-20 20:43:18.926 | error | Failed writing data to socket
e3dc-rscp.0 | 2022-03-20 20:42:54.606 | warn | E3/DC connection closed
e3dc-rscp.0 | 2022-03-20 20:42:54.604 | error | E3/DC connection error
e3dc-rscp.0 | 2022-03-20 20:40:41.251 | info | Reconnecting to E3/DC ...
e3dc-rscp.0 | 2022-03-20 20:40:31.253 | warn | E3/DC connection closed
e3dc-rscp.0 | 2022-03-20 20:40:31.248 | error | E3/DC connection error
e3dc-rscp.0 | 2022-03-20 20:38:18.801 | info | starting. Version 1.0.2 in /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v16.14.2, js-controller: 4.0.21

Systeminfo
[image]
Adaptereinstellungen
[image]

die node v14.19 habe ich gerade auf 16.14 aktualisiert.
Der Fehler tritt bei beiden Versionen auf.

git-kick, 2022-03-21:
Die signifikate Zeile im Log ist 
`e3dc-rscp.0 | 2022-03-20 20:40:31.248 | error | E3/DC connection error`
Diese Meldung kommt ausschließlich dann, wenn die TCP-Verbindung einen Fehler wirft.
Wenn das immer gleich so auftritt, kann `net.Socket()` keine (stabile) Verbindung zu 192.168.170.125:5033 aufbauen.

Findet die iobroker.discovery deine E3/DC-Anlage?

bushmills1608, 2022-03-21:
Ja, das ist richtig, eine Pink Antwort bekomme ich auch nicht.
Ich habe auch Probleme die e3DC über Modbus abzufragen. 
Dann bekomme ich Socket Error

git-kick, 2022-03-21:
Dann ist dein LAN noch nicht ganz o.k., da kann der Adapter nichts machen...

closed 2022-03-21 by git-kick

bushmills1608, 2022-03-21:
Ich blockiere mir mit Modbuszugriff auf einen MYPV Elwa-E die TCP Verbindung zur E3DC, der auch mit  dem S10  in Verbindung steht.

bushmills1608, 2022-03-21:
Nach dem Neustart aller Geräte funktionierte Der Ping auf den S10.
Jedoch bekomme ich nach Neustart des Adapters diese Meldung:
Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)

git-kick, 2022-03-21:
Das spricht für sich: deine Anlage betrachtet - anders als andere Anlagen - das Tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER als nicht verfügbar, warum auch immer...
Da ich keine Dokumentation von E3/DC über solche "feinen Unterschiede" habe, kann sie der Adapter auch nicht abfangen.
Die restliche Funktion des Adapters wird dadurch nicht beeinträchtigt.

bushmills1608, 2022-03-22:
Der Verbindungsfehler ist nicht mehr aufgetreten.
Frage : kann ich von der, an die S10 Angeschlossene MYPV ELWA-E,  auch Daten abfagen?

git-kick, 2022-03-22:
Der Adapter spricht nur mit einer S10 - wenn das angeschlossene Gerät entsprechend von E3/DC integriert ist, könnten auch Daten des angeschlossenen Geräts im Adapter sichtbar sein (Beispiel Wallbox).
Der Adapter spricht kein anderes Gerät direkt an.

bushmills1608, 2022-03-22:
Das ELWA-E ist in ein der S10 sichtbarer Heizstab der überschüssigen Stom durch S10 gesteuert wird