git-kick/ioBroker.e3dc-rscp #100: Decoding of payload for tag TAG_WB_EXTERN_DATA_ALG

rzumbuehl, 2022-04-02:
By doing some experiments with my S10 system and the E3/DC Wallbox I figured out that the decoding of the tag `TAG_WB_EXTERN_DATA_ALG` as specified in `RscpTags.json` is probably wrong. 

I observed the following mapping of the `EXTERN_DATA` payload:
(Byte 1: uint8, PreCharge in [%])
Byte 2: uint8, Anzahl akt. Phasen [0-3]
Byte 3: uint8, 0x10 : Auto lädt, 0x20: Type2 verriegelt
Byte 4: uint8, Ladestrom [A]
Byte 5: uint8, 0x40: Sonnenmode, 0x00: Mischbetrieb
(Byte 6: uint4 low, 1: Schuko belegt / uint4 high, 1: Schuko an)

(Could not verify the mapping for the bytes in brackets)

This might be relevant if this decoding is or will be available in the ioBroker (which I can't tell).

ArnoD15, 2022-04-02:
It would be interesting what firmware version is on your E3DC and Wallbox.
With version P10_2022_022, E3DC has also installed a firmware update for the Wallbox. It may be that the descriptions are no longer correct with the current firmware.

rzumbuehl, 2022-04-02:
I'm currently running S10_2022_022 on the S10. 
Where could I check the FW version running on the wallbox?

ArnoD15, 2022-04-02:
Enter the IP address of the Wallbox in the Browser, and the firmware version will appear on the bottom left of the login page.
[image]

rzumbuehl, 2022-04-02:
To my knowledge, our E3/DC WB is connected via some sort of (S?)bus to the S10 system. I'm not aware that it has its own IP address.

git-kick, 2022-04-03:
The decoding of the tag `TAG_WB_EXTERN_DATA_ALG` as specified in RscpTags.json is taken 1:1 from the original E3/DC tag list. Is it possible that `TAG_WB_EXTERN_DATA_ALG` has a deviating encoding for non-E3/DC wallboxes?
Can we see the type of connected wallbox somewhere and then derive the encoding schema?
(I cannot help in finding out such rules because I have no E3/DC attached wallbox.)

rzumbuehl, 2022-04-03:
I can confirm that we have an E3/DC original wallbox installed but it's a rather old pre-Easy Connect wallbox (installed in 2016). 
Where could I look up the E3/DC tag list on my own? Is there any sort of specification available online? I could not find related specs in the technical documentation of the E3/DC system (other than some general RSCP protocol specs). 
But never mind, it's not that critical for me since I figured out a mapping that works for me. It just might be relevant if other users experience issues with this particular tag. Thanks for your support!

closed 2022-04-14 by rzumbuehl