git-kick/ioBroker.e3dc-rscp #102: MAX_DISCHARGE_POWER has no effect

christoph-spielmann, 2022-04-09:
If I set MAX_DISCHARGE_POWER to a value, for example 0 Watt or 100 Watt, the battery will still be discharged with a higher power (e.g. 5000 Watt).

I would like to reduce the discharge power to 0 if I want to charge my Tesla and the power for this should not come from the battery but from the grid. This is sometimes preferable when I need to fully charge the Tesla but have no solar power.

git-kick, 2022-04-09:
Thanks for your report. Have you made sure that POWER_LIMITS_USED ist `true`?
(Otherwise, the E3/DC device ignores MAX_DISCHARGE_POWER.)

christoph-spielmann, 2022-04-09:
>> Have you made sure that POWER_LIMITS_USED is true?
>> (Otherwise, the E3/DC device ignores MAX_DISCHARGE_POWER.)

This was the cause of my Problem. Thank you very much for your help!

closed 2022-04-09 by christoph-spielmann