git-kick/ioBroker.e3dc-rscp #103: SET_POWER Wiederholintervall ist auch von Einstellung Abfrageintervalle abhängig

ArnoD15, 2022-04-19:
**Describe the bug**  
Das Sende Intervall von SET_POWER_VALUE ist aktuell von zwei einstellbaren Zeiten im Adapter abhängig.
Einmal von der Einstellung SET_POWER Wiederholintervall und dann noch mal von der Einstellung der Abfrageintervalle im Adapter.

**Versions:**  
 
    Adapter version: 1.0.4
    JS-Controller version: 4.0.21
    Node version: 14.19.0
    Operating system: linux

**Additional context**  
Das Sende Intervall von SET_POWER_VALUE darf nur von der Einstellung SET_POWER Wiederholintervall abhängig sein, da es sonst zu Verwirrungen führt.

git-kick, 2022-04-20:
@ArnoD15 , danke für den Report.
Ich stehe aber auf der Leitung: mir ist nur das Feld `SET_POWER Wiederholintervall [s]` in den Adaptereinstellungen bewusst. In der `Einstellung der Abfrageintervalle` kann man nur einstellen, wie oft REQ_ Tags gesendet werden, nicht wie oft das SET_POWER Tag gesendet wird.
Was genau hast du beobachtet?

ArnoD15, 2022-04-20:
Ja, das ist ja das komische. Es wird aber trotzdem die Zeit in den Einstellungen Abfrageintervalle/Abfrageintervalle S=kurz [s] berücksichtigt bzw. verwendet.
Kannst du einfach testen, mach mal folgende Einstellung:
SET_POWER Wiederholintervall = 2 sek.
Abfrageintervall S = kurz = 60 sek.

Dann wirst du im LOG sehen das alle 2 sek. "queueEmsSetPower( 3, 1100 )" auftaucht, aber erst nach 60 sek. tatsächlich alle in der Warteschlange gespeicherten anfragen auf einmal an die E3DC Steuerung geschickt werden. Wenn du die Zeit bei den Abfrageintervalle Abfrageintervall S = kurz dann auf 120 sek. änderst, wird auch erst nach 120 sek. alle gespeicherten anfragen auf einmal an die E3DC Steuerung geschickt.

git-kick, 2022-04-21:
Ok verstanden, danke - dann habe ich offenbar bei den Timern was falsch programmiert. Muss ich tiefer reinschauen...

git-kick, 2022-04-23:
Ja, tatsächlich wurden die SET_POWER tags nur in die Queue geschoben und die wird nur im Takt der S-Abfrageintervalle bearbeitet. Ich habe jetzt die "schnelle Schiene" für SET_POWER eingebaut. Das ging etwas in die Eingeweide des Codes, deshalb bitte ich dich erstmal eine Weile zu testen:
https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23103

Bei mir sehe ich jetzt erstmals eine saubere, flache Ladekurve, wenn ich die Leistung einstelle (blaue Kurve ab 12:13:40 auf 350 Watt):
[image]

Wichtig dabei: der bisherige Default von 15 Sekunden für die SET_POWER Wiederholung reicht nicht (obwohl in der E3/DC Doku von 30 Sekunden die Rede ist). Ich habe jetzt 10 Sekunden eingestellt, dann bekomme ich obiges Bild.

Bin gespannt, wie es bei dir läuft...

ArnoD15, 2022-04-23:
Ok super, werde ich morgen mal testen, wenn es hoffentlich nicht zu bewölkt ist.

ArnoD15, 2022-04-24:
Wollte es heute testen, aber der Adapter scheitert bereits bei der Anmeldung am E3DC.
Die Anmeldedaten sind sicher richtig eingetragen, das habe ich zweimal überprüft, es kommt aber immer folgende Fehlermeldung im LOG:
`e3dc-rscp.0 | 2022-04-24 11:37:09.583 | error | Authentication against E3/DC failed - check adapter settings, then restart instance.
`
Was mir aufgefallen ist, dass das Passwort nicht mehr im LOG auftaucht, hast du da was geändert ?

ArnoD15, 2022-04-24:
Ok konnte es jetzt testen und es funktioniert jetzt so weit das nur noch die Zeit SET_POWER Wiederholintervall berücksichtigt wird.
Dass man für die SET_POWER Wiederholung min. 10 sek. einstellen muss hatte ich bei meinem ersten Test auch festgestellt und deswegen auf 9 sek. eingestellt.
Ich habe jetzt nur noch ein Problem bzw. ein ungutes Gefühl was die automatischen Wiederholungen von SET_POWER angeht.
Ich versuche das mal an einem Beispiel zu erklären.
Wenn ich über mein Script den Wert SET_POWER und SET_POWER_MODE ändere, um die Regelung zu beeinflussen, würde es keiner mehr mitbekommen, wenn mein Script wegen eines Fehlers stoppt oder auch einfach angehalten wird.
Das würde jetzt bedeuten, dass die Regelung mit der letzten Einstellung einfriert und meine Batterie dauerhaft geladen oder entladen wird, da der Adapter die letzte Einstellung ja ständig wiederholt.
Ist es möglich das so zu ändern, dass wenn in den Instanz-Einstellungen SET_POWER Wiederholintervall 0 sek. eingestellt wird, dann nicht automatisch vom Adapter wiederholt wird, sondern nur wenn sich der State SET_POWER und SET_POWER_MODE ändern, bzw. aktualisieren um auch gleiche Werte zu berücksichtigen?
Vermutlich ist nämlich genau, dass der Grund, warum E3DC alle 10 sek. eine Wiederholung erwartet, um sicher zu sein, dass noch jemand da ist, der regelt.

git-kick, 2022-04-24:
Gute Idee. Hab ich im selben [Branch ](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23103) eingebaut. Das kommt auch meiner Denkweise "der Adapter soll möglichst wenig eigene Logik haben und damit anwendungsneutral bleiben" entgegen. (Hab aber das Wiederholen vorläufig auch noch drin gelassen, um mögliche Verwender nicht mit einem Rückschritt zu enttäuschen.)

Btw, man kann auch ohne PV-Leistung testen: mit SET_POWER_MODE = GRID_CHARGE holt sich die E3/DC den nötigen Strom aus dem Netz.

ArnoD15, 2022-04-24:
Funktioniert jetzt super :-) 
Danke für die schnelle Umsetzung.
Kann man dir eigentlich mal einen Kasten Bier spendieren für deine tolle Arbeit?
Konnte keinen donate Button finden.

git-kick, 2022-04-25:
> Konnte keinen donate Button finden.

OK, diesen Bug behebe ich natürlich sehr gerne:-;  siehe README (gleich unter der Überschrift)
Obwohl das für mich wirklich Hobby im besten Sinne ist, außerdem habe ich manchmal den Eindruck, dass du genauso viel Zeit in den Adapter investierst wie ich, danke für die super Beiträge!

(Release v1.0.5 mit dem bugfix folgt demnächst.)

closed 2022-04-25 by git-kick