git-kick/ioBroker.e3dc-rscp #106: Wallbox-Ladeleistung einstellen

mccrossen, 2022-06-08:
Es wäre praktisch wenn man die maximale Ladeleistung der E3DC-Wallbox via ioBroker konfigurieren könnte.

Über die RSCPGui (https://github.com/rxhan/RSCPGui) funktioniert es, nur leider weiß ich nicht genau welche der fünf Vorkommen die richtige Bitmaske für das setzen der Ladeleistung wäre:
- https://github.com/rxhan/RSCPGui/blob/400c36bac3d3258be9307eb77b381acdb13b871b/e3dc/rscp_tag.py#L1063
- https://github.com/rxhan/RSCPGui/blob/400c36bac3d3258be9307eb77b381acdb13b871b/e3dc/rscp_tag.py#L1120
- https://github.com/rxhan/RSCPGui/blob/400c36bac3d3258be9307eb77b381acdb13b871b/e3dc/rscp_tag.py#L1141 (wohl die richtige)

Eventuell kannst du es zum setzen vorbereiten und schreibst mir an welcher Stelle ich es ändern muss um es ausprobieren zu können.

mccrossen, 2022-06-10:
Hallo,
ja leider kenne ich mich mit der Adapter-Entwicklung für ioBroker so überhaupt nicht aus. Ich kann das gern versuchen über einen Fork zu erarbeiten oder mir einen Branch aus deinem Repo ziehen. Ja ich habe eine E3DC Easy Connect Wallbox und kann es über meinen ioBroker auf dem Raspi testen, habe mir eine entsprechende Entwicklungsumgebung für diesen Adapter bereits erstellt.

Ich habe heute schon mal folgendes heraus gefunden:

Die aktuelle Ladeleistung wird tatsächlich bereits von deiner Lösung ausgelesen.
Im Feld `e3dc-rscp.0.WB.WB_0.EXTERN_DATA_ALG` sind diese Daten in Hex enthalten.
Hier ein Beispiel-Datensatz: `50 01 80 06 00 00 00 00`

Dabei habe ich folgendes nachvollziehen können:
`50 XX YY ZZ 00 00 00 00`

**XX** stellt die Anzahl der konfigurierten Phasen dar. 
- ``01`` steht für 1 Phase 
- ``03`` steht für 3 Phasen

**YY** stellt die Konfiguration des Sonnenmodus dar
- ``00`` Sonnenmodus nicht aktiv
- ``80`` Sonnenmodus aktiv

**ZZ** stellt die eingestellte Ladeleistung in Ampere als HEX 
- ``06`` steht für 6 A (min. Wert)
- ``07`` steht für 7 A
- ``08`` steht für 8 A
- ``09`` steht für 9 A
- ``0A`` steht für 10 A
- ``0B`` steht für 11 A
- ``0C`` steht für 12 A
...
- ``1E`` steht für 30 A
- ``1F`` steht für 31 A
- ``20`` steht für 32 A

Ist jetzt die Frage wie könnte man die Werte in einzelne ioBroker-Objekte zerlegen. Hast du da eine Idee?

Im Feld `e3dc-rscp.0.WB.WB_0.EXTERN_DATA` erscheint auch die eingestellte Ladeleistung: `00 00 06 00 00 00 00 00`
Der Datenwert ist dort hier: `00 00 ZZ 00 00 00 00 00`
Doch leider springt das gesamte Feld auch immer wieder auf `00 00 00 00 00 00 00 00` um und ich weiß nicht warum.

Das Abfrageintervall von `e3dc-rscp.0.WB.WB_0.EXTERN_DATA_ALG` könnte man jedenfalls schon mal auf `S` ändern.

git-kick, 2022-06-13:
Das Zerlegen solcher binär codierten Werte ist an sich kein großes Problem, wenn man erstmal weiß, was sie bedeuten!
Also das kann ich einbauen, allerdings bin ich ab morgen bis Ende Juni auf Reisen, d.h. es geht erst im Juli weiter...

Streit187, 2022-07-31:
Hallo. 
Hier ist noch jemand mit der ed3c Wallbox und großem Interesse an der Funktionalität. Ausser testen werde ich nur vermutlich nicht viel beitragen können. 
VG

rzumbuehl, 2022-07-31:
Ich hatte ähnliche Erkenntnisse gehabt beim Reverse-Engineering von EXTERN_DATA_ALG: siehe #100 

Das Setzen der Ladeleistung habe ich in einem eigenen, non-ioBroker Fork so gelöst (Verwendung auf eigene Gefahr 😉): 

```
	queueWbSetChargingCurrent( value ) {
		this.clearFrame();
		const pos = this.startContainer( "TAG_WB_REQ_DATA" );
		this.addTagtoFrame( "TAG_WB_INDEX", "", 0 );
		this.addTagtoFrame( "TAG_WB_REQ_PARAM_1" );

		// Sub-container
		const wbFrame = this.startContainer( "TAG_WB_REQ_SET_EXTERN" );
		this.addTagtoFrame( "TAG_WB_EXTERN_DATA_LEN", "", 6);
		let wbValue = Buffer.alloc(6);
		wbValue[1] = value % 0x100
		this.addTagtoFrame( "TAG_WB_EXTERN_DATA", "",  bufferToString(wbValue));
		this.endContainer(wbFrame)
		this.pushFrame( pos );
		this.clearFrame();
	}
```

ArnoD15, 2022-07-31:
Nur mal eine Frage, warum verbindet ihr eure E3DC Wallbox nicht einfach per Modbus, um diese zu steuern?
Importdateien für die einzelnen Modbus Register könnt ihr hier finden:
[Modbus Register](https://github.com/ArnoD15/iobroker_E3DC/tree/master/Modbus%20Import%20Dateien%20Register)

Die Steuerung über die Verbindung e3dc-rscp Adpter -> E3DC Hauskraftwerk-> Modbus -> E3DC Wallbox finde ich etwas umständlich.
Wenn es euch nur um das Auslesen der Werte geht, ist der e3dc-rscp Adapter die bessere Wahl.

mccrossen, 2022-08-01:
> Nur mal eine Frage, warum verbindet ihr eure E3DC Wallbox nicht einfach per Modbus, um diese zu steuern? Importdateien für die einzelnen Modbus Register könnt ihr hier finden: [Modbus Register](https://github.com/ArnoD15/iobroker_E3DC/tree/master/Modbus%20Import%20Dateien%20Register) Anleitung findet ihr hier: [Wallbox.pdf](https://github.com/ArnoD15/iobroker_E3DC/blob/master/Anleitung%20Script%20Wallbox.pdf) Die Steuerung über die Verbindung e3dc-rscp Adpter -> E3DC Hauskraftwerk-> Modbus -> E3DC Wallbox finde ich etwas umständlich. Wenn es euch nur um das Auslesen der Werte geht, ist der e3dc-rscp Adapter die bessere Wahl.

Eigentlich wurde von E3DC immer kommuniziert, das man via Modbus nur Daten auslesen, jedoch nicht setzt kann.

rzumbuehl, 2022-08-01:
> Nur mal eine Frage, warum verbindet ihr eure E3DC Wallbox nicht einfach per Modbus, um diese zu steuern?
> Importdateien für die einzelnen Modbus Register könnt ihr hier finden:
> [Modbus [...]

Ich steuere die Wallbox grundsätzlich via Modbus (auch das Setzen des Solarmodus, Starten des Ladevorganges etc.). Nur das Setzen des Ladestromes scheint nur via RSCP möglich zu sein.

ArnoD15, 2022-08-01:
> Eigentlich wurde von E3DC immer kommuniziert, dass man via Modbus nur Daten auslesen, jedoch nicht setzt kann.

Das ist bei E3DC Hauskraftwerk auch richtig, nur bei der Wallbox steuert E3DC diese auch über Modbus.

> Nur das Setzen des Ladestromes scheint nur via RSCP möglich zu sein.

Ist auch über Modbus möglich, mit dem Holdingregister 528 kann man den Ladestrom in A einstellen und über das Holdingregister 300 bekommt man die Rückmeldung vom Fahrzeug, mit wie viel A gerade geladen wird.

Da E3DC unterschiedliche Hardware bei den Wallboxen verwendet, muss man natürlich immer prüfen, ob das bei euch funktioniert. 
Ich habe die E3DC easy connect mit wallbe EV Charge Control und dieser ist baugleich mit der Phoenix Contact EV-CC-AC1-M3-CBC-RCM-ETH. Wenn man im Internet nach diesem Typ sucht, findet man auch ein Anwenderhandbuch, wo die einzelnen Modbus Register genauer beschrieben sind.

git-kick, 2022-08-01:
> Hallo, ja leider kenne ich mich mit der Adapter-Entwicklung für ioBroker so überhaupt nicht aus. Ich kann das gern versuchen über einen Fork zu erarbeiten oder mir einen Branch aus deinem Repo [...]

@mccrossen ich habe mir das angesehen und folgenden Vorschlag: die "Zerlegung" (welche E3/DC zwar dokumentiert aber nicht macht), kann man im ioBroker javascript-Adapter nachrüsten und dabei auch beliebig formatieren. Das sieht in etwa so aus:

```
on( {id: 'e3dc-rscp.0.WB.WB_0.EXTERN_DATA_ALG', change: 'ne'}, (obj) => {
    setState('e3dc-rscp.0.WB.WB_0.PHASES', getState('e3dc-rscp.0.WB.WB_0.EXTERN_DATA_ALG').val.substring(2,4) )
	...
})
```
Wenn man dieses Skript im ioBroker startet, wird bei jeder Änderung von `EXTERN_DATA_ALG` der Wert von `PHASES` aktualisert, indem das 3. und 4. Zeichen aus `EXTERN_DATA_ALG` herausgeschnitten wird. D.h. der Code-Schnipsel würde nur das Byte in Hex-Darstellung ausgeben, aber eine Umrechnung in dezimal oder auch Decodierung in aktiv/inaktiv geht natürlich auch in Javascript.

Das Thema "Zerlegung" ist also auch ohne Änderung am Adapter lösbar.
(Wenn ich eine passende Wallbox hätte, würde ich es schon einbauen, aber so...)

git-kick, 2022-08-01:
> Das ist bei E3DC Hauskraftwerk auch richtig, nur bei der Wallbox steuert E3DC diese auch über Modbus.
>
>
> Ist auch über Modbus möglich, mit dem Holdingregister 528 kann man den Ladestrom in A [...]

@ArnoD15 danke für den Lösungsvorschlag!
Die Erweiterung am Adapter kann ich selbst nicht machen, wie [oben](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/106#issuecomment-1151283346) beschrieben.
Weiterhin gilt: wer mag, kann das natürlich in den e3dc-rscp Adaper einbauen und einen Pull Request anlegen, das wäre Klasse!

ka-vaNu, 2022-11-27:
@git-kick Ich habe ein wenig Erfahrung in der Adapter-Entwicklung und einer E3DC-Wallbox zuhause. Leider erschlägt mich die Adapter-Komplexität etwas. Daher kann ich meiner Hilfe beim entwickeln und testen anbieten. Finden wir hier einen gemeinsamen Weg zur Lösung?

git-kick, 2022-11-27:
@ka-vaNu, danke für dein Angebot. Es ist aber so (wie [oben](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/106#issuecomment-1151283346) beschrieben), dass man bei der Code-Entwicklung dauernd die Wallbox als "Gegenüber" braucht, um das Verhalten auszutesten. Da sehe ich aktuell keinen gangbaren Weg für eine verteilte Arbeit (coden hier, testen da).

ka-vaNu, 2022-11-27:
@git-kick Die Entwicklung kann ich relativ problemlos übernehmen. Ich verstehe nur nicht, wie ich zusätzliche Status z.B. für die Set-Request und die decodierten Werte erstellen kann. Hast du irgendeine Doku für RSCP? Wie kannst du die Kommunikation debuggen? Die main.js ist doch recht umfangreich...

git-kick, 2022-11-28:
**RSCP-Dokumentation**: eine wichtige Quelle ist das e3dc Portal:
[image]
In der "RSCP-Beispielapplikation" ist die Tag-Liste als Excel-Datei enthalten, da habe ich die meisten Informationen rausgezogen. Leider ist die Tagliste von 2016, also nicht vollständig und nicht aktuell. Es gibt hier und da Ergänzungen aus der Community, z.B. von rxhan.
Im e3dc-rscp-Adapter ist mein aktueller Kenntnisstand zusammengefasst in der Datei `RscpTags.json`.

Für das **Setzen von Werten** ist die Funktion `queueSetValue` (in main.js) zuständig.
Die greift via `getSetTags` auf die Werte in `mapChangedIdToSetTags` zu: hier (main.js#272ff) muss man am Anfang eintragen, welches (optional) Container- und Setter-Tag nötig ist, um einen Wert an die S10 zu übergeben. Am besten mal mit z.B. mit `EMS_MAX_CHARGE_POWER` durchspielen...

Das **Mapping/Codieren der Werte** erledigen Konstanten (main.js#70ff). Hier muss man für neue Wertebereiche neue Konstanten anlegen oder (wenn möglich) vorhandene wiederverwenden. Einige sind auch noch auskommentiert (z.B. `rscpWbType`), weil ich sie bisher nicht verwende. 

**Debugging** sollte erst ab Ebene der RSCP-Tags nötig sein (also nicht TCP, Verschlüssselung etc., das läuft inzwischen recht stabil). Hier arbeite ich mit log-level "silly" - der wirft schon alle Nachrichten mit Inhalt aus - und bei Bedarf zeitweise eingestreuten `this.log.debug()`.
Eleganter ist eigentlich ein richtiger Debugger - es gibt auch Anleitungen z.B. für VS Code, aber ich bin nicht so richtig damit warm geworden.

ka-vaNu, 2022-12-06:
Besten Dank! Das hat schon einmal sehr geholfen! 

Die Kommunikation mit der Wallbox funktioniert nun und ich kann die Ladeparameter schon übermitteln.

Hast du noch einen Tipp, wie ich neue ioBroker-Objekte in deinem Adapter korrekt anlege? Muss ich dazu einen Eintrag in der JSON-Datei vornehmen, zu dem kein RSCP-TAG spezifiziert ist. Existiert dazu schon ein Beispiel? Hintergrund: Ich würde das decodieren und codieren der Parameter gerne im Adapter lösen.

ka-vaNu, 2022-12-08:
Leider habe ich die Verarbeitung der RSCP-Container und -Tags noch nicht verstanden. Der u.g. Wallbox-Container wird leider nicht richtig in die Objekte e3dc-rscp.0.WB.WB_0.EXTERN_DATA_ALG, e3dc-rscp.0.WB.WB_0.EXTERN_DATA_ALL, e3dc-rscp.0.WB.WB_0.EXTERN_DATA_NET und e3dc-rscp.0.WB.WB_0.EXTERN_DATA_SUN übertragen. Stattdessen wird immer nur das innere Tag TAG_WB_EXTERN_DATA in das Objekt e3dc-rscp.0.WB.WB_0.EXTERN_DATA geschrieben. 

`processTree: path = , tree = [ {TAG_WB_CONNECTED_DEVICES(Container) = [ {TAG_WB_INDEX(Int32) = 0}, ]}, {TAG_WB_DATA(Container) = [ {TAG_WB_INDEX(UInt16) = 0}, {TAG_WB_EXTERN_DATA_SUN(Container) = [ {TAG_WB_EXTERN_DATA_LEN(Int32) = 8}, {TAG_WB_EXTERN_DATA(ByteArray) = 00 00 C8 84 00 00 64 00}, ]}, {TAG_WB_EXTERN_DATA_NET(Container) = [ {TAG_WB_EXTERN_DATA_LEN(Int32) = 8}, {TAG_WB_EXTERN_DATA(ByteArray) = 00 00 17 5B 02 00 00 00}, ]}, {TAG_WB_EXTERN_DATA_ALL(Container) = [ {TAG_WB_EXTERN_DATA_LEN(Int32) = 8}, {TAG_WB_EXTERN_DATA(ByteArray) = 00 00 DF DF 02 00 00 64}, ]}, {TAG_WB_EXTERN_DATA_ALG(Container) = [ {TAG_WB_EXTERN_DATA_LEN(Int32) = 8}, {TAG_WB_EXTERN_DATA(ByteArray) = 64 01 00 0E 00 00 00 00}, ]}, ]}, ]`

ka-vaNu, 2022-12-09:
Die Steuerung ist jetzt implementiert. Bitte einmal unter https://github.com/ka-vaNu/ioBroker.e3dc-rscp/archive/refs/heads/Support-Wallbox.zip prüfen. Wenn OK, erzeuge ich einen Pull-Request.

git-kick, 2022-12-09:
Danke und Gratulation! Ich habe gerade viel anderes um die Ohren und deshalb kann es etwas dauern mit einer Antwort. Aber ich werde mir das definitiv ansehen!

ka-vaNu, 2022-12-11:
@mccrossen, @Streit187, @rzumbuehl, @ArnoD15, @Matis-1, @christianreiss : Bitte Testen, bei euch hatte ich ebenfalls ein Interesse wahrgenommen

Folgende ioBroker-Objekte sind im Baum WB hinzugekommen:

**SUNMODE**
Tanken über PV-Anlage oder aus dem Speicher

**POWER_LIMITATION**
Begrenzung der Ladeleistung in Ampere

**PRECHARGE**
Funktion so nicht über die Oberfläche einstellbar, +/- 5%

**TOGGLE_PHASES**
Umschalten zwischen 1 und 3-phasigem Laden. Da im Moment die Befüllung DATA_EXTERN_ALG defekt ist (wir nur bei der initialen Erstellung des Objekts bei erstem Adapterstart befüllt) kann der aktuelle Stand nicht abgebildet werden.

**ABORT_CHARGING_TYP2**
Ladevorgang Typ2 Stecker abbrechen

`Byte 6: > 0: Schuko, Best�tigung für ?AN?`
Ist mir unklar und deshalb immer statisch auf 0 -> Vielleicht hat ja noch jemand eine bessere Information. Bei Zeiten kann ich es sonst vielleicht aus der RSCPGui-Implementierung herauspuzzeln...

christianreiss, 2022-12-12:
Yes1

Pure awesome! Thanks for doing this.
Got the zip file and... and then what? Javascript? main.js? Mhh, no Makefile. Checking Readme (grep -i install README.md) -- nothing but a License. Docs Directory? INSTALL file?

Does it plug into Homeassistant? According to homepage "Settings are organized in Tabs." -- what tabs? Where?

I am afraid I need some pointers here. Is it standalone? Do I need to compile something? Do I include it?

christianreiss, 2022-12-12:
Okay, I am dumb. Iobroker is an Home Automation Software... I am unaware just how I came across this.
As I am a home assistant user, is there a way to run this standalone and trigger things?

I just need a way to cancel charging...

ka-vaNu, 2022-12-12:
OK, this is about a pure ioBroker adapter. Only an installation via the GUI is required. These are installed via expert mode in beta phase or developer test. Please turn on the expert mode in the adapter menu, then you can select "Install from custom URL". There you can select "Custom" and paste the URL.
Only the main.js and admin/words.js have changed.
If the installation with the ZIP file does not work, please try https://github.com/ka-vaNu/ioBroker.e3dc-rscp/tree/Support-Wallbox.

git-kick, 2022-12-16:
> Die Steuerung ist jetzt implementiert. Bitte einmal unter https://github.com/ka-vaNu/ioBroker.e3dc-rscp/archive/refs/heads/Support-Wallbox.zip prüfen. Wenn OK, erzeuge ich einen Pull-Request.

Ich habe heute den Code gelesen und im Kern (`queueWbSetData`) passt das gut zu Adapter.
Ein paar kleinere Anmerkungen habe ich noch:

1. Durch das pretty printing habe ich sehr viele (Hunderte) Änderungen in `main.js`, kannst du deinen Editor so einstellen, dass er keine Leerzeichen einfügt usw?
2. Um den Objektbaum konsistent zu halten, sollten die neuen Werte unter einem Knoten `EXTERN_DATA` stehen und Namen in Gross/Kleinschreibung haben. Die Analogie sieht man unter EMS.SYS_SPECS, auch dort sind Werte, denen nicht 1:1 ein RSCP-Tag zugeordnet ist.
3. Die `log.debug` in den Zeilen 1065,1073, 1078 sollten eher `log.silly` sein.
4. Nur eine Wallbox mit Index 0 - das wäre wahrscheinlich lösbar analog `queueBatProbe, queuePviProbe`
5. Die `states`Wertebereiche ab Zeile 1850 sollten oben (ab 170) als Konstanten definiert werden. Dann kann man sie weiderverwenden, z.B. zum Decodieren.
6. A propos decodieren: kann man `EXTERN_DATA` eigentlich auch abfragen oder ist das "write-only". Wenn möglich würde ich auch abfragen, dann erkennt man Fehler beim Setzen.

Ich kann einiges davon auch selbst noch machen und dir dann zum Testen geben, es ist ja kaum etwas Funktionales.
Kannst du mir bitte mal einen gefüllten WB-Objektbaum posten (Grafik reicht), damit ich eine bessere Vorstellung habe.

Freue mich auf den Pull Request!

ka-vaNu, 2022-12-23:
zu 1.: Ich nutze immer die .eslinrc.json, da die Editoren hier oft unterschiedlich arbeiten. Die Konfiguration, die dem bestehenden Code am nächsten kam, habe ich unter https://github.com/ka-vaNu/ioBroker.e3dc-rscp/commit/b31628a9b6d4c30a49acaafc3064de887a190d8a commited. Falls du noch eine passende Konfiguration hast, kann ich die gerne übernehmen.

zu 2.: EinKnoten EXTERN_DATA ist so nicht korrekt. Die Daten liegen eigentlich in Containern wie z.B.: TAG_WB_EXTERN_DATA_SUN, TAG_WB_EXTERN_DATA_NET, TAG_WB_EXTERN_DATA_ALL etc. die dann wiederum ein ein EXTERN_DATA Tag enthalten. Aus meiner Sicht sollte eher etwas wie folgendes entstehen:
`
EXTERN_DATA_SUN
- RAW_DATA
- SUNPOWER
- SUNENERGY
- SUNPERCENTAGE
EXTERN_DATA_NET
- NETPOWER
- NETENERGY
- NETPERCENTAGE `

zu 3.: Ändere ich

zu 4.: Schau ich mir an, sollte vielleicht ein paralleles Enhancement werden, da mehrere WB eher die Ausnahme sein werden

zu 5. Ändere ich

zu 6.: Hängt eng mit 2. zusammen. Hier müssten die TAG_WB_REQ_EXTERN_DATA_ALL und TAG_WB_REQ_EXTERN_DATA_ALG regelmäßig abgefragt werden und dann entsprechend 2. die Container als Unterordner abgelegt werden.

Anbei der Screenshot: [image]

Dieser enthält schon die neuen Objekte und EXTERN_DATA ist aus meiner Sicht falsch dekodiert, da es immer wieder in unterschiedlichen Containern vorkommt.

christianreiss, 2023-01-02:
> @mccrossen, @Streit187, @rzumbuehl, @ArnoD15, @Matis-11, @christianreiss : Es wäre schön, wenn ihr den Code ebenfalls testen könntet, bei euch hatte ich ein Interesse wahrgenommen
>
> **ABORT_CHARGING_TYP2** Ladevorgang Typ2 Stecker abbrechen

Soo,  hab rscp mal "extern" zum laufen bekommen. Welchen Tag sendest du denn bei dem ABORT_CHARGING_TYP2 event? 

Das was ich finden konnte ist das hier:

```
                                                        const str = sunmode + " " + powerLimitation + " " + precharge  + " " + togglePhases  + " " + abortCharging  + " " + schuko;
                                                        this.log.debug( "String: " + str );
                                                        const buf = stringToBuffer( str );

```

Also du setzt Dir vorher sunmode, powerLimitation, Precharge, TogglePhases, AbortCharging und Schuko und sendest dann denn konkatenierten String an das e3dc?

Gibt es eine Möglichkeit, dass statisch via rscp zu machen? Oder sind alle Parameter von oben zwingend für jeden Abbruch der Ladung nötig?

git-kick, 2023-01-03:
> @git-kick Können wir uns auf eine eslint-Konfig einigen?

Ja, bestimmt. Ich bin noch ein paar Tage unterwegs, dann melde ich mich. 
Bisher habe ich nur nach Gusto Einstellungen im VS Code gemacht. Ich schaus mir an und melde dann, ob ich das hinbekommen habe und mit welcher eslint-config.

ka-vaNu, 2023-01-04:
Ich verwende ebenfalls VS Code und habe dort die ESLint (v2.2.6) Extension installiert. Diese formatiert beim Speichern entsprechend der Regeln in der lokalen .eslintrc.json

ka-vaNu, 2023-01-04:
Ich habe heute noch alle Wallbox-relevanten Teile in eine Bibliothek ausgelagert, dadurch wird der Code deutlich übersichtlicher und wir kommen uns hoffentlich nicht groß in die quere, dafür habe ich ein paar Hilfsmethoden in die tools.js auslagern müssen.

ka-vaNu, 2023-01-04:
Zum Testen kann der Adapter mit folgendem Befehl installiert werden:
`npm i https://github.com/ka-vaNu/ioBroker.e3dc-rscp/tarball/Support-Wallbox`

Danach einfach eine Instanz erzeugen.

git-kick, 2023-01-06:
> Ich verwende ebenfalls VS Code und habe dort die ESLint (v2.2.6) Extension installiert. Diese formatiert beim Speichern entsprechend der Regeln in der lokalen .eslintrc.json

`.eslintrc.json` war ja bei mir auch aktiv, nur etwas laxer eingestellt bzgl. der Spaces. Ich habe deine Änderungen (PR#135) jetzt  übernommen, damit wir auf dem gleichen Stand sind. Für die Zukunft habe ich aber gelernt, dass Contributors die `.eslintrc.json` einfach übernehmen können, dann sollte es keine Diskussionen über die Code-Formatierung geben.

git-kick, 2023-01-06:
> Ich habe heute noch alle Wallbox-relevanten Teile in eine Bibliothek ausgelagert, dadurch wird der Code deutlich übersichtlicher und wir kommen uns hoffentlich nicht groß in die quere, dafür habe ich ein paar Hilfsmethoden in die tools.js auslagern müssen.

Die `tools.js` stammt nicht von mir, deshalb würde ich Adapter-eigene Funktionen nicht dort einfügen. Besser würden mir eine `helper.js` und `wallbox.js` auf root-Ebene gefallen.   

Gibt es einen git-Branch, wo ich den aktuellen Stand sehen kann?

ka-vaNu, 2023-01-06:
> Die `tools.js` stammt nicht von mir, deshalb würde ich Adapter-eigene Funktionen nicht dort einfügen. Besser würden mir eine `helper.js` und `wallbox.js` auf root-Ebene gefallen.

OK, habe die Bibliotheken auf die Root-Ebene verschoben und die tools.js wieder aufgespalten und in helper.js ausgelagert.
 
> Gibt es einen git-Branch, wo ich den aktuellen Stand einsehen kann? 

Ich habe dein Repository geforked und einen "Support-Wallbox"-Branch eröffnet, dort habe ich die Entwicklung durchgeführt:
https://github.com/ka-vaNu/ioBroker.e3dc-rscp/tree/Support-Wallbox

> Schön wäre auch (zu Info) nochmal den aktuellen Baum mit den `EXTERN_DATA...` Werten in den Container-Knoten zu sehen.

[image]

> Ich würde dann eine v1.1 anstreben, weil wir hier eine größere Änderung haben.

Sehr gerne!

ka-vaNu, 2023-01-15:
Pull-Request ist gestellt

git-kick, 2023-01-15:
Danke @ka-vaNu. Ich bitte um etwas Geduld bis zur v1.1. Ich möchte noch die Dokumentation anpassen und ein paar andere Kleinigkeiten fixen.

Lucifor1976, 2023-01-19:
Nice, danke euch für die Arbeit! Mein e3dc samt Wallbox ist seit 2 Tagen im Betrieb. Haut raus das Ding, ich freue mich tierisch darauf.

git-kick, 2023-01-22:
So, der Pull Request #136 ist jetzt merged und der Adapter läuft mit dem Wallbox-Modul.
**@Wallbox-Besitzer**, könnt ihr bitte diesen [Branch](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/prepare-v1.1.0)
testen, bei mir erscheint naturgemäß keine Wallbox.
Sobald ich hier ein oder zwei "ok" habe, geht die v1.1.0 auf release.

ka-vaNu, 2023-01-22:
Hab den Branch mittels `npm i https://github.com/git-kick/ioBroker.e3dc-rscp/tarball/prepare-v1.1.0` unter `/opt/iobroker/node_modules` installiert und getestet. 

Aus meiner Sicht ist alles OK. 

- Die Phase lassen sich umschalten
- Die Ladeleistung kann begrenzt werden
- Es kann zwischen Sonnenmodus und Mixedmode umgeschaltet werden 
- Der Ladvorgang kann gestartet bzw. abgebrochen werden.

Lucifor1976, 2023-01-22:
> So, der Pull Request #136 ist jetzt merged und der Adapter läuft mit dem Wallbox-Modul. **@Wallbox-Besitzer**, könnt ihr bitte diesen [...]

Lüppt .. :)

ka-vaNu, 2023-01-22:
> Falls es noch auf die schnelle geht: Könnt Ihr die Schlüsselschalterfunktion Laden verboten 0 / Laden erlaubt 1 einbauen?
> Das Kraftwerk lässt zu zu entscheiden was der Schlüsselschalter tun soll Nativ: Stellung Null Überschussladung PV / Stellung 1 Laden PV und Netz.
> Mir wäre die Funktion Laden verboten/erlaubt lieber, da die Station öffentlich erreichbar ist. Da ich die Box aber vermiete ist das gerenne in den Keller um die Schlüsselfunktion zu ändern recht lästig.
Ich habe deine Anforderung noch nicht ganz verstanden, willst du die Funktionen des Schlüsselschalters fernsteuern oder willst du Laden erlauben/verbieten umschalten. Ich habe spontan in der Tag-Liste kein entsprechendes RSCP-Tag gefunden...

git-kick, 2023-01-22:
So, v1.1.0 ist released und PR für [ioBroker.repositories](https://github.com/git-kick/ioBroker.repositories) ist erstellt.
Danke für das Teamwork!

closed 2023-01-22 by git-kick

j64bit, 2023-10-03:
Hallo liebe Gurus der Wallboxerweiterung!

Habt Ihr Lust vielleicht noch eine sinnvolle Erweiterung einzubauen?
Vorschlag findet sich hier: 
[](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/185#issue-1923455318)
#185 

Das wäre super - Danke