git-kick/ioBroker.e3dc-rscp #109: E3DC e3dc-rscp.0.EMS.WEATHER_REGULATED_CHARGE_ENABLED cannot be set to TRUE again

Assemblerguru, 2022-06-28:
In Version 1.0.5. e3dc-rscp.0.EMS.WEATHER_REGULATED_CHARGE_ENABLED can be set to FALSE to set off the weather regulation feature. But If the value is set to TRUE it is not set back to true. Instead E3DC`s value is left false and the valuer of the adapter is set back to FALSE again after some time. This does not happen in previous used Version 1.0.0 that I have used before.

Steps to reproduce the behavior:
1. In IoBroker "Objects" go to ''E3DC-rscp" and open EMS-folder. Go to the last entry "e3dc-rscp.0.EMS.WEATHER_REGULATED_CHARGE_ENABLED" and check if it it set to FALSE. Also check the state of the Weather-Regulation on your E3DC via Web-Portal or on the Display. It is OFF, too. -> OK !
2. Enable Weather regulation of E3DC via Web_portal or Display -> The object in IoBroker is also set to true after some time -> OK !
3. Disable Weather regulation of E3DC via Web or Display -> The object is set to false again -> OK !
4. Repeat step 2 (switch on via Portal again) ->OK
5. Set object in ioBroker to FALSE -> E3DC switches the state also to Off after some time -> OK !
6. Enable e3dc-rscp.0.EMS.WEATHER_REGULATED_CHARGE_ENABLED and set it to TRUE -> E3DC switches NOT to true but after some time the object in IOBroker is se to FALSE again -> NOK !

So it is not possible to switch the weather regulation on again.

**Expected behavior**  
The Weather-Regulation can also be switched on via IoBroker object e3dc-rscp.0.EMS.WEATHER_REGULATED_CHARGE_ENABLED.

Story:
Using the weather regulation to manually start my on charging script that calculates loading time and prevent block loading in the morning if SOC is 0. POWER_LIMITS_USED can only be switched and used if STATUS_0 (Laden nicht erlaubt, Lock-Symbol in Display)  is false, but setting STATUS_0 to false cannot be set manually - instead you must set Weatherregulation to false and true again for resetting STATUS_0.

**Versions:**  
 - Adapter version: 1.0.5
 - JS-Controller version: 4.0.23
 - Node version: v14.19.1
 - Operating system: raspberry pi4 IoBroker

git-kick, 2022-07-01:
@Assemblerguru, thanks for your detailed report. The described behaviour is not intended, but I currently cannot say why it was introduced with v1.0.3. 

I will do some code analysis/testing and then update here. (Due to holiday, this will take some time; sorry for that.)

git-kick, 2022-07-10:
I found the mistake. After setting boolean values (i.e. Switches) in the object tree, the adapter always sent 0 (zero) due to a type conversion issue.
You can try the solution by installing from [master branch](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/master).
Will be released with v1.0.6 soon.

closed 2022-07-14 by Assemblerguru