git-kick/ioBroker.e3dc-rscp #111: State bei Erstellung undefiniert

ArnoD15, 2022-07-13:
**Describe the bug**  
Die beiden State e3dc-rscp.0.EMS.SET_POWER_VALUE und e3dc-rscp.0.EMS.SET_POWER_MODE sind bei der Erstellung undefiniert (null)

**To Reproduce**  
State Löschen und Instanz neu starten

**Expected behavior**  
State werden mit 0 definiert

**Versions:**  
 - Adapter version: 1.0.6

**Additional context**  
Hintergrund ist, dass mein Script bei der Prüfung, ob diese State vorhanden sind, mit dem Befehle "existsState" in einen Fehler läuft, wenn dort kein Wert eingetragen ist.

git-kick, 2022-07-14:
Ich habe folgendes Experiment gemacht:
1. State Löschen und Instanz neu starten
2. Folgendes Skript starten: 
`if( existsState('e3dc-rscp.0.EMS.SET_POWER_VALUE') ) {console.log("existiert")} else {console.log("existiert nicht")}`

Ergebnis: das Skript schreibt wie erwartet "existiert nicht" und terminiert ganz normal.
Ich sehe das Problem nicht, kannst du es näher beschreiben?

Btw, undefiniert (null) ist eigentlich korrekter als 0, denn die Werte sind ja nicht 0, sondern eben noch undefiniert.
Werte im Objektbaum werden generell nur aus der Schnittstelle oder vom Anwender befüllt, da möchte ich möglichst wenig Ausnahmen machen.

ArnoD15, 2022-07-14:
Genau das ist mein Problem. Weil bei der Abfrage 
if( existsState('e3dc-rscp.0.EMS.SET_POWER_VALUE') ) {console.log("existiert")} else {console.log("existiert nicht")}

"false" zurückgegeben wird und dadurch bei mir eine Fehlermeldung ausgegeben wird, die darauf hinweist, dass eben diese beiden State nicht existieren, was ja eigentlich nicht stimmt, sie sind nur nicht definiert.
Ich könnte das dahingehend ändern, dass ich nur den Adapter abfrage, ob dieser vorhanden ist und dann einfach davon ausgehe, dass auch die beiden State angelegt worden sind.
Kann man eigentlich die ID abfrage und nicht direkt den State ?
Las mal alles so und ich schaue wie ich das anders lösen kann, du hast ja nicht unrecht mit deiner Argumentation.

git-kick, 2022-07-15:
Ah gut. Bevor ich deine Antwort las, habe ich folgenden Workaround mit `existsState` gefunden:

```
//id = 'e3dc-rscp.0.EMS.POWER_BAT'
id = 'e3dc-rscp.0.EMS.SET_POWER_VALUE'
existsState( id, function (err, isExists) {
        if( isExists === null ) { 
            console.log("State exisitiert nicht") } 
        else {
            console.log( "State existiert, Wert ist " )
            if( isExists === false ) { 
                console.log( "null")
            } else {
                console.log( getState(id).val )
            } 
        }
    }
)
```

Vielleicht kann man's mal brauchen ;-)