git-kick/ioBroker.e3dc-rscp #112: No PV-String Data with V1.0.6

creecool, 2022-07-18:
After the installation of Version 1.0.6 the system does not receive data from/for the PV String 1 & 2.
There are no error information in the Logfile.

After a downgrade to 1.0.5 it is working. 

Picture V1.0.6

[image]

Picture V1.0.5
[image]

creecool, 2022-07-24:
Hi, I have installed the version 1.0.6 via GitHub and now it is working. I will test it for the next days. 
I have an other question, where can I find the actual software version f.e S10_2022_02x of the device? Is this working via RSCP request? 

Thx

git-kick, 2022-07-24:
No bug.

closed 2022-07-26 by creecool