git-kick/ioBroker.e3dc-rscp #113: EP.PARAM_0.PARAM_EP_RESERVE no update

valbuz, 2022-07-29:
Hi

When i change the Parameter for the Batt Reserve in my E3DC System, the Paramter in the Object doesn't change.
Only when i restart the Instanz, the correct value is reading.

Is there a workaround for that?

pat

git-kick, 2022-07-29:
Hi pat, 
I cannot figure out what exactly you mean by `Batt Reserve` - please always specify the object id (copy/paste from object tree) to avoid ambiguities. 
I assume your observation is due to polling intervals. By default, not all values are updated frequently - we saw performance issues on small ioBroker Hardware (Raspi). So we decided to group parameters in S, M, L polling intervals. These are set and assigned by default, but can be adjusted in the adapter's admin tab "Polling Intervals".
If the parameter you are observing has "M" or even "L" there, set it to "S" and you will get instant updates.
Uli

valbuz, 2022-07-29:
Hi Uli
thanks for reply.
Itry it again :-) (perhaps next time in german)

object id: e3dc-rscp.0.EP.PARAM_0.PARAM_EP_RESERVE

i have change this Paramter on the LCD of my E3DC System. Now after one week, the value isn't updatet in the tree.
After Restart the Instance, the value is updatet.

thanks again
Pat

valbuz, 2022-08-06:
Hi
Do you found the Parameter?
pat

git-kick, 2022-08-06:
Did you change the polling interval to "S" ?
Is the problem still there afterwards?

valbuz, 2022-08-06:
I can't find the Parameter in the Polling List. So i can change it to "S" or else.

object id: e3dc-rscp.0.EP.PARAM_0.PARAM_EP_RESERVE

valbuz, 2022-08-09:
Did you see that the parameter is not in the polling list?

git-kick, 2022-08-14:
As described in [README](https://github.com/git-kick/ioBroker.e3dc-rscp#tab-polling-intervals), due to the E3/DC interface design there is no one-to-one relationship between polling requests and object tree tags.

After scrolling thru the silly log, I assume that `TAG_EP_REQ_IS_READY_FOR_SWITCH` is the request tag you have to address in the polling interval tab, because it's response contains `PARAM_EP_RESERVE`.

Please note that an interface adapter's primary task is to make the original interface available, not to explain or "transform" it somehow. So everybody using it has to face the peculiarities of the interface; this is not per se a shortcoming of the adapter.

closed 2022-08-14 by git-kick