git-kick/ioBroker.e3dc-rscp #116: Falsche Objektdaten

peterle-k, 2022-09-08:
**Describe the bug**  
im Ordner e3dc-rscp.0.BAT.BAT_0.DCB_0.DCB_CELL_TEMPERATURE ist bei Zelle 02 der Datenpunkt:

`  "common": {
    "name": "Zellspannung",
    "type": "number",
    "role": "value",
    "read": true,
    "write": false,
    "states": null,
    "unit": "V"
  },`

müsste aber so in der Art sein:
`
  "common": {
    "name": "Zelltemperatur",
    "type": "number",
    "role": "value",
    "read": true,
    "write": false,
    "states": null,
    "unit": "°C"
  },`

**Screenshots & Logfiles**  
[image]

**Versions:**  
 - Adapter version: 1.07
 - JS-Controller version: v 4.0.23
 - Node version: v 16.17.0
 - Operating system: raspbian

git-kick, 2022-09-09:
@peterle-k, danke für den Report.

Anscheinend hat das E3DC-Interface einen Bug und liefert Spannungswerte in Temperatur-Containern und umgekehrt. Das habe ich pragmatisch gelöst und unterscheide am willkürlich gewählten Schwellert 4,5
```
	if( tagName == "DCB_CELL_TEMPERATURE" ) {
		if( token.content > 4.5 ) {
			unit = "°C";
		} else { // low values are regarded as voltage (seems to be a flaw in RSCP interface)
			unit = "V";
			dictionaryIndex = "DCB_CELL_VOLTAGE";
		}
	} else if( tagName == "DCB_CELL_VOLTAGE" ) {
		unit = "V";
	}
```
D.h. nur Werte größer 4,5 werden als Temperatur behandelt, alle andern (auch `null`) als Spannung.
Mir fällt keine "saubere" Lösung ein, Vorschläge sind willkommen.

peterle-k, 2022-09-09:
Soll ich bei E3DC eine Supportanfrage machen?

Before printing think about environment and costs

git-kick, 2022-09-09:
Also ich würde nicht folgern, dass an _deiner_ E3DC etwas nicht stimmt. Ich vermute, dass das RSCP-Interface auf Seite E3DC nicht sauber programmiert ist und deshalb Temperatur- und Spannungswerte nicht klar unterscheidbar sind. Die Fehlerbeschreibung wäre in etwa "Tag DCB_CELL_TEMPERATURE liefert offensichtlich teilweise Spannungswerte (um 3,3) - nicht nur Temperaturwerte (um 20)"

Zu der `null` kann ich nicht wirklich etwas sagen, aber vielleicht ist da einfach kein Sensor - es hängt ja vom physischen Aufbau der Batterie ab, wie viele Zellen und zugehörige Sensoren da sind. Die E3DC kann natürlich nur Werte weitergeben, die das jeweilige Batteriemodul abliefert. (Wobei dann die nächste Frage wäre, wozu für `02` überhaupt etwas via RSCP übertragen wird ...)

Meine Batterie (10 kWh) hat übrigens zwei DCBs und für jedes davon bekomme ich 6 Temperaturen und 16 Spannungen, keine `null`. Zwei der sechs Spannungswerte kommen immer im `DCB_CELL_TEMPERATURE` Container von RSCP, der Rest passt.

Da ich selbst die Spannungs- und Temperaturwerte nicht verwende, war mir das bisher nicht So wichtig. Aber wenn du Info von E3DC bekommst, mit der ich den Adapter verbessern kann: immer her damit!

peterle-k, 2022-09-09:
ich hab mal eine Anfrage bei E3DC gemacht ... auch wenn deren RSCP Interface das nicht genau liefert hätten sie Handlungsbedarf. Verwirren tut mich das null - und ich hab e3dc gebeten zu prüfen ob alles ok ist -> können sie sicher mit Fernabfrage machen

Mir war das bis jetzt auch schnurz ... allerdings gibts ein neues youtube video vom Piepenbrink wo er die Einflussfaktoren für eine lange Batterielebensdauer beschreibt und die Zelltemperatur ist da ein wesentlicher Treiber. 2° bringen da Jahre. Jetzt ist mein Gedanke, dass ich bei steigender Temperatur und passenden Rahmenbedingungen die Ladeleistung runtersetze und somit die Temperatur senke ... vielleicht völliger humbug aber mal sehen

git-kick, 2022-09-11:
Ich reduziere auch meistens die Ladeleistung, aber aus einem anderen Grund: ich lade nur mit geringerer Leistung, so dass der Akku erst am Nachmittag voll wird um zu vermeiden, dass er z.B. schon um 12 Uhr voll ist und ich danach in die Abregelung gerate (Einspeisung = max. 70% der Peakleistung). Aber das Temperaturthema ist auch interessant!

Diesen Thread schließe ich damit erstmal...

closed 2022-09-11 by git-kick