git-kick/ioBroker.e3dc-rscp #117: Objekte e3dc-rscp.0.EP.PARAM_0. werden nicht aktualisiert.

ArnoD15, 2022-09-09:
**Describe the bug**  
Es werden alle EP Parameter unter e3dc-rscp.0.EP.PARAM_0. bei mir nicht aktualisiert.
Wenn die Instanz neu gestartet wird, werden diese einmal aktualisiert und anschließend nicht mehr.
Wenn ich die Notstromreserve beim E3DC ändere, wird der geänderte Wert nur nach einem Neustart der Instanz übernommen.
Auch wenn ich diese lösche, werden sie nicht neu erstellt, sondern nur wenn die Instanz neu gestartet wird.
[image]

Das Abfrageintervall bei allen EP Tag ist s
[image]

Hier das LOG 
[iobroker.2022-09-09.log](https://github.com/git-kick/ioBroker.e3dc-rscp/files/9538588/iobroker.2022-09-09.log)

Habe jetzt alles probiert und schaffe es nicht das diese aktualisiert werden. 

**Versions:**  
 - Adapter version: 1.0.7
 - JS-Controller version: 4.0.23
 - Node version: 16.17.0

git-kick, 2022-09-11:
@ArnoD15 , lange nichts gehört :-)

Ich denke ich weiß woran es liegt: in deinem log sehe ich keine Abfrage von `TAG_EP_REQ_EP_RESERVE`
Grund ist wohl, dass ich dieses Tag in der Abfrageintervall-Liste vergessen habe.
Verblüffend ist, dass es bei mir trotzdem einwandfrei funktioniert...

Könntest du mal von hier installieren:
https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23117

Wenn es dann auch bei dir funktioniert, ist der Fehler gefunden und ich baue das ins nächste Release ein.

ArnoD15, 2022-09-11:
War sehr mit meinem Script beschäftigt :-)

Habe die Version von Github installiert, aber das gleiche Problem.
Wenn ich die Instanz neu starte, wird der Wert aktualisiert.
Das komische ist, dass gleich nach dem Neustart der Instanz, die ersten 10 sek. auch funktioniert. 
Wenn ich gleich danach den Wert beim E3DC ändere, wird dieser aktualisiert, wenn ich dann wieder beim E3DC ändere, nicht mehr.
Hier noch mal das LOG mit zweimal ändern der Notstromreserve beim E3DC:
[iobroker.2022-09-11.log](https://github.com/git-kick/ioBroker.e3dc-rscp/files/9542720/iobroker.2022-09-11.log)

git-kick, 2022-09-12:
Okay, zu früh gefreut :-\\
Ich kann das Problem auch bei mir nachstellen - da war ich letztes Mal zu eilig gewesen.
Ich sehe, dass der EP-Request vom Adapter ans E3DC etwa alle 3-4 sec rausgeht.
Der hier relevante `TAG_EP_REQ_EP_RESERVE` ist aber nur etwa alle 7 sec dabei und auch nur 4- oder 5-mal nach dem Start der Adapterinstanz, danach ist Ruhe (d.h. kein solcher Request mehr und damit auch keine Aktualisierung der `PARAM`-Werte).

Das Ganze sieht recht merkwürdig aus, ich vermute irgendwas mit Timern oder Concurrency. Muss ich mir mal in Ruhe vornehmen, kann etwas dauern...

ArnoD15, 2022-09-12:
Mir ist vor einiger Zeit auch aufgefallen, dass die Temperaturwerte `e3dc-rscp.0.PVI.PVI_0.TEMPERATURE.00 bis 03` einfach einfrieren, also nicht mehr aktualisiert werden. Nach einem Neustart der Instanz war alles wieder normal. Danach ist das aber sporadisch immer wieder mal vorgekommen.

[image]

[image]

Eventuell ist das der gleiche Fehler.
Bei Problemen mit Timern oder Concurrency könnte eventuell Async \ await helfen, das ist aber jetzt nur ins blaue geraten, da ich dazu deine code zu wenig kenne.

git-kick, 2022-10-03:
So, endlich was Neues. Ich denke ich habe den Fehler gefunden. War kein Timingproblem, nur banale Programmierfehler beim Vergleichen der Tagnames. Kannst du bitte nochmal neu installieren von https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23117
**Wichtig:** nicht nur `Dateiupload`, sondern den Adapter komplett löschen und von der URL neu installieren.
Dann manuell die `Instanzeinstellungen: e3dc-rscp.0` eintragen (nicht aus einer json-Datei importieren).

Hintergrund: ich habe in io-package.json Array-Werte ergänzt. Diese werden nur bei einer echten Neuinstallation ausgelesen. Ein Import der Einstellungen aus einer älteren json-Datei löscht die neu hinzugekommenen Werte.

Bei mir wird jetzt `TAG_EP_REQ_EP_RESERVE` ordentlich ausgeführt, ausserdem lässt sich das Abfrageintervall einstellen.
(Ebenso fehlten die Abfrageintervalle für `TAG_BAT_REQ_INFO` und `TAG_WB_REQ_CONNECTED_DEVICES`, auch die habe ich ergänzt.)

Die `e3dc-rscp.0.PVI.PVI_0.TEMPERATURE.00 bis 03` Werte habe ich länger beobachtet, alles lief bei mir normal. 
Den von dir beschriebenen sporadischen Fehler lagere ich in den neuen Issue #120 aus.

ArnoD15, 2022-10-04:
Ok Super :-)
Werde ich dann heute Abend mal testen.

ArnoD15, 2022-10-04:
Gerade getestet und funktioniert jetzt.
Tolle Arbeit. Danke.

git-kick, 2022-10-15:
Kommt demnächst mit v1.0.8

closed 2022-10-15 by git-kick