git-kick/ioBroker.e3dc-rscp #119: Received data type error

otze75, 2022-09-13:
**Describe the bug**  
The adapter makes error messages for some tags. See screenshot. The rest works fine.

**Versions:**  
 - Adapter version: v1.0.7
 
 PlatformBetriebssystem:linux
Architektur:arm64
CPUs:4
Geschwindigkeit:1800 MHz
Modell:Cortex-A72
RAM:7.6 GB
System-Betriebszeit:25 T. 19:59:44
Node.js:v16.17.0
time:1663084236629
timeOffset:-120
Adapter-Anzahl:433
NPM:8.15.0
Datenträgergröße:29.0 GB
Freier Festplattenspeicher:18.5 GB
Betriebszeit:17 T. 09:52:32
Aktive Instanzen:22
Pfad:/opt/iobroker/
aktiv:true
_nodeCurrent:16.17.0
_nodeNewest:16.17.0
_nodeNewestNext:16.17.0
_npmCurrent:8.15.0
_npmNewest:8.15.0
_npmNewestNext:8.15.0

E3DC S10X Compact Baujahr 2022

[image]

git-kick, 2022-09-19:
@otze75, thanks for your report.
Looks like there is some mismatch between what number of batteries/DCBs the adapter expects vs. whats really ther in your S10X.
For further analysis, I need information about your BAT/DCB setup, and a debug log from instance start up to the messages shown above.

git-kick, 2022-09-26:
Thank you @icke-pp - I will look into that as soon as possible.

git-kick, 2022-10-31:
It looks like there goes something wrong detecting the number of DCBs.

@otze75 and @icke-pp, could you please send me a screenshot of your object tree showing the number of BATs and DCBs, like
[image]
(please add one picture for each BAT_x)

Btw, do you know what "DCB" stands for? "device control block"?

git-kick, 2022-11-01:
Thank you. Is it correct that you have in your object tree
- only 1 BAT, i.e BAT_0
- only 1 DCB under BAT_0, i.e. DCB_0

If so, then it is clear why the adapter throws warnings: having DCB_COUNT = 4 under BAT_0, the adapter requests info for DCB_0, DCB_1, DCB_2, DCB_3 - and fails for three of them. (At least, the adapter learns from failure and stops querying DCB_1..3)

So, perhaps I am misinterpreting DCB_COUNT?
I am curious about what you get from other tools!

git-kick, 2022-11-01:
Ok, ich mach in deutsch weiter, wenn ich den Screenshot sehe :-)

Also, das ist alles plausibel (1 BAT mit 4 DCB, DCB_COUNT=4) und analog zu meiner Installation (1 BAT mit 2 DCB, DCB_COUNT=2). Allerdings antwortet meine S10 für beide DBC_0 und DCB_1 mit Daten:

[image]

[image]

Ich kann nicht nachvollziehen, warum die S10X offenbar anders reagiert. Immerhin läuft der Adapter weiter (und vermeidet weitere Anfragen die scheitern), aber dir fehlen halt die Daten von DCB_1..3

Ich habe aktuell keine Idee, wie ich das angehen könnte, weil ich keine S10X zum testen habe und die RSCP-Dokumentation von Hager dürftig und veraltet ist (Tag-Liste von 2016).

closed 2022-11-05 by git-kick

otze75, 2022-12-31:
Vielen Dank für die Antwort. Ich wünsche dir einen guten Start ins neue Jahr.... BGGesendet von Outlook für Android

git-kick, 2023-01-06:
> Ich habe nun endlich eine offizielle Rückmeldung vom E3DC Support erhalten. Behebt das Problem zwar immer noch nicht, aber wieder etwas gelernt das die Batteriemodule zum Stand heute nicht einzeln ausgelesen werden können. Der Status ist also weiterhin "closed". Vielleicht wird das trotzdem von E3DC irgendwann mal gelöst. Ich bleibe dran.

Danke @icke-pp das ist gut wenn der Hersteller weiß, dass wir auf Informationen warten.
Und jetzt kann ich auch `DCB` richtig ausschreiben...