git-kick/ioBroker.e3dc-rscp #120: PVI_0.TEMPERATURE.00..03 frieren sporadisch ein

git-kick, 2022-10-03:
Mir ist vor einiger Zeit auch aufgefallen, dass die Temperaturwerte `e3dc-rscp.0.PVI.PVI_0.TEMPERATURE.00 bis 03` einfach einfrieren, also nicht mehr aktualisiert werden. Nach einem Neustart der Instanz war alles wieder normal. Danach ist das aber sporadisch immer wieder mal vorgekommen.

[image]

[image]

Eventuell ist das der gleiche Fehler.
Bei Problemen mit Timern oder Concurrency könnte eventuell Async \ await helfen, das ist aber jetzt nur ins blaue geraten, da ich dazu deine code zu wenig kenne.

_Originally posted by @ArnoD15 in https://github.com/git-kick/ioBroker.e3dc-rscp/issues/117#issuecomment-1244152880_

git-kick, 2022-10-03:
@ArnoD15, die `e3dc-rscp.0.PVI.PVI_0.TEMPERATURE.00 bis 03` Werte habe ich länger beobachtet, alles lief bei mir normal. Der #117  Fehler trifft hier nicht analog zu. 

Das Auftreten nach ein paar Stunden lenkt meinen Verdacht eher auf ein ioBroker-Problem (kein adapterinternes) - gibt es im `admin.0`-Log Einträge? Kannst du parallel mal Werte wie `system.adapter.e3dc-rscp.0.cpu` oder `system.adapter.e3dc-rscp.0.memRss` aufzeichnen, vielleicht gibt es einen Zusammenhang mit der Systemauslastung.

ArnoD15, 2022-10-04:
Uli ich habe den Grund dafür gefunden. In den letzten Tagen wurde meine Batterie öfter bis auf 0 kWh entleert und dabei konnte ich feststellen, das jedes mal wenn die Batterie leer war, der Wechselrichter in einen Standby-Modus wechselt um keine Strom zu verbrauchen.
In diesem Standby-Modus werden auch die Temperatur Werte nicht mehr aktualisiert. 
Konnte das dann auch nachstellen indem ich die Lade-/Entladeleistung der Batterie manuell auf 0W eingestellt habe, da passiert dann das gleiche.
Das ist somit kein Problem vom Adapter sondern von E3DC so gewollt :-)
Das Issue kann somit geschlossen werden.

git-kick, 2022-10-15:
OK danke @ArnoD15 - wieder interessante Erkenntnisse!

closed 2022-10-15 by git-kick