git-kick/ioBroker.e3dc-rscp #122: E3DC Firmware Update S10_2022_04

ArnoD15, 2022-10-25:
**Describe the bug**  
Hallo Uli,
bin mir nicht sicher, ob es am Adapter liegt oder E3DC was umgestellt hat.
Die Objekt ID e3dc-rscp.0.EMS.DERATE_AT_POWER_VALUE wird nicht mehr übertragen und ich vermute das es seit dem Update von E3DC ist.
In den Release Notes für Release-Versionen: P10_2022_04 unter 1.12 finde ich nur diesen Hinweis:
• RSCP-Tags um Netzfrequenz erweitert.
• RSCP-Tag für Kommunikationsstatus der Wallbox erstellt (gilt für alle Gerätetypen).

Kannst du dir das mal ansehen, ob vom Adapter was umgestellt werden muss.

git-kick, 2022-12-15:
Hallo Arno,

bei mir wird das Tag `DERATE_AT_POWER_VALUE` ordentlich gelesen, hat aber immer den Wert 0.
Der direkt proportionale `DERATE_AT_PERCENT_VALUE` dagegen zeigt einen sinnvollen Wert (70%).

Die Definition der beiden Tags in `RscpTags.json` ist praktisch gleich, sie sind beide vom Typ `Float32`.
Im Silly-Log steht, wie die beiden Werte empfangen werden:
`{TAG_EMS_DERATE_AT_PERCENT_VALUE(Float32) = 0.7}, {TAG_EMS_DERATE_AT_POWER_VALUE(Float32) = 0}`
Einen Fehler im Adapter halte ich deshalb für nahezu ausgeschlossen, es erscheint auch nirgendwo eine Fehlermeldung.

Ich kann nicht nachvollziehen ob `DERATE_AT_POWER_VALUE` jemals einen Wert größer 0 hatte.
Wenn ja, hat E3/DC wohl am Interface etwas geändert; dann haben wir das bekannte Dokumentationsproblem.

Ich setze den Issue-Status mal auf `waiting`, vielleicht gibt es noch weitere Hinweise oder Ideen.

ArnoD15, 2022-12-16:
Ok, Danke.
Dann hat E3DC hier was umgestellt oder eher verstellt. 
DERATE_AT_POWER_VALUE hat auf alle Fälle mal funktioniert.
Habe es bei mir im Script bereits geändert und berechnen den Wert über DERATE_AT_PERCENT_VALUE

git-kick, 2022-12-16:
Ok. Warten wir auf das nächste Firmware Update.
Immerhin gibt's einen einfachen Workaround.

closed 2022-12-16 by git-kick