git-kick/ioBroker.e3dc-rscp #129: Missing data for external inverter

christoph-spielmann, 2022-12-15:
The mainboard of my S10 E Pro was defect and replaced with a new one. Since then I do not get any values for my external E3/DC inverter. PVI_1 contains zero values only. Values were present here before the mainboard was replaced.

Also, the log shows some errors like:

`Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)`

Any ideas?

git-kick, 2022-12-15:
There is an open issue https://github.com/git-kick/ioBroker.e3dc-rscp/issues/126 with similar message.

If your problem is different, please post your PVI object subtree and a debug log starting with adapter startup.
Still, I cannot promise a solution since differences between E3/DC models and versions are hard to test&debug for me.

christoph-spielmann, 2022-12-16:
Log: 

`
2022-12-16 20:19:05.682 - info: e3dc-rscp.0 (17039) starting. Version 1.0.8 in /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v16.18.0, js-controller: 4.0.23
2022-12-16 20:19:05.834 - info: e3dc-rscp.0 (17039) Connection to E3/DC is established
2022-12-16 20:19:07.831 - warn: e3dc-rscp.0 (17039) TAG_EP_REQ_EP_RESERVE has no assigned polling interval - assuming 'M' - please check io-package.json
2022-12-16 20:19:07.834 - warn: e3dc-rscp.0 (17039) TAG_SYS_REQ_IS_SYSTEM_REBOOTING has no assigned polling interval - assuming 'M' - please check io-package.json
2022-12-16 20:19:13.105 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:13.306 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:13.503 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:20.002 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:20.115 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:20.229 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:27.027 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:27.156 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:27.275 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:34.008 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:34.144 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2022-12-16 20:19:34.255 - warn: e3dc-rscp.0 (17039) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
`

PVI-Subtree:

`
{
  "e3dc-rscp.0.PVI": {
    "type": "device",
    "common": {
      "name": "Photovoltaik-Wechselrichter",
      "role": "photovoltaic.inverter"
    },
    "native": {},
    "from": "system.adapter.e3dc-rscp.0",
    "user": "system.user.admin",
    "ts": 1648562211105,
    "_id": "e3dc-rscp.0.PVI",
    "acl": {
      "object": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    }
  },
  "e3dc-rscp.0.PVI.PVI_0": {
    "type": "channel",
    "common": {
      "role": "info.module"
    },
    "from": "system.adapter.e3dc-rscp.0",
    "user": "system.user.admin",
    "ts": 1671222125910,
    "_id": "e3dc-rscp.0.PVI.PVI_0",
    "acl": {
      "object": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    }
  },
  "e3dc-rscp.0.PVI.PVI_0.AC_MAX_APPARENTPOWER": {
    "type": "state",
    "common": {
      "name": "Maximale AC-Scheinleistung",
      "type": "number",
      "role": "value",
      "read": true,
      "write": false,
      "states": "",
      "unit": "VA"
    },
    "native": {},
    "from": "system.adapter.e3dc-rscp.0",
[... 5042 lines cut ...]
  "e3dc-rscp.0.PVI.PVI_2.VOLTAGE_MONITORING_THRESHOLD_TOP": {
    "type": "state",
    "common": {
      "name": "Spannungsüberwachung obere Schwelle",
      "type": "number",
      "role": "value",
      "read": true,
      "write": false,
      "states": null,
      "unit": "V"
    },
    "native": {},
    "from": "system.adapter.e3dc-rscp.0",
    "user": "system.user.admin",
    "ts": 1670318238329,
    "_id": "e3dc-rscp.0.PVI.PVI_2.VOLTAGE_MONITORING_THRESHOLD_TOP",
    "acl": {
      "object": 1636,
      "state": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    }
  }
}
`

christoph-spielmann, 2022-12-30:
Is there something else I can do?

git-kick, 2023-01-06:
Yes, please 
- stop the adapter
- delete the complete e3dc-rscp.0 object tree
- set the e3dc-rscp.0 instance from `info` to `debug` 
- start the adapter and wait 30-60 secs
- post the log as formatted file (with line breaks)
- post the object tree (PVI part) as screenshot

chrispi68, 2023-01-09:
[image]
[image]
[image]

chrispi68, 2023-01-09:
I created a new ioBroker vm/instance to create the log file and the screenshots. Hope this helps.

git-kick, 2023-01-09:
Thank you @chrispi68 for your Input.

From the debug log I can see that the warning 
`Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)`
probably has the same cause as in #126 - we assume that tag is not longer supported. From adapter version v1.0.9 one can set 'N' for never in the polling intervals list to exclude such unsupported tags. 

From the object tree I see that the adapter only zeros for `String_x` values for 'PVI_1' - from your report, I assume that more meaningful values are displayed for `PVI_0`, and they used to be o.k. also for `PVI_1` with the original mainboard in your S10 E Pro. (pls correct me if I'm wrong)

Background: as RSCP has no means to request the number of PVIs directly, the adapter sends probing requests for indexes 0, 1, 2, and builds up the object tree according to the (non-error) RSCP responses. 

In your case, the adapter got o.k.-responses für `PVI_0` and `PVI_1`, but no valid response for `PVI_2`. I assume this is correct as you still have two PVIs configured, right? Also, I can see that many other (at least the more static) data are coming in for `PVI_1`, i.e. only the values under `String_x` are missing.

The "bad" thing is: there is no hint in the debug log about any issue getting data for `PVI_1`. In other words: I am quite sure that the adapter "cleanly" receives the zero values you see in the `PVI_1` subtree. As long as I have no other evidence, I can do nothing about it in the adapter, but assume it is an issue introduced with the new mainboard. Please keep in mind that I can only test against a S10E device and there may be (possibly undocumented) differences between S10 models and versions. 

Maybe a request to E3/DC (Hager) brings some clarification, like [here](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/119#issuecomment-1368218756)

chrispi68, 2023-01-09:
> From the object tree I see that the adapter shows only zeros for `String_x` values for 'PVI_1' - from your report, I assume that more meaningful values are displayed for `PVI_0`, and they used to be o.k. also for `PVI_1` with the original mainboard in your S10 E Pro. (pls correct me if I'm wrong)

Correct.

> In your case, the adapter got o.k.-responses für `PVI_0` and `PVI_1`, but no valid response for `PVI_2`. I assume this is correct as you still have two PVIs configured, right? 

Yes, I own one S10 Pro and a external inverter.

> Please keep in mind that I can only test against a S10E device and there may be (possibly undocumented) differences between S10 models and versions.

Would it help if a temporarily open the port on my firewall so you can access my S10 Pro?

git-kick, 2023-01-09:
> Would it help if a temporarily open the port on my firewall so you can access my S10 Pro?

Not as long as I have no evidence about something going wrong _in the adapter_. (It would certainly help to test & debug a fix.)
Sorry.

git-kick, 2023-02-18:
Closed assuming this is not an adapter issue.

closed 2023-02-18 by git-kick