git-kick/ioBroker.e3dc-rscp #137: Funktionserweiterung: Schlüsselschalter/Schalter im Adapter (Sunmode/Mixedmode/forbidden/permittet

Lucifor1976, 2023-01-22:
Am Hauskraftwerk kann die Funktion des Schlüsselschalters der EasyConnect eingestellt werden (Laden PV 0/Laden Netz und PV 1 oder Laden gesperrt 0 / Laden erlaubt 1).

Könnt ihr diese Funktion mit einbauen? Derzeitig müsste man immer in den Keller rennen, die Wallbox einschalten oder aus, bzw im Menü des Speichers die Funktion umstellen.

Da viele die Wallbox vermieten (zumindest ich) wäre es toll wenn man die Wallbox vom ioBroker aus steuern könnte ohne ständig in den Keller rennen zu müssen um das Laden zu erlauben bzw. zu verbieten.

Herzlichen Dank dafür.

git-kick, 2023-02-03:
Da ich keine Wallbox zum Testen habe, müsste @ka-vaNu sich das ansehen.

ka-vaNu, 2023-02-05:
Ich kenne kein RSCP-Tag mit dem diese Funktion möglich wäre. Ich habe die Funktion vielleicht auch noch nicht ganz verstanden. Bei mir kann ich nur die Funktion des Schlüsselschalters umschalten. Also "Sunmode/Mixedmode" oder "Laden erlaubt/gesperrt". Ein Sperren der Wallbox per Software ist mir nicht bekannt. Kannst du mal ein Foto von der Funktion im Steuerungsmenü hochladen.

git-kick, 2023-05-26:
(keine Aktivität.)