git-kick/ioBroker.e3dc-rscp #139: ERROR wb.queueWbRequestData is not a function

ArnoD15, 2023-01-23:
Hallo Uli,

erstmal super, dass ihr auch die Wallbox eingebunden habt.
Habe heute die neue Version 1.1.0 installiert und bekomme einen Fehler, wenn ich in den Adaptereinstellungen die WB anwähle.
Ohne WB funktioniert der Adapter, soweit ich bis jetzt testen konnte.
Mit Wallbox kommt im LOG beim Starten der Instanz folgender Fehler und die Instanz bleibt rot.
Ich habe den Adapter bereits komplett deinstalliert und neu installiert, jedoch ohne Erfolg.

--
2023-01-23 18:59:03.254  - info: host.iobroker "system.adapter.e3dc-rscp.0" enabled
2023-01-23 18:59:03.498  - info: host.iobroker instance system.adapter.e3dc-rscp.0 started with pid 5853
2023-01-23 18:59:04.870  - debug: e3dc-rscp.0 (5853) Redis Objects: Use Redis connection: 0.0.0.0:9001
2023-01-23 18:59:04.920  - debug: e3dc-rscp.0 (5853) Objects client ready ... initialize now
2023-01-23 18:59:04.923  - debug: e3dc-rscp.0 (5853) Objects create System PubSub Client
2023-01-23 18:59:04.924  - debug: e3dc-rscp.0 (5853) Objects create User PubSub Client
2023-01-23 18:59:05.002  - debug: e3dc-rscp.0 (5853) Objects client initialize lua scripts
2023-01-23 18:59:05.068  - debug: e3dc-rscp.0 (5853) Objects connected to redis: 0.0.0.0:9001
2023-01-23 18:59:05.110  - debug: e3dc-rscp.0 (5853) Redis States: Use Redis connection: 0.0.0.0:9000
2023-01-23 18:59:05.322  - debug: e3dc-rscp.0 (5853) States create System PubSub Client
2023-01-23 18:59:05.324  - debug: e3dc-rscp.0 (5853) States create User PubSub Client
2023-01-23 18:59:05.634  - debug: e3dc-rscp.0 (5853) States connected to redis: 0.0.0.0:9000
2023-01-23 18:59:08.285  - info:  e3dc-rscp.0 (5853) starting. Version 1.1.0 in  /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v16.18.1,  js-controller: 4.0.24
2023-01-23 18:59:08.386  - debug: e3dc-rscp.0 (5853) config.*: (xxx.xxx.xxx.xxx, 5033, xxxxxx.xxxxxx@xxxxxxxxx.xx, 3, 5, 12, 0)
2023-01-23 18:59:08.411  - debug: e3dc-rscp.0 (5853) Probing for BAT units - 0..1.
2023-01-23 18:59:08.413  - debug: e3dc-rscp.0 (5853) Probing for PVI units - 0..2.
2023-01-23 18:59:08.423  - error: e3dc-rscp.0 (5853) uncaught exception: wb.queueWbRequestData is not a function
2023-01-23 18:59:08.424  - error: e3dc-rscp.0 (5853) TypeError: wb.queueWbRequestData is not a function
at E3dcRscp.requestAllData (/opt/iobroker/node_modules/iobroker.e3dc-rscp/main.js:1153:34)
at Timeout._onTimeout (/opt/iobroker/node_modules/iobroker.e3dc-rscp/main.js:563:10)
at listOnTimeout (node:internal/timers:559:17)
at processTimers (node:internal/timers:502:7)
2023-01-23 18:59:08.425  - error: e3dc-rscp.0 (5853) wb.queueWbRequestData is not a function
2023-01-23 18:59:08.428  - info: e3dc-rscp.0 (5853) Connection to E3/DC is established
2023-01-23 18:59:08.429  - debug: e3dc-rscp.0 (5853) Sending request TAG_RSCP_REQ_AUTHENTICATION
2023-01-23 18:59:08.451  - info: e3dc-rscp.0 (5853) terminating
2023-01-23 18:59:08.452  - warn: e3dc-rscp.0 (5853) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-01-23 18:59:08.456  - warn: e3dc-rscp.0 (5853) E3/DC connection closed
2023-01-23 18:59:08.458  - info: e3dc-rscp.0 (5853) Stop communication with E3/DC and pause a minute before retry ...
2023-01-23 18:59:09.047  - error: host.iobroker instance system.adapter.e3dc-rscp.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-01-23 18:59:09.047  - info: host.iobroker Restart adapter system.adapter.e3dc-rscp.0 because enabled
2023-01-23 18:59:14.940  - info: host.iobroker "system.adapter.e3dc-rscp.0" disabled

**Versions:**  
 - Adapter version: 1.1.0
 - JS-Controller version: 4.0.24
 - Node version: 16.18.1
 - Operating system: Docker / Linux

git-kick, 2023-01-28:
Hallo Arno,
die bei dir als fehlend gemeldete Funktion `wb.queueWbRequestData` ist in der Datei `wallbox.js` definiert, welche im selben Verzeichnis liegt wie `main.js`. Ist das bei dir so gegeben?
Wenn ja, wie genau hast du den Adapter installiert?

ArnoD15, 2023-01-29:
Hallo Uli,
ja die Datei wallbox.js ist im Verzeichnis vorhanden.
[image]

Erste Installation über die Adapter Update Funktion von IoBroker, als dann der Fehler im LOG war, habe ich den Adapter e3dc-rscp gelöscht und dann neu installiert. 
Werde noch mal alles löschen und dann von Github installieren, mal sehen, ob das was hilft.

ka-vaNu, 2023-02-04:
Aus Sicht der Wallbox ist die Anwahl der DB nicht erforderlich. Ich fixe es.

ka-vaNu, 2023-02-04:
Hab den Buchstabe ausgetauscht und unter https://github.com/git-kick/ioBroker.e3dc-rscp/pull/141 alls PR gestellt.

ArnoD15, 2023-02-06:
ka-vaNu super Danke. :-)

git-kick, 2023-02-18:
Den Fehler hatte ich per cop/paste eingebaut - danke @ka-vaNu für den Fix.
v1.1.1 ist im Release-Prozess.

closed 2023-02-18 by git-kick