git-kick/ioBroker.e3dc-rscp #140: WB/Control/PowerLimitation switch back to previous value

kkeintzel, 2023-01-31:
If i write into the object "e3dc-rscp.0.WB.WB_0.Control.PowerLimitation" a value (e.g. 6, 10 or 16).
it will be switch back to the previous value, but it will be set correctly to the E3/DC-S10.

config:
iobroker (beta channel) up to date
e3dc-rscp: v1.1.0

https://user-images.githubusercontent.com/83504161/215748889-41eeee56-947a-41ea-ae0c-f9b2304cb39a.mp4

thx

j64bit, 2023-01-31:
I find the same, also for the other control elements like SunMode.

git-kick, 2023-02-03:
As I have no wallbox for testing, I assign this Issue to @ka-vaNu

kkeintzel, 2023-02-04:
Following interesting stuff I found:
if you send the value 2times (like a doubleclick) e.g. via VIS-Button >> then it works!
[image]

ka-vaNu, 2023-02-04:
OK, i'll take it, but i've got insufficient rights to assign it myself

ka-vaNu, 2023-02-04:
It looks like a timing problem. My S10E with WB Easy Connect seems to send the EXTERN_DATA_ALG packets in a different order. The value in e3dc-rscp.0.WB.WB_0.EXTERN_DATA_ALG.PowerLimitation shows the value from the wallbox/S10E, the value in e3dc-rscp.0.WB.WB_0.Control.PowerLimitation shows the value set by the user. If the value is set via the app, the value in e3dc-rscp.0.WB.WB_0.Control.PowerLimitation is also set. I can disable this, but then it is no longer synchronous in case of an external update.

What do you think, how should the control-object behave?

ka-vaNu, 2023-02-04:
Maybe I should merge the display object and the control object and leave the confirmation to the RSCP packet.

git-kick, 2023-02-05:
There are similar cases where we have one RSCP tag for setting a value and another tag for reading the value (which should be set), e.g. `EMS.SET_POWER_VALUE`and `EMS.SET_POWER`.
As the adapter follows the "no processing, just transmit" policy, **I would prefer to keep both tags visible in the object tree**.
Otherwise, we will hide all those timing issues beneath the surface and we will have a hard time reporting and solving them.

ka-vaNu, 2023-03-12:
Fix via  PR #144, please test and review. The RSCP-Response does not longer update the Object e3dc-rscp.0.WB.WB_0.Control.PowerLimitation and e3dc-rscp.0.WB.WB_0.Control.PowerLimitation. So the timing-issue caused by different versions is fixed.

closed 2023-03-23 by git-kick

kkeintzel, 2023-04-16:
Many Thx for your perfect job....

PR #144 (2023-03-23): Control.* no longer updated by rscp response
Fix #140, Control.* no longer updated by rscp response