git-kick/ioBroker.e3dc-rscp #142: Funktion: Manuelle Speicherladung

Borilu, 2023-02-05:
Ist es möglich eine manuelle Speicherladung anzustoßen? Also x Wh über das Netz in die Batterie zu laden?
Vielen Dank für die Hilfe.

git-kick, 2023-02-05:
[image]
An dem Schloß-Symbol links und an der Rolle "indicator" bzw. "value" kann man erkennen, dass dies beides read-only-tags sind (wären sie schreibbar, hätten sie die Rolle "switch" bzw. "level").

Wie man die manuelle Speicherladung anstößt, kann ich aus der Tagliste nicht erkennen; evtl. hat das `SET_POWER_MODE` Tag etwas damit zu tun, aber das ist reine Spekulation.

Borilu, 2023-02-06:
Kann ich bestätigen.
[image]
Grid_Charge startet die Speicherladung aus dem Netz. Danke

git-kick, 2023-02-06:
Sehr gut, dass es klappt.

closed 2023-02-06 by git-kick