git-kick/ioBroker.e3dc-rscp #143: Several Warnings

Draygv2, 2023-02-28:
Every interval I get this errors
`
<html>
<body>

e3dc-rscp.0 | 2023-02-28 16:55:40.044 | warn | Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
-- | -- | -- | --
e3dc-rscp.0 | 2023-02-28 16:55:40.004 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-02-28 16:55:40.004 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-02-28 16:55:40.004 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-02-28 16:55:40.004 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-02-28 16:55:40.003 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-02-28 16:55:40.002 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-02-28 16:55:40.002 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)

</body>
</html>
`

**Versions:**  
 - Adapter version: <1.1.1>
 - JS-Controller version: <16.19.1> 
 - Node version: <6.19.1> 
 - Operating system: <ubuntu>

git-kick, 2023-03-02:
Hi, please supply
1. your E3/DC configuration (Model, how many BATs/DCBs)
2. a debug log (it will be long, so please attach as file, do not paste as text)

Draygv2, 2023-03-11:
Hi sorry for my late answer. I was in holiday.

I have an S10X System with 7 DCBs. At the moment I have just one error. The might be because of the restart for the debug log
[E3DC.txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/10948712/E3DC.txt)

git-kick, 2023-03-12:
I cannot see any error in the log you attached.
Warning `RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)` just says that your device does not support this tag (other E3/DC models do). This is not an error, just ignore this tag.

If you want to provide a debug log containing the `ERROR: RSCP_ERR_OUT_OF_BOUNDS (5)`error, please make sure the log contains everything from _adapter start_.

Draygv2, 2023-03-12:
Sorry I mean warning not error. 
Okey I changed the interval to never. I guess that's what the N stands for.
I changed also all other warning from the beginning. 
Maybe now its done like this. Thanks so far.

jh537, 2023-03-19:
I have the same problem s10-x compact 14 with batteries of cn
`

javascript.0 | 2023-03-19 05:04:24.008 | info | script.js.common.Charge_Control: ##{"from":"Charge-Control", "message":" -==== Laderegelung wird gestoppt ====- "}##
-- | -- | -- | --
javascript.0 | 2023-03-19 05:04:21.009 | info | script.js.common.Charge_Control: ##{"from":"Charge-Control", "message":" -==== Laderegelung wird gestoppt ====- "}##
e3dc-rscp.0 | 2023-03-19 05:04:18.512 | warn | Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
e3dc-rscp.0 | 2023-03-19 05:04:18.494 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-03-19 05:04:18.494 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-03-19 05:04:18.494 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-03-19 05:04:18.494 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-03-19 05:04:18.494 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-03-19 05:04:18.494 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-03-19 05:04:18.494 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_INFO (0x3800042)
e3dc-rscp.0 | 2023-03-19 05:04:18.493 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-03-19 05:04:18.493 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)
e3dc-rscp.0 | 2023-03-19 05:04:18.493 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_VOLTAGES (0x380001a)
e3dc-rscp.0 | 2023-03-19 05:04:18.492 | warn | Received data type ERROR: RSCP_ERR_OUT_OF_BOUNDS (5) - tag TAG_BAT_DCB_ALL_CELL_TEMPERATURES (0x3800018)

`

git-kick, 2023-03-20:
> I have the same problem s10-x compact 14 with batteries of cn `

See https://github.com/git-kick/ioBroker.e3dc-rscp/issues/143#issuecomment-1465158505 preceding your post. It is a good idea to read the thread before posting.
If you decide to post a debug log, please attach a file (not inline text).

git-kick, 2023-03-20:
continued in #145

closed 2023-03-20 by git-kick