git-kick/ioBroker.e3dc-rscp #145: Battery data incorrect or incomplete

jh537, 2023-03-19:
**Describe the bug**  
Battery data incorrect or incomplete  

4x95V Module cn on e3dc s10x compact 14 with ca. 12000 Wh

**To Reproduce**  

**Expected behavior**  

**Screenshots & Logfiles**  

[image]
[image]

**Versions:**  
 - Adapter version: v1.1.1
 - JS-Controller version: 4.0.24 
 - Node version: v16.19.1 
 - Operating system: debian 11 x64

**Additional context**  
Add any other context about the problem here.

git-kick, 2023-03-20:
(Ich schreibe deutsch, weil der Screenshot in deutsch ist. Bitte melden, wenn englisch besser ist.)

Ich denke das ist der Fehler, der zu https://github.com/git-kick/ioBroker.e3dc-rscp/issues/143 gehört: aus irgendeinem Grund bekommt der Adapter von der _S10X compact_ keine Werte für `BAT_DCB_INFO` und `BAT_DCB_ALL_CELL_{VOLTAGES,TEMPERATURES}`.

Um das Problem weiter verfolgen zu können, brauche ich ein debug log ab Adapter-Start - bitte als Dateianhang, nicht inline (ist sehr groß).

jh537, 2023-03-20:
Ok gern in D. Wie bekommt man das debug-log am besten?

jh537, 2023-03-20:
[log.txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/11016768/log.txt)

git-kick, 2023-03-20:
Ich sehe den Start nicht.
1. Instanz stoppen
2. log löschen
3. Instanz im debug Modus starten
4. etwa 3 Minuten laufen lassen
5. log speichern

jh537, 2023-03-20:
[log.txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/11017202/log.txt)

jh537, 2023-03-21:
klar gern
[log.txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/11024711/log.txt)

jh537, 2023-03-21:
Dort ist aber das Portal-Passwort hinterlegt.... 
`TAG_RSCP_AUTHENTICATION_USER - type: 0x0D - CString - length: 
TAG_RSCP_AUTHENTICATION_PASSWORD - type: 0x0D - CString - length: `

Außerdem ist mir noch eingefallen: Im Portal konnte auch nie die Batterietemperatur angezeigt werden. Andere User haben das Problem auch. Ich vermute, dass die bei E3DC sebst erst mal Anpassungen wegen neuem Batterietyp durchführen müssen. Vielleicht kann es deshalb nicht funktionieren.

git-kick, 2023-03-21:
Ja, das Passwort am besten rauslöschen.
Es ist aber "nur" das für den lokalen RSCP-Login, für den man im lokalen LAN sein muss.
(Ich mach mir eine Notiz, dass ich das mal ändere und das Passwort nicht mehr ausgegeben wird.) 

Dass die fehlenden Daten von der E3/DC Anlage (modellabhängig) nicht gemeldet werden, kann gut sein. In diesem Fall kann ich natürlich nichts machen. Aber ich schau mir das log noch genauer an - kann ja auch sein, dass der Adapter noch Fehler hat.
(Dauert noch ein paar Tage, bin gerade anderweitig beschäftigt.)

jh537, 2023-03-22:
> lEs ist aber "nur" das für den lokalen RSCP-Login, für den man im lokalen LAN sein muss.

Dann hätte ich da das falsche Passwort. Ich hatte das die LoginDaten vom E3DC-Portal. Das stellt man da in der Instanz ein. Irgendwie scheint, da etwas nicht korrekt zu sein?

jh537, 2023-03-22:
Ich habe mal die PW getauscht, aber da läuft die Instanz nur mit Warnung.

git-kick, 2023-03-23:
> > lEs ist aber "nur" das für den lokalen RSCP-Login, für den man im lokalen LAN sein muss.
> 
> Dann hätte ich da das falsche Passwort. Ich hatte da die LoginDaten vom E3DC-Portal. Das stellt man in der Instanz ein. Irgendwie scheint, da etwas nicht korrekt zu sein?

Du hast recht, es ist das Portal-Passwort, das aktuell noch im silly-log steht. Das zweite, lokale Passwort (im GUI `E3/DC RSCP Passwort`) dient der Verschlüsselung der Netzwerkverbindung und wird nicht in einem Tag übergeben, deshalb erscheint es nicht im silly-log. Da hat mich der Name des Portal-Passwort-Tags (`TAG_RSCP_AUTHENTICATION_PASSWORD`) in die Irre geführt.

jh537, 2023-03-23:
wenn es weiter nichts ist :-)

git-kick, 2023-03-23:
> klar gern [log.txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/11024759/log.txt)

Im silly-log sehe ich folgendes:

Bei der Initialisierung meldet die Anlage (korrekt) die 4 DCBs: 
```
2023-03-20 17:11:23.508 - silly: e3dc-rscp.0 (41460) TAG_BAT_DATA
DCB_COUNT - type: 0x03 - UChar8 - length: 1 value: 0x04
```

Der Adapter sendet etwas später Anfragen für alle 4 DCBs:
```
2023-03-20 17:11:26.425 - debug: e3dc-rscp.0 (41460) Sending request TAG_BAT_REQ_DATA
TAG_BAT_REQ_DCB_ALL_CELL_TEMPERATURES - type: 0x05 - UInt16 - length: 2 value: 0
TAG_BAT_REQ_DCB_ALL_CELL_VOLTAGES - type: 0x05 - UInt16 - length: 2 value: 0
TAG_BAT_REQ_DCB_INFO - type: 0x05 - UInt16 - length: 2 value: 0
TAG_BAT_REQ_DCB_ALL_CELL_TEMPERATURES - type: 0x05 - UInt16 - length: 2 value: 1
TAG_BAT_REQ_DCB_ALL_CELL_VOLTAGES - type: 0x05 - UInt16 - length: 2 value: 1
TAG_BAT_REQ_DCB_INFO - type: 0x05 - UInt16 - length: 2 value: 1
TAG_BAT_REQ_DCB_ALL_CELL_TEMPERATURES - type: 0x05 - UInt16 - length: 2 value: 2
TAG_BAT_REQ_DCB_ALL_CELL_VOLTAGES - type: 0x05 - UInt16 - length: 2 value: 2
TAG_BAT_REQ_DCB_INFO - type: 0x05 - UInt16 - length: 2 value: 2
TAG_BAT_REQ_DCB_ALL_CELL_TEMPERATURES - type: 0x05 - UInt16 - length: 2 value: 3
TAG_BAT_REQ_DCB_ALL_CELL_VOLTAGES - type: 0x05 - UInt16 - length: 2 value: 3
TAG_BAT_REQ_DCB_INFO - type: 0x05 - UInt16 - length: 2 value: 3

```
Aber in der Antwort kommen nur Fehlercodes, als DCB_INDEX kommt nur 0 vor (in der fünften Zeile), die anderen (1,2,3) gar nicht:
```
2023-03-20 17:11:30.510 - silly: e3dc-rscp.0 (41460) TAG_BAT_DATA
TAG_BAT_DCB_ALL_CELL_TEMPERATURES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_ALL_CELL_VOLTAGES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_INFO - type: 0x0E - Container - length: 48
TAG_BAT_DCB_INDEX - type: 0x06 - Int32 - length: 4 value: 0
TAG_BAT_DCB_ALL_CELL_TEMPERATURES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_ALL_CELL_VOLTAGES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_INFO - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_ALL_CELL_TEMPERATURES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_ALL_CELL_VOLTAGES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_INFO - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_ALL_CELL_TEMPERATURES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_ALL_CELL_VOLTAGES - type: 0xFF - Error - length: 4 value: 5
TAG_BAT_DCB_INFO - type: 0xFF - Error - length: 4 value: 5
```

Wohlgemerkt, der Adapter empfängt lauter i.O. Datenpakete an der RSCP-Schnittstelle und kann diese auch entpacken, aber die andere Seite liefert halt nicht die gewünschten Inhalte.

Wie du schon vermutet hast, ist das ein Thema der Anlagen-Firmware. Schade, dass sich anscheinend jede Anlage etwas anders verhält bzw. nicht alle Daten liefert.

jh537, 2023-03-24:
haben die bei e3dc auch irgendwas zum nachvollziehen? ich hatte da schon mal etwas als fehler gemeldet aber nie eine anwort bekommen :( da könnte man das problem ja mit etwas nachdruck schieben. die haben einen anderen fehler jetzt im system-log kurzfristig beseitigt aber die temp wird immer noch nicht angezeigt

jh537, 2023-03-24:
Ich habe mal ein Ticket geöffnet, mit der Hoffnung, dass dieses Mal eine Antwort kommt. 
[image]

jh537, 2023-03-24:
Ich habe  mal ein Ticket geöffnet, in der Hoffnung, dass dieses Mal eine Antwort/Lösung kommt.

[image]
[image]

jh537, 2023-03-24:
ok das ging dieses mal sehr schnell
[image]

git-kick, 2023-04-23:
Nicht seitens Adapter zu lösen.

closed 2023-04-23 by git-kick