git-kick/ioBroker.e3dc-rscp #148: .PVI_0.String_0.DC_POWER and .PVI_0.String_1.DC_POWER not updating and not in setup list

smartboart, 2023-04-14:
**Describe the bug**  

Hello, PVI_0.String_0 and 1 is missing in the Tag List. Only TAG_PVI_REQ_DC_STRING_ENERGY_ALL is present.
So the states for String 0 and 1 not getting updated ...

see pictures..

[image]
[image]

**Versions:**  
 - Adapter version: 1.1.1
 - JS-Controller version: 4.0.24
 - Node version: v18.15.0
 - Operating system:  debian buster / RPI4 8GB 
 - NPM: 9.5.0

**Additional context**  
I need the states for String 1 and 2 stable and constantly polling for my vis.  ...It would be great to habhave this states in the setup tag list...thanks..
The firmware Version of the E3DC would also be great to have. TheModbus Adapter is providing this...

git-kick, 2023-04-14:
Hi and thanks for your report.

> The firmware Version of the E3DC would also be great to have.
This is content of tbe INFO namespace, which is not yet supported. I created #149 for this.

smartboart, 2023-04-14:
Hello, here is the log file...
I need to inform you, that in this log everything went fine. I have watched this behavior 2 times. A restart always fixed the issue.
My E3DC-S10EPro is from 2020. 
[RSCP_silly_log.txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/11232041/RSCP_silly_log.txt)

git-kick, 2023-04-14:
Okay... then it is not worth inspecting the log now.

I find it hard to estimate how large a silly-log will grow untli the error occurs. Perhaps it is better to do something like
1. start normal operation (info log)
2. wait until error occurs
3. delete log
4. set instance to "silly" _without restart_
5. wait some time until error reoccurs (see Polling Interval setting)
6. save the log...

ArnoD15, 2023-04-15:
Hallo Smartboart,
kann es sein, dass es in der Zeit passiert, wo dein Wechselrichter in den Standby-Modus wechselt, um Strom zu sparen?
Mir ist das bei den Temperatur Werten vom WR aufgefallen, dass dann keine Werte mehr vom E3DC übertragen werden.
Habe es jetzt bei mir mal geprüft und in dem gleichen Zeitraum werden auch die Volt, Watt Werte er einzelnen Strings nicht mehr übertragen und auch keine Watt werte der drei Phasen.
Wenn der WR sich wieder einschaltet, funktioniert alles wieder ohne Neustart vom Adapter.
Komisch ist nur, dass es über den Modbus Adapter ja funktioniert.
[image]

[image]

smartboart, 2023-04-15:
Hi Arno...Könnte gut sein...habe mal ein Chart zusammen gebaut mit den Historie Daten...
[image]

smartboart, 2023-04-18:
Heute wieder keine Aktualisierung der PV String Weerte....siehe Bild.leider war der log nicht auf silly, habe aber im laufenden  Betrieb umgestellt ohne Adapter neustart. Der log ist angehängt.
[RSCP_silly_log_ohne Neustart.txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/11260317/RSCP_silly_log_ohne.Neustart.txt)

[image]

smartboart, 2023-04-18:
irgend wann um 12:30 rum wurden dann wieder Daten für String Leistung übergeben.
[image]

ArnoD15, 2023-04-18:
Hatte heute um 6:00 Uhr noch eine Fehlermeldung im LOG, leider hatte ich die LOG Stufe wieder auf Info gestellt. 
Eventuell hilft dir die Fehlermeldung weiter:

2023-04-18 06:26:20.925  - warn: e3dc-rscp.0 (5462) Received message with inconsistent length: 5792 vs 6515
2023-04-18 06:26:20.950  - error: e3dc-rscp.0 (5462) uncaught exception: The value of "offset" is out of range. It must be >= 0 and <= 5788. Received 6511
2023-04-18 06:26:20.952  - error:  e3dc-rscp.0 (5462) RangeError [ERR_OUT_OF_RANGE]: The value of "offset"  is out of range. It must be >= 0 and <= 5788. Received 6511
at new NodeError (node:internal/errors:399:5)
at boundsError (node:internal/buffer:88:9)
at Buffer.readInt32LE (node:internal/buffer:392:5)
at E3dcRscp.processFrame (/opt/iobroker/node_modules/iobroker.e3dc-rscp/main.js:1338:74)
at Socket. (/opt/iobroker/node_modules/iobroker.e3dc-rscp/main.js:522:10)
at Socket.emit (node:events:513:28)
at addChunk (node:internal/streams/readable:324:12)
at readableAddChunk (node:internal/streams/readable:297:9)
at Readable.push (node:internal/streams/readable:234:10)
at TCP.onStreamRead (node:internal/stream_base_commons:190:23)
2023-04-18 06:26:20.952  - error:  e3dc-rscp.0 (5462) Exception-Code: ERR_OUT_OF_RANGE: The value of  "offset" is out of range. It must be >= 0 and <= 5788. Received  6511
2023-04-18 06:26:21.035  - warn: e3dc-rscp.0 (5462) Received message with invalid MAGIC: >2BD1<
2023-04-18 06:26:21.036  - warn: e3dc-rscp.0 (5462) Received message with invalid CTRL: >69A8<
2023-04-18 06:26:21.036  - warn: e3dc-rscp.0 (5462) Received message with inconsistent length: 6528 vs 21841
2023-04-18 06:26:21.040  - warn: e3dc-rscp.0 (5462) Unknown tag: tagCode=0xa9a66a4e, len=30773, typeCode=0x96
2023-04-18 06:26:21.044  - info: e3dc-rscp.0 (5462) terminating
2023-04-18 06:26:21.045  - warn: e3dc-rscp.0 (5462) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-04-18 06:26:21.054  - warn: e3dc-rscp.0 (5462) E3/DC connection closed
2023-04-18 06:26:21.054  - info: e3dc-rscp.0 (5462) Stop communication with E3/DC and pause a minute before retry ...
2023-04-18 06:26:21.642  - error: host.iobroker instance system.adapter.e3dc-rscp.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-04-18 06:26:21.643  - info: host.iobroker Restart adapter system.adapter.e3dc-rscp.0 because enabled
2023-04-18 06:26:51.766  - info: host.iobroker instance system.adapter.e3dc-rscp.0 started with pid 16441
2023-04-18 06:26:56.505  - info:  e3dc-rscp.0 (16441) starting. Version 1.2.0 (non-npm:  git-kick/ioBroker.e3dc-rscp#4f870640e4806923fa86221ca129283714f9eb98) in  /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v18.15.0,  js-controller: 4.0.24
2023-04-18 06:26:56.673  - info: e3dc-rscp.0 (16441) Connection to E3/DC is established
2023-04-18 06:26:57.233  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:04.315  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:11.297  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:18.259  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:25.220  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd

git-kick, 2023-04-23:
So wie ich das sehe, liegt das Problem höchstwahrscheinlich nicht im Adapter, sondern auf der S10-Seite.

closed 2023-04-23 by git-kick

git-kick, 2023-05-08:
Das Thema wird in #160 weiter bearbeitet.