git-kick/ioBroker.e3dc-rscp #149: Feature request: support INFO namespace

git-kick, 2023-04-14:
(created from #148)

Include tags of the INFO namespace in the adapter.
This will also show the E3/DC firmare version, as requested by @smatboart in #148

git-kick, 2023-04-17:
Vorabversion ist im master branch.
Wer Lust hat, kann ja schon mal probieren :)

ArnoD15, 2023-04-17:
Dann bin ich mal so mutig und teste die Version. :-)
Muss der Adapter vorher deinstalliert werden?

git-kick, 2023-04-17:
Eigentlich sollte es reichen, die Instanz zu löschen (damit die Einstellungen weg sind). Aber sicher bin ich mir nicht, da fehlt mir noch ioBroker-Knowhow...
Klares Zeichen ist, dass in den Einstellungen jetzt eine Checkbox für INFO sein muss.

ArnoD15, 2023-04-17:
Ok, habe es jetzt ohne Deinstallation einfach über Github neu installiert und es funktioniert. :-)
Bei den Abfrageintervallen sind aber keine Tags mit Info, ist das richtig?

ArnoD15, 2023-04-17:
Ok, jetzt sieht es besser aus. Es kommt nur noch eine Warnung die ständig wiederholt wird:
e3dc-rscp.0 | 2023-04-17 18:10:44.409 | warn | Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd

git-kick, 2023-04-17:
Ok danke, das invalid tag: 0xA800036 schau ich mir bei Gelegenheit noch an, vielleicht steht da ja etwas sinnvolles drin, obwohl die tag id bisher nicht dokumentiert ist.

ArnoD15, 2023-04-18:
Hatte heute um 6:00 Uhr noch eine Fehlermeldung im LOG, leider hatte ich die LOG Stufe wieder auf Info gestellt. 
Eventuell hilft dir die Fehlermeldung weiter:

2023-04-18 06:26:20.925  - warn: e3dc-rscp.0 (5462) Received message with inconsistent length: 5792 vs 6515
2023-04-18 06:26:20.950  - error: e3dc-rscp.0 (5462) uncaught exception: The value of "offset" is out of range. It must be >= 0 and <= 5788. Received 6511
2023-04-18 06:26:20.952  - error:  e3dc-rscp.0 (5462) RangeError [ERR_OUT_OF_RANGE]: The value of "offset"  is out of range. It must be >= 0 and <= 5788. Received 6511
at new NodeError (node:internal/errors:399:5)
at boundsError (node:internal/buffer:88:9)
at Buffer.readInt32LE (node:internal/buffer:392:5)
at E3dcRscp.processFrame (/opt/iobroker/node_modules/iobroker.e3dc-rscp/main.js:1338:74)
at Socket. (/opt/iobroker/node_modules/iobroker.e3dc-rscp/main.js:522:10)
at Socket.emit (node:events:513:28)
at addChunk (node:internal/streams/readable:324:12)
at readableAddChunk (node:internal/streams/readable:297:9)
at Readable.push (node:internal/streams/readable:234:10)
at TCP.onStreamRead (node:internal/stream_base_commons:190:23)
2023-04-18 06:26:20.952  - error:  e3dc-rscp.0 (5462) Exception-Code: ERR_OUT_OF_RANGE: The value of  "offset" is out of range. It must be >= 0 and <= 5788. Received  6511
2023-04-18 06:26:21.035  - warn: e3dc-rscp.0 (5462) Received message with invalid MAGIC: >2BD1<
2023-04-18 06:26:21.036  - warn: e3dc-rscp.0 (5462) Received message with invalid CTRL: >69A8<
2023-04-18 06:26:21.036  - warn: e3dc-rscp.0 (5462) Received message with inconsistent length: 6528 vs 21841
2023-04-18 06:26:21.040  - warn: e3dc-rscp.0 (5462) Unknown tag: tagCode=0xa9a66a4e, len=30773, typeCode=0x96
2023-04-18 06:26:21.044  - info: e3dc-rscp.0 (5462) terminating
2023-04-18 06:26:21.045  - warn: e3dc-rscp.0 (5462) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-04-18 06:26:21.054  - warn: e3dc-rscp.0 (5462) E3/DC connection closed
2023-04-18 06:26:21.054  - info: e3dc-rscp.0 (5462) Stop communication with E3/DC and pause a minute before retry ...
2023-04-18 06:26:21.642  - error: host.iobroker instance system.adapter.e3dc-rscp.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-04-18 06:26:21.643  - info: host.iobroker Restart adapter system.adapter.e3dc-rscp.0 because enabled
2023-04-18 06:26:51.766  - info: host.iobroker instance system.adapter.e3dc-rscp.0 started with pid 16441
2023-04-18 06:26:56.505  - info:  e3dc-rscp.0 (16441) starting. Version 1.2.0 (non-npm:  git-kick/ioBroker.e3dc-rscp#4f870640e4806923fa86221ca129283714f9eb98) in  /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v18.15.0,  js-controller: 4.0.24
2023-04-18 06:26:56.673  - info: e3dc-rscp.0 (16441) Connection to E3/DC is established
2023-04-18 06:26:57.233  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:04.315  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:11.297  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:18.259  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd
2023-04-18 06:27:25.220  - warn: e3dc-rscp.0 (16441) Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd

git-kick, 2023-04-20:
> Hatte heute um 6:00 Uhr noch eine Fehlermeldung im LOG, leider hatte ich die LOG Stufe wieder auf Info gestellt. Eventuell hilft dir die Fehlermeldung weiter:
>
> 2023-04-18 06:26:20.925 - warn: [...]

Das passiert bei mir auch (selten). Vermutlich zeigt das ein Problem auf TCP/IP Ebene an. Die größeren Nachrichten werden nämlich in Pakete unterteilt und wenn da eines nicht ankommt, dann bleibt die Nachricht unvollständig, der Adapter erkennt das an den Längenangaben und verwirft die Nachricht.

Solange das sehr sporadisch auftritt, kann man es m.E. ignorieren, weil es ja immer eine nächste Abfrage gibt, die dann hoffentlich gut geht. Wenn es häufiger auftritt, bitte mal mit silly-log laufen lassen. Unter Verdacht steht aber dann vor allem das LAN (incl. Treiber).

Grundsätzlich könnte der Adapter solche Fehler auch einfach unterdrücken, aber ich denke ein "warn" ist es schon wert.

git-kick, 2023-04-23:
Die Erweiterung um `INFO` wird demnächst als V1.2.0 released.

ArnoD15, 2023-04-23:
Habe gerade die neue Version installiert und habe immer noch die Meldung "Unknown tag: tagCode=0xa800036, len=8, typeCode=0xd".

ArnoD15, 2023-04-23:
OK, da war ich wieder mal zu schnell :-)

git-kick, 2023-04-28:
Nachdem v1.2.0 ein Flop war, kann man jetzt v1.2.1 im "latest" channel probieren. Nicht "stable", ohne Garantie ;-)

closed 2023-05-06 by git-kick