git-kick/ioBroker.e3dc-rscp #150: SG Ready status

nase30, 2023-04-16:
Ist it possible to get the SG Ready status as well (Relais A and B) ?

git-kick, 2023-04-16:
I don't know what "SG ready status" means.

If you want find potential RSCP Tags containing the information you need, I recommend to look into the official tag list, an Excel contained in the RscpExample:
[image]

nase30, 2023-04-16:
I reviewed all the RSCP Data on iobroker and haven't found the SG-Ready status, nor in the documentation. May be they are named different and I'm not aware of. I'm looking for the status of the SG0 and SG1 actor.
[image]

git-kick, 2023-04-17:
Seems like SG ready is a separate interface.
If you have reviewed all RSCP tags and the information you are looking for is not there, then the e3dc-rscp adapter cannot fulfill your need.

closed 2023-04-17 by git-kick