git-kick/ioBroker.e3dc-rscp #153: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 5788. Received 6533

walterw3, 2023-04-24:
**Versions:**  
 - Adapter version: v1.2.0
 - JS-Controller version: 4.0.24
 - Node version: 16.17.1
 - Operating system: QNAP Firmware-Version QTS 5.0.1.2346 Build 20230322

1. Installed new version v1.2.0
2. received error RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 5788. Received 6533

Happens so often that host stops adapter. Attached lofgile.
[iobroker.2023-04-24.log.html.zip](https://github.com/git-kick/ioBroker.e3dc-rscp/files/11309063/iobroker.2023-04-24.log.html.zip)

git-kick, 2023-04-24:
v1.2.0 was not published in ioBroker. In fact, it is an erroneous build and marked 'deprecated' on npm. Please downgrade to v1.1.2

closed 2023-04-24 by git-kick

git-kick, 2023-04-28:
You can now try v1.2.1 from the "latest" channel.
Not "stable", no guarantees ;-)