git-kick/ioBroker.e3dc-rscp #157: I have set the TAG_PVI_REQ_FREQUENCY_UNDER_OVER to 'N' , but it shows a warning.

SurfGargano, 2023-04-28:
**Describe the bug**  
I have set the TAG_PVI_REQ_FREQUENCY_UNDER_OVER to 'N' , but it shows a warning.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
With disabled tag by 'N' it should not show a warning 

**Screenshots & Logfiles**  
<html>
<body>

3dc-rscp.0 | 2023-04-28 09:11:00.519 | warn | Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
-- | -- | -- | --

</body>
</html>
**Versions:**  
 - Adapter version: <1.2.1>
 - JS-Controller version: <js-controller-version> 
 - Node version: <node-version> 
 - Operating system: <os-name>

**Additional context**  
Add any other context about the problem here.

git-kick, 2023-04-28:
Could you please retry after #156 is solved for you. I suspect the same root cause. Thanks.

SurfGargano, 2023-04-28:
I have delete the instance and add a new one, but the problem still remains

SurfGargano, 2023-04-28:
yes

git-kick, 2023-05-06:
I think I have found the root cause: the adapter sends 5 initial requests for "all data", ignoring the S/M/L/N-setting.
Is it true that you see max. 5 of those warnings after every instance start?

SurfGargano, 2023-05-06:
Sorry, in the Moment I can't check.

git-kick, 2023-05-26:
Release v1.2.2 available at npm (not yet "stable" in ioBroker-Repo).

closed 2023-05-26 by git-kick