git-kick/ioBroker.e3dc-rscp #159: Invalid CTRL / MAGIC in received message

cavernerg, 2023-05-06:
**Describe the bug**  
Bei Ausführung erscheinen die Warnings aus dem Logauszug unten. Einstellungen sind standard mit den Anpassungen für das E3DC Control Skript. 

**To Reproduce**  
Steps to reproduce the behavior:
1. start instance
2. watch log

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
`

e3dc-rscp.0 | 2023-05-06 19:01:33.365 | debug | IN: magic: >3032< is WRONG  - ctrl: >3220< is  WRONG  - seconds: 63982970484017 - nseconds: 1395138653 - length: 20054   - invalid tag: 0x3336363A  no CRC32
-- | -- | -- | --
e3dc-rscp.0 | 2023-05-06 19:01:33.365 | warn | Received message with inconsistent length: 928 vs 20072
e3dc-rscp.0 | 2023-05-06 19:01:33.364 | warn | Received message with invalid CTRL: >3220<
e3dc-rscp.0 | 2023-05-06 19:01:33.364 | warn | Received message with invalid MAGIC: >3032<
e3dc-rscp.0 | 2023-05-06 19:01:33.348 | debug | IN: magic: >E3DC< is OK  - ctrl: >0011< is OK -  Version 1, with CRC  - seconds: 1683392496 - nseconds: 158406000 -  length: 6691  TAG_INFO_SERIAL_NUMBER - type: 0x0D - CString - length: 16 value:  S10-722138016407   TAG_INFO_PRODUCTION_DATE - type: 0x0D - CString - length: 9 value: KW38  2021    - invalid tag: 0xA800003  no CRC32
e3dc-rscp.0 | 2023-05-06 19:01:33.348 | warn | Received message with inconsistent length: 5792 vs 6713
e3dc-rscp.0 | 2023-05-06 19:01:33.154 | debug | state e3dc-rscp.0.RSCP.AUTHENTICATION changed: 10 (ack = true)
e3dc-rscp.0 | 2023-05-06 19:01:33.151 | debug | Sending request TAG_INFO_REQ_SERIAL_NUMBER
e3dc-rscp.0 | 2023-05-06 19:01:33.143 | debug | Sending request TAG_SYS_REQ_IS_SYSTEM_REBOOTING
e3dc-rscp.0 | 2023-05-06 19:01:33.109 | debug | Sending request TAG_EP_REQ_IS_READY_FOR_SWITCH
e3dc-rscp.0 | 2023-05-06 19:01:33.104 | debug | Sending request TAG_EMS_REQ_GET_POWER_SETTINGS
e3dc-rscp.0 | 2023-05-06 19:01:33.098 | debug | Sending request TAG_PVI_REQ_DATA
e3dc-rscp.0 | 2023-05-06 19:01:33.091 | debug | Sending request TAG_PVI_REQ_DATA
e3dc-rscp.0 | 2023-05-06 19:01:33.085 | debug | Sending request TAG_PVI_REQ_DATA
e3dc-rscp.0 | 2023-05-06 19:01:33.076 | debug | Sending request TAG_BAT_REQ_DATA
e3dc-rscp.0 | 2023-05-06 19:01:33.066 | debug | Sending request TAG_BAT_REQ_DATA
e3dc-rscp.0 | 2023-05-06 19:01:33.065 | info | Connection to E3/DC is established
e3dc-rscp.0 | 2023-05-06 19:01:33.061 | debug | Sending request TAG_RSCP_REQ_AUTHENTICATION
e3dc-rscp.0 | 2023-05-06 19:01:33.057 | debug | Probing for PVI units - 0..2.
e3dc-rscp.0 | 2023-05-06 19:01:33.057 | debug | Probing for BAT units - 0..1.
e3dc-rscp.0 | 2023-05-06 19:01:33.050 | debug | config.*: (192.168.178.zzz, 5033, xx@yy.de, 2, 2, 12, 0)
e3dc-rscp.0 | 2023-05-06 19:01:32.970 | info | starting.  Version 1.2.2 (non-npm:  git-kick/ioBroker.e3dc-rscp#84610977ec33cef296526ed72a1a2792418e6c3c) in  /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v16.19.1,  js-controller: 4.0.24
`

git-kick, 2023-05-08:
Die RSCP-Nachrichten werden auf TCP/IP Ebene in mehrere Teile zerlegt und beim Empfänger wieder zusammengesetzt. Wenn das schiefgeht, erscheinen solche Warnungen. Was genau diese Situation im LAN hervorruft, kann ich nicht sagen.

Wenn das nicht zu oft auftritt, ist es kein Problem, weil die Werte ja bald wieder aktualisiert werden.

Ich werde die Meldungen auf `debug` herabstufen, weil sie schon öfter irritiert haben.

git-kick, 2023-05-08:
Merged into master.
Kommt mit V1.2.2