git-kick/ioBroker.e3dc-rscp #163: Enhancement: enable writing into InfluxDB

github-ps1304, 2023-05-17:
**Describe the bug**  
Enhancement: it would be great to write all values into a InfluxDB instance of iobroker. Otherwise you need to select each value.

**To Reproduce**  
a setting to pick an InfluxDB instance from iobroker would be great

**Expected behavior**  
write all or selected values into InfluxDB instance

buanet, 2023-05-17:
Really? This is not how ioBroker works. Use Filters in Objects, then use the wrench button. You will be able to set influxdb settings for all object matching your filter!

For example: 
[image]

[image]

Regards,
André

github-ps1304, 2023-05-17:
I wasnt aware of that feature - cool; thanks for the headsup - will try it immediately. !!!

closed 2023-05-17 by github-ps1304