git-kick/ioBroker.e3dc-rscp #166: Please consider fixing problems reported by adapter checker

mcm1957, 2023-05-25:
Please add missing languages. a simple ```npm run translate``` should do the joib if using the up-to-date release of dev-tools (https://github.com/ioBroker/adapter-dev)

Thanks for your support
McM1957

              ## Automated adapter checker

### [ioBroker.e3dc-rscp](https://github.com/git-kick/ioBroker.e3dc-rscp)
[image] [image] [image]
[image]

:thumbsup: No errors found
- [ ] :eyes: [W171] "common.title" is deprecated in [io-package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/io-package.json)
- [ ] :eyes: [W105] "common.titleLang" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W109] "common.desc" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W156] Adapter should support admin 5 UI (jsonConfig) if you do not use a React based UI
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W145] Each "common.news" should be translated into all supported languages (en, de, ru, pt, nl, fr, it, es, pl, uk, zh-cn)
- [ ] :eyes: [W513] "gulpfile.js" found in repo! Think about migrating to @iobroker/adapter-dev package

*Add comment "RE-CHECK!" to start check anew*

_Originally posted by @GermanBluefox in https://github.com/ioBroker/ioBroker.repositories/issues/2310#issuecomment-1563345786_

git-kick, 2023-05-26:
I fixed the translation stuff and removed `gulpfile.js`

But after searching for an hour, I cannot figure out how to fix:

> [W156] Adapter should support admin 5 UI (jsonConfig) if you do not use a React based UI

@mcm1957, could you give me some starting point or template which I have to follow?

ka-vaNu, 2023-05-26:
Perhaps you want to have a Look at my blebox-Adapter. I've implemented/configured a AdminUI5 https://github.com/ka-vaNu/ioBroker.blebox/blob/master/admin/jsonConfig.json

git-kick, 2023-05-28:
Ok @mcm1957 , thanks for your quick check & response.

closed 2023-05-28 by git-kick