git-kick/ioBroker.e3dc-rscp #167: SUM.CONSUMPTION, SUM.BAT_POWER_OUT values are wrong

MrIngenieur, 2023-05-29:
**Describe the bug**  
In every query all the other response values are correct except:
e3dc-rscp.0.DB.HISTORY_DATA_***.SUM.CONSUMPTION

and 
e3dc-rscp.0.DB.HISTORY_DATA_***.SUM.BAT_POWER_OUT

This goes for DAY/WEEK/MONTH/YEAR. 

Typically the value is too small, compared to the app value. 

Year:
iobroker: 1621857Wh = 1,62 MWh 
E3DC-Portal: 2,23 MWh
2592000
31536000
2023-01-01 00:00:00.000

Month:
iobroker: 320277 = 320 kWh
E3DC-Portal: 366 kWh
86400
2678400
2023-05-01 00:00:00.000

Week:
iobroker: 4569
E3DC-Portal: 4,96
14400
604800
2023-05-29 00:00:00.000

If I sum up VALUE_00 to VALUE_XY values of "e3dc-rscp.0.DB.HISTORY_DATA_xxx.VALUE_xy.CONSUMPTION" then I get the same value as in e3dc-rscp.0.DB.HISTORY_DATA_xxx.SUM the the iobroker adapter seems to sum up correctly (or whoever does build the sum) but the value reported is wrong. 

Similar errors can be observed in e3dc-rscp.0.DB.HISTORY_DATA_xxx.SUM.BAT_POWER_OUT

**Expected behavior**  
Values are nearly identical to E3DC Portal (maybe wrong for minimum step size 15 minutes values, but not more)

**Versions:**  
 - Adapter version: v1.2.2
iobroker
Plattform
linux
Betriebssystem
linux
Architektur
arm
CPUs
4
Geschwindigkeit
1500 MHz
Modell
unknown
RAM
3.75 GB
System-Betriebszeit
09:05:00
Node.js
v18.16.0
time
1685358335336
timeOffset
-120
Anzahl der Adapter
474
NPM
9.5.1
Datenträgergröße
116.56 GB
freier Festplattenspeicher
94.56 GB
Betriebszeit
01:51:22
Aktive Instanzen
24
Weg
/opt/iobroker/
Hostname
iobroker

**Additional context**  
Add any other context about the problem here.

git-kick, 2023-05-30:
Thanks for your detailed issue report.

I repeated your measurements in my environment:

Year 2023 (kWh)
BAT_POWER_OUT  ioBroker=893,069 Portal=893,07
CONSUMPTION  ioBroker=2768,858 Portal=2768,78

Month May 2023 (kWh)
BAT_POWER_OUT  ioBroker=231,439 Portal=231,01
CONSUMPTION  ioBroker=585,230 Portal=584,85

Week from 29 May 2023 (kWh)
BAT_POWER_OUT  ioBroker=12,182 Portal=11,66
CONSUMPTION  ioBroker=36,182 Portal=35,68

I.e. the values are similar, as expected. Since the adapter does no calculation or interpretation on the values delivered via RSCP, I have no idea how to research the deviations you see within the adapter, **unless the adapter writes suspicious log entries** (switch on debug or silly log). If so, please post a log-file.

For further analysis, I recommend to measure intervals that end in the past, e.g. year 2022, month April 2023 - on order to exclude deviations due to asynchronous updates.

Please note: I have no access to documentation about E3/DC's firmware or portal software, so maybe we cannot trace this further.

git-kick, 2023-08-11:
Close after 2 months of no activity.

closed 2023-08-11 by git-kick