git-kick/ioBroker.e3dc-rscp #169: Warning entries in the iobroker log

github-ps1304, 2023-06-05:
**Describe the bug**  
entries in iobroker log

**To Reproduce**  

2023-06-05 06:14:47.808 | warn | Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
-- | -- | --
e3dc-rscp.0 | 2023-06-05 06:14:47.807 | warn | Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)

**Expected behavior**  
no warnings in the log

**Versions:**  
 - Adapter version: 1.2.3
 - JS-Controller version: 4.0.24
 - Node version: 18.16.0
 - Operating system: <Proxmox LXC Container (Debian 11)>

**Additional context**  
Add any other context about the problem here.

git-kick, 2023-06-05:
Thanks for your report.

This is a common observation because many E3/DC devices do not report these two tags. **You can avoid the warn log entries by setting both REQ-tags to 'N'** in adapter settings, tab "Polling Intervals".

I will keep this issue open and enhance the adapter so that it will stop polling these tags after receiving  RSCP_ERR_NOT_AVAILABLE.

git-kick, 2023-06-05:
The enhancement is included in [master branch](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/master) and will be released with v1.2.3

closed 2023-06-05 by git-kick