git-kick/ioBroker.e3dc-rscp #172: many Errors

Matis-11, 2023-08-04:
**Describe the bug**  
Version 1.2.2 throws many log-errors

**To Reproduce**  
just use it, see errorlog

**Expected behavior**  
no errors

**Screenshots & Logfiles**  

`2023-08-04 13:59:14.242  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)
2023-08-04 13:59:14.245  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2023-08-04 13:59:14.288  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)
2023-08-04 13:59:14.290  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2023-08-04 13:59:21.249  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)
2023-08-04 13:59:21.250  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2023-08-04 13:59:21.358  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)
2023-08-04 13:59:21.359  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2023-08-04 13:59:28.285  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)
2023-08-04 13:59:28.289  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2023-08-04 13:59:28.378  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)
2023-08-04 13:59:28.380  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)
2023-08-04 13:59:28.923  - warn: e3dc-rscp.0 (3808) TAG_EMS_REQ_POWER_PV has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.926  - warn: e3dc-rscp.0 (3808) TAG_EP_REQ_EP_RESERVE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.937  - warn:  e3dc-rscp.0 (3808) TAG_SYS_REQ_IS_SYSTEM_REBOOTING has no assigned  polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.937  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_SERIAL_NUMBER has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.938  - warn:  e3dc-rscp.0 (3808) TAG_INFO_REQ_PRODUCTION_DATE has no assigned polling  interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.938  - warn:  e3dc-rscp.0 (3808) TAG_INFO_REQ_MODULES_SW_VERSIONS has no assigned  polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.939  - warn:  e3dc-rscp.0 (3808) TAG_INFO_REQ_A35_SERIAL_NUMBER has no assigned  polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.939  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_IP_ADDRESS has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.940  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_SUBNET_MASK has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.940  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_MAC_ADDRESS has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.941  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_GATEWAY has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.941  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_DNS has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.942  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_DHCP_STATUS has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.942  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_TIME has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.942  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_UTC_TIME has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.943  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_TIME_ZONE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.943  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_SW_RELEASE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.944  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_GUI_TARGET has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.944  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_GET_FACILITY has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.944  - warn: e3dc-rscp.0 (3808) TAG_INFO_REQ_GET_FS_USAGE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.945  - warn:  e3dc-rscp.0 (3808) TAG_WB_REQ_CONNECTED_DEVICES has no assigned polling  interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.945  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_STATUS has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.946  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_ERROR_CODE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.946  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_MODE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.947  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_DEVICE_ID has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.947  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_DEVICE_NAME has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.948  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_DEVICE_STATE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.948  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_APP_SOFTWARE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.949  - warn:  e3dc-rscp.0 (3808) TAG_WB_REQ_BOOTLOADER_SOFTWARE has no assigned  polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.949  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_HW_VERSION has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.950  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_FLASH_VERSION has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.950  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_ENERGY_ALL has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.951  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_ENERGY_SOLAR has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.951  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_SOC has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.952  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_PM_POWER_L1 has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.952  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_PM_POWER_L2 has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.952  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_PM_POWER_L3 has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.953  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_PM_ACTIVE_PHASES has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.953  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_PM_MODE has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.954  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_PM_ENERGY_L1 has no assigned polling interval  - assuming 'M' - please check io-package.json
2023-08-04 13:59:28.954  - warn: e3dc-rscp.0 (3808) TAG_WB_REQ_PM_ENERGY_L2 has no assigned polling interval  - assuming 'M' - please check io-package.json
[... cut ...]
2023-08-04 13:59:42.504  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_VOLTAGE_MONITORING (0x2000070)
2023-08-04 13:59:42.505  - warn: e3dc-rscp.0 (3808) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_PVI_REQ_FREQUENCY_UNDER_OVER (0x2000080)`

**Versions:**  
 - Adapter version: 1.2.2
 - JS-Controller version: 4.0.24
 - Node version: 18.17.0
 - Operating system: linux (Synology docker)

**Additional context**  
Add any other context about the problem here.

git-kick, 2023-08-06:
Most of the messages (`RSCP_ERR_NOT_AVAILABLE (6)`) are warnings due to E3/DC model differences, please see #169 
(there is a work-around, and it will be resolved with v1.2.3).

Some (fewer) messages are like: `TAG_EMS_REQ_POWER_PV has no assigned polling interval` - this message says there are entries missing in `io-package.json` - please make sure you have the v1.2.2 file installed, I just checked and it has entries for the tags warned about.

Matis-11, 2023-08-06:
Thanks, I checked again and even so I had installed 1.2.3 and 1.2.2 again, there were many labels missing the polling interval page. I had to remove the whole adapter and reinstall 1.2.2 again to get all the labels. Thanks again.

closed 2023-08-06 by Matis-11