git-kick/ioBroker.e3dc-rscp #174: Some tags like TAG_EP_REQ_IS_GRID_CONNECTED are not updated with the set polling interval of 15 s

trapezoid677, 2023-08-13:
They are not updated at all.

**Versions:**  
 - S10-E8
 - 1.2.1 and 1.2.2
 - JS-Controller version: 5.0.11
 - Node version: 16.20.1
 - Operating system: Manjaro Linux

git-kick, 2023-08-13:
please see https://github.com/git-kick/ioBroker.e3dc-rscp#bug-reports

No input, no output ;-)

trapezoid677, 2023-08-19:
Is this sufficient?
[iobroker.current_e3dc.log.html.zip](https://github.com/git-kick/ioBroker.e3dc-rscp/files/12385953/iobroker.current_e3dc.log.html.zip)

git-kick, 2023-08-21:
Thank you for providing the log-file.

I looked at IS_GRID_CONNECTED. It is requested for times, and every time there is a valid response with value "0x01" (true).

```
// Request 1
2023-08-19 15:20:28.339  - silly:  e3dc-rscp.0 (20803) OUT: magic: >E3DC< is OK  - ctrl:  >0011< is OK - Version 1, with CRC  - seconds: 1692451228 -  nseconds: 0 - length: 42
TAG_EP_REQ_IS_GRID_CONNECTED - type: 0x00 - None - length: 0
...
2023-08-19 15:20:28.412  - silly:  e3dc-rscp.0 (20803) IN: magic: >E3DC< is OK  - ctrl: >0011<  is OK - Version 1, with CRC  - seconds: 1692450936 - nseconds:  116093000 - length: 132
TAG_EP_IS_GRID_CONNECTED - type: 0x01 - Bool - length: 1 value: 0x01

// Request 2
2023-08-19 15:20:35.372  - silly:  e3dc-rscp.0 (20803) OUT: magic: >E3DC< is OK  - ctrl:  >0011< is OK - Version 1, with CRC  - seconds: 1692451235 -  nseconds: 0 - length: 42
TAG_EP_REQ_IS_GRID_CONNECTED - type: 0x00 - None - length: 0
...
2023-08-19 15:20:35.378  - silly:  e3dc-rscp.0 (20803) IN: magic: >E3DC< is OK  - ctrl: >0011<  is OK - Version 1, with CRC  - seconds: 1692450943 - nseconds:  135383000 - length: 132
TAG_EP_IS_GRID_CONNECTED - type: 0x01 - Bool - length: 1 value: 0x01

// Request 3
2023-08-19 15:20:42.360  - silly:  e3dc-rscp.0 (20803) OUT: magic: >E3DC< is OK  - ctrl:  >0011< is OK - Version 1, with CRC  - seconds: 1692451242 -  nseconds: 0 - length: 42
TAG_EP_REQ_IS_GRID_CONNECTED - type: 0x00 - None - length: 0
...
2023-08-19 15:20:42.367  - silly:  e3dc-rscp.0 (20803) IN: magic: >E3DC< is OK  - ctrl: >0011<  is OK - Version 1, with CRC  - seconds: 1692450950 - nseconds:  122385000 - length: 132
TAG_EP_IS_GRID_CONNECTED - type: 0x01 - Bool - length: 1 value: 0x01

// Request 4
2023-08-19 15:20:43.329  - silly:  e3dc-rscp.0 (20803) OUT: magic: >E3DC< is OK  - ctrl:  >0011< is OK - Version 1, with CRC  - seconds: 1692451243 -  nseconds: 0 - length: 35
TAG_EP_REQ_IS_GRID_CONNECTED - type: 0x00 - None - length: 0
...
2023-08-19 15:20:43.336  - silly:  e3dc-rscp.0 (20803) IN: magic: >E3DC< is OK  - ctrl: >0011<  is OK - Version 1, with CRC  - seconds: 1692450951 - nseconds: 94850000  - length: 40
TAG_EP_IS_GRID_CONNECTED - type: 0x01 - Bool - length: 1 value: 0x01
```
Timestamps say that the intervals between requests are: 7s, 7s, 1s
Shorter intervals after adapter startup are intended to quickly fill the object tree.
So far, everything looks fine. I cannot see anything unexpected. 

How exactly do you reproduce and observe that IS_GRID_CONNECTED does not get updated? 
Are the other EP_ tags showing data as expected?

trapezoid677, 2023-08-21:
These datapoints are updated:

e3dc-rscp.0.EP.PARAM_0.PARAM_EP_RESERVE_MAX_W
e3dc-rscp.0.EP.PARAM_0.PARAM_LAST_SOC
e3dc-rscp.0.EP.PARAM_0.PARAM_TIME_LAST_FULL

For the other EP tags, the last updates have been between 2 and 5 months ago.

[image]
[image]

git-kick, 2023-08-21:
I think some of the EP tags may not change for months. Do you have a sample where you know the value changed in the E3/DC, but the change was not taken to the e3dc-rcps object tree?
I still can't get your point what's going wrong.

trapezoid677, 2023-08-21:
I use an interval S (15 sec) to request the status of the IS_GRID_CONNECTED tag. I expect to receive a response every 15 sec. Whether I save these responses or not is handled by the SQL-module where I can neglect unchanged values for a certain timeframe. But if an interval of 15 sec is set, the module should not decide on its own if the value is signaled or not. At least I don't know any other modules which neglect values on their own, thereby ignoring the interval being set.

git-kick, 2023-08-21:
ok, got it.
Indeed, the e3dc-rscp adapter currently does not call setState() unless the value changed. This is an intentional optimization motivated by the massive amount of tags the E3/DC has (which overloadef some Raspi installations). 
But keep in mind the interval is *not* ignored - the adapter polls according to the interval.
Also, I could not find any spec defining whar ts is (time of last change vs time of last message). To me, it makes no sense relying on the latter: imagine the E3/DC had an interface throwing an event on every value change; you never would have those periodical "it's still the same value". I.e. your application should not depend on such implementation details of the interface. 

Nevertheless, I could add a configuration switch for that, if you could provide some evidence that ts *is* positively defined as "time when last checked", not "time when last changed".

trapezoid677, 2023-08-22:
There is at least one use case which I personally have: I want to debug Node-Red-Scripts which is impossible if no events are generated. And waiting for a power grid outage for debugging...

ArnoD15, 2023-08-22:
Ich antworte mal auf Deutsch, da es hier anscheinend alle verstehen und es für mich einfacher ist :-)
Ich will mich hier wirklich nicht einmischen, aber ich bin auch schon über das gleiche Problem gestolpert.
In der ioBroker Dokumentation und auch in anderen wird erklärt, dass es sich bei "ts" um einen Unix Timestamp (Zeitstempel in Millisekunden) wann der Zustand **zuletzt aktualisiert wurde** handelt und zwar auch, wenn sich der Wert nicht geändert hat.
Die letzte Änderung bekommt man über "lc" mit.

Wenn man es aber weiß, kann man auch damit leben, dass sich der Zeitstempel "ts" auch nur bei einer Änderung aktualisiert. Es ist aber so nicht möglich festzustellen, ob die Schnittstelle noch richtig funktioniert, da man ja keine Rückmeldung bei Aktualisierung erhält.

git-kick, 2023-08-22:
Hallo @ArnoD15, danke für die Erklärung.

Da ich wirklich bemüht bin, immer zuerst die Doku konsultieren: wo steht, "dass es sich bei "ts" um einen Unix Timestamp (Zeitstempel in Millisekunden) wann der Zustand zuletzt aktualisiert wurde handelt *und zwar auch, wenn sich der Wert nicht geändert hat*."?

ArnoD15, 2023-08-22:
Muss mal suchen, wo ich das gelesen habe. 
Es ist jedenfalls so umgesetzt worden. 
Wenn du dir im ioBroker bei Objekte die Statusansicht einschaltest, kannst du bei allen Objekte feststellen, dass es so gehandhabt wird.
Der Zeitstempel "ts" würde ja sonst auch keinen Sinn ergeben, denn für Änderungen wo sich auch der Wert geändert hat, gibt es ja bereits den Zeitstempel "lc"

git-kick, 2023-08-28:
Neue Konfigurations-Checkbox ist im master branch und kommt mit dem v1.2.4 Release.

closed 2023-08-28 by git-kick

ArnoD15, 2023-09-01:
Danke, werde ich dann gleich testen :-)