git-kick/ioBroker.e3dc-rscp #175: Abfrage Zählerstand

Berndhafen, 2023-08-14:
Hallo,
ich benutze den Adapter um einige Parameter an KNX zu übergeben und dort auf der Visualisierung anzuzeigen.
Gibt es die Möglichkeit die Zähler für Produktion/Hausverbrauch/Netzbezug/Einspeisung auszulesen ? Evtl. für den aktuellen Tag, für den letzten Tag / Woche / Jahr ? einfache Anzeige in Wh oder kWh ? Im Prinzip die Werte die auch das E3DC Portal zur Verfügung stellt.
Vielen Dank.

git-kick, 2023-08-14:
Hallo,
ja, die genannten Momentanwerte stehen alle im EMS Namespace. Kumulierte Werte kann man im DB Namespace abrufen. Das ist aber kaum dokumentiert, da muss man etwas mit den Werten spielen um zu verstehen, was die Anlage liefert.

Berndhafen, 2023-08-14:
Danke für die schnelle Rückmeldung, die Momentanwerte habe ich, die frage ich auch ab, das klappt super. Aber im Ordner DB sind eben die History-Ordner und jeweils drei Zeit-Objekte aber keine Werte dazu.

git-kick, 2023-08-14:
Du musst die Werte TIME_INTERVAL/_SPAN/_START setzen, das triggert die DB-Anfrage.

Berndhafen, 2023-08-14:
Super, vielen Dank !!!
Genau das habe ich gesucht :-)

Danke & Grüße

closed 2023-08-14 by git-kick

git-kick, 2023-08-14:
Da hilft nur Probieren.
Meine bescheidenen Erkenntnisse habe ich hier aufgeschrieben: https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/README-dev.md#db-namespace-semantics