git-kick/ioBroker.e3dc-rscp #184: Feature request to support Changing "e3dc-rscp.0.EMS.MANUAL_CHARGE_ACTIVE" parameter

bwarnsing, 2023-09-30:
**Describe the bug/ It is not a bug**  
In current version the paramter e3dc-rscp.0.EMS.MANUAL_CHARGE_ACTIVE is not writeable.
Would be great to make that changable/ writeable.

**Versions:**  
 - E3/DC model and battery type : S10 / Panasonic
 - Adapter version: 1.2.3
 - JS-Controller version: 4.0.24
 - Node version: v16.20.2
 - Operating system: Linux

**Additional context**  
I need that writeable to charge Battery from Grid if Stock prices go down!

git-kick, 2023-10-02:
Thanks for your request.

To start manual charge, REQ_START_MANUAL_CHARGE must be sent to the E3/DC. I will do my best to add this to the next release.

git-kick, 2023-10-08:
Note: according to the E3/DC App, manual charging is a testing function and executed only once a day.
(I did not confirm this by own testing.)

bwarnsing, 2023-10-08:
thanks for letting me know. I'm aware. Let me know if you need somebody to test or play.

git-kick, 2023-10-08:
You can try starting a manual charge:
https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23184

It worked for me, but only once, as announced. 
But the code is not finalized yet (naming, reset energy value after charging is finished, ack energy value after it was confirmed)

bmause, 2023-10-09:
Hallo, bin auch daran sehr Interessiert ein manuelles Laden mit den Tibberlink Adapter und nach besten Preis zu laden.
So ganz habe ich die Datenpunkte zum manuellen Laden per Skript noch nicht gefunden.
EMS.SET_MANUAL_CHARGE_ENERGY = e3dc-rscp.0. EMS.MANUAL_CHARGE_ENERGY_COUNTER ? REQ_START_MANUAL_CHARGE = habe ich noch nicht gesehen in den Adapter
installierte Version 1.2.3

git-kick, 2023-10-14:
@bmause: ja, kann noch nicht da sein, kommt (frühestens) mit V1.2.4, siehe [oben](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/184#issuecomment-1743411224)

git-kick, 2023-10-16:
Code for this feature is now completed and merged to the [master branch](url).
Note that the object is now called e3dc-rscp.0.EMS.MANUAL_CHARGE_ENERGY (without "SET")

Will be released with v1.2.4, sooner or later ;-)

git-kick, 2023-10-27:
Implemented in v1.2.4

closed 2023-10-27 by git-kick