git-kick/ioBroker.e3dc-rscp #185: "Wallbox" -> "Im Sonnenmodus" -> "Bis SOC" Wert nicht auffindbar

tommy-rk, 2023-10-03:
Hallo,
ich möchte den Wert "Wallbox" -> "Im Sonnenmodus" -> "Bis SOC" nach Tageszeit ändern. Kann diesen aber nirgends finden.
Hier ein Foto von meiner E3DC:
[image]

Meine Idee ist:
Morgens ist mein Akku immer noch zu 40-50% geladen und ich möchte mein Auto bevor ich losfahre noch laden bis die Hausbatterie bei 5% ist.

Aktuell stelle ich den Wert manuell von Hand an der E3DC im Display um.

Wenn ich aber am Mittag das Auto laden möchte, ist die Batterie meist schon voll geladen, oder zumindest bei 90%-100%. Dann möchte ich nicht mehr die Hausbatterie entladen, da ich sie ja noch für die Nacht benötige. In dem Fall stelle ich aktuell den Wert "Bis SOC" manuell auf 95% um. Dann lädt mein Auto nur noch mit dem Strom aus der PV Anlage.

Gibt es eine Möglichkeit, dies zu automatisieren? Leider kann ich nirgends den Wert "Bis SOC" in den Objekten finden. Das nächste, das ich finden konnte, ist:
e3dc-rscp.0.WB.WB_0.Control.SunMode

j64bit, 2023-10-03:
Hi,
Danke für den tollen Adapter. Der ist so hilfreich!

An diesem neuen Feature wäre ich auch sehr interessiert.
In [https://github.com/pvtom/rscp2mqtt/blob/main/RscpTags.h](url)
habe ich dies gefunden:
```
#define TAG_EMS_GET_WB_DISCHARGE_BAT_UNTIL                   	0x0180027D
#define TAG_EMS_REQ_GET_WB_DISCHARGE_BAT_UNTIL               	0x0100027D
#define TAG_EMS_SET_WB_DISCHARGE_BAT_UNTIL                  	0x0180027C
#define TAG_EMS_REQ_SET_WB_DISCHARGE_BAT_UNTIL               	0x0100027
```
Sorry wenn das schon bekannt ist

git-kick, 2023-10-03:
Danke für das Interesse am e3dc-rscp Adapter und für die Vorschläge.

Leider kann ich nichts bearbeiten, was die Wallbox steuert. Der Grund ist hier ausführlich dargelegt:
https://github.com/git-kick/ioBroker.e3dc-rscp/issues/106

Wenn sich ein anderer Entwickler mit Wallbox findet, der das entwickelt und testet, wäre das fein.

tommy-rk, 2023-10-03:
Kann der Request ein paar Wochen offen bleiben? Geschlossene Requests findet man nicht in den Issues und ist die Chance nicht höher, wenn dieser Request eine höhere Visibilität hat, gelöst zu werden? Vielleicht findet sich ja jemand, der das angehen möchte, wie im Request 106?

ka-vaNu, 2023-10-04:
Ich kann mir das gerne in den nächsten Wochen ansehen. Muss mich allerdings erstmal wieder reinfuchsen 😉

j64bit, 2023-10-04:
Sehr cool, Leute. Wenn ich mit testen zB helfen kann, dann gerne.

tommy-rk, 2023-10-05:
Ich bin mir nicht sicher, ob das wirklich ein Wallbox-Feature ist? Ich habe die Wallbox abgeschaltet und das WB Logo ist aus dem Hauptscreen verschwunden.
Das Menüfeld findet sich **auch** unter:
Smart-Funktionen -> Smart Charge -> Ladepriorisierung

Es wird keine WB benötigt, um an diese Funktion zu kommen. Siehe Screenshot. Oben Links steht der Menüpunkt "Smart Charge" und nicht "Wallbox", wie in meinem ersten Bild. Es ist aber exakt der gleiche Inhalt.
[image]

Vielleicht ist dies ein Parameter, der auch ohne Wallbox verfügbar ist?

ka-vaNu, 2023-10-12:
> #define TAG_EMS_GET_WB_DISCHARGE_BAT_UNTIL                   	0x0180027D
> #define TAG_EMS_REQ_GET_WB_DISCHARGE_BAT_UNTIL               	0x0100027D
> #define TAG_EMS_SET_WB_DISCHARGE_BAT_UNTIL                  	0x0180027C
> #define TAG_EMS_REQ_SET_WB_DISCHARGE_BAT_UNTIL               	0x0100027

Wie am Namespace zu sehen ist, handelt es sich um EMS (Energy Management System?). Es ist also kein Wallbox-Namespace. Die beiden Punkte im Display beziehen sich intern auf den gleichen Wert. 

Dieser kann nun mittels des neuen State **e3dc-rscp.0.EMS.SET_WB_DISCHARGE_BAT_UNTIL** gesetzt werden. Der aktuelle Wert wird immer unter **e3dc-rscp.0.EMS.WB_DISCHARGE_BAT_UNTIL** angezeigt. Wenn also eine Änderung am iobroker vorbei am Display vorgenommen wird, wird dieser unter letztgenanntem State angezeigt.

Unter https://github.com/ka-vaNu/ioBroker.e3dc-rscp/tree/set_wb_bat_discharge_limit liegt der geändert Code. Bitte einmal testen und gerne Feedback, falls etwas nicht funktioniert.

git-kick, 2023-10-13:
Danke @ka-vaNu - der Code läuft auch bei mir, die Tags erscheinen im Objektbaum und der Wert lässt sich setzen. Aber richtig testen kann ich mangels E3/DC-Wallbox halt nicht.

Wenn die Beteiligten hier sagen, dass es funktioniert, dann baue ich das gerne in v1.2.4 ein. (Ich bitte um Geduld, habe aktuell wenig übrige Zeit.)

Woher hast du die Info über die WB_DISCHARGE_BAT Tags? Ich bin immer interessiert an Quellen zur RSCP-Schnittstelle!

ka-vaNu, 2023-10-13:
@j64bit hat mich auf das Projekt [rscp2mqtt](https://github.com/pvtom/rscp2mqtt) hingewiesen s. [hier](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/185#issuecomment-1744466020). Ich habe in der Entwicklerdoku einen entsprechenden Abschnitt hinzugefügt, in der die offizielle Doku und die mir bekannten Projekte hinterlegt sind.

ka-vaNu, 2023-10-13:
Habe soeben die anderen 3 Attribute auf der Maske implementiert (s. [Readme ](https://github.com/git-kick/ioBroker.e3dc-rscp/commit/8f008c6b2ec85e124341efbd6a2961f3a88b201e#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5))

Bitte ausgiebig testen. SET_BATTERY_BEFORE_CAR_MODE in bestimmten Situationen nicht deaktiviert werden (0xFF = Aktivierung nicht möglich (BatteryToCar noch aktiv?))

j64bit, 2023-10-14:
Hallo zusammen,

Danke für die super Arbeit. Bei mir funktionieren die Setting bestens wie erwartet. Man kann sogar "Bis SOC" einstellen, selbst wenn oben "Erst Batterie" steht. 
Den Hinweis oben zu SET_BATTERY_BEFORE_CAR_MODE verstehe ich nicht richtig und ich hatte auch kein 'nicht möglich'. 

Also soweit alles bestens und releasefähig würde ich sagen.

Danke nochmal

git-kick, 2023-10-15:
Ich habe im Branch [Issue#185](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23185) die Erweiterung in Absprache mit @ka-vaNu an die bisherige Implementierung angepasst und vervollständigt (z.B. polling intervals für die WB-tags). Die änderbaren Objekte heißen jetzt  

- e3dc-rscp.0.EMS.BATTERY_BEFORE_CAR_MODE
- e3dc-rscp.0.EMS.BATTERY_TO_CAR_MODE
- e3dc-rscp.0.EMS.WB_DISCHARGE_BAT_UNTIL
- e3dc-rscp.0.EMS.WB_ENFORCE_POWER_ASSIGNMENT

Bitte gerne testen... (kommt mit v1.2.4)

@ka-vaNu: in diesem Branch habe ich auch das [README-dev](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/Issue%23185/README-dev.md) ergänzt um ein Kapitel ["Introducing a new  setter tag"](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/Issue%23185/README-dev.md#introducing-a-new--setter-tag) - ich hoffe das hilft etwas. Wenn du dir den Code anschaust wirst du sehen, dass für die vier neuen setter keine zusätzliche `queue....()` Funktion nötig war; geht alles über `queueSetValue()`.

j64bit, 2023-10-15:
Hi
testen kann ich erst Morgen. Aber bei der Installation wird folgendes ausgegeben:
`2023-10-15 23:16:12.344  - warn:  e3dc-rscp.0 (373754) TAG_EMS_REQ_WB_DISCHARGE_BAT_UNTIL has no assigned  polling interval  - assuming 'M' - please check io-package.json`
`2023-10-15 23:16:12.347  - warn:  e3dc-rscp.0 (373754) TAG_EMS_REQ_WB_ENFORCE_POWER_ASSIGNMENT has no  assigned polling interval  - assuming 'M' - please check io-package.json`

Und für ca 20 weitere. Fehlt da noch etwas? Danke und Gruß

git-kick, 2023-10-16:
Hi, bitte den Adapter komplett löschen, neu installieren und die Konfiguration neu von Hand eingeben - Laden aus json file löscht die neuen polling interval Einträge.

git-kick, 2023-10-16:
Der Code ist jetzt fertig und merged in den [master branch](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/url).

Kommt mit v1.2.4, früher oder später ;-)

tommy-rk, 2023-10-19:
Wow! Tolle Leistung an alle Beteiligten. Vielen Dank!

ka-vaNu, 2023-10-19:
> Ich habe im Branch [Issue#185](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23185) die Erweiterung in Absprache mit @ka-vaNu an die bisherige Implementierung angepasst und [...]

Danke, das hilft auf jeden Fall sehr! E3dc scheint in der Regel den gleichen Pattern zu verwenden. Wenn dies ohne queueSet... zu lösen ist, vereinfacht dies die Implementierung erheblich ;-)

git-kick, 2023-10-27:
Implementiert in v1.2.4

closed 2023-10-27 by git-kick