git-kick/ioBroker.e3dc-rscp #192: S10 X High Voltage Battery RSCP_ERR_OUT_OF_BOUNDS Error on 12 Tags

mxbode, 2023-10-24:
Connect to a S10 X High Voltage System. The Tags get no updates.

Expected: Tags get Updated (Battery Voltage). I've data in my history but I guess its not valid.

Example logfile attached and history data attached.

 - E3/DC S10 X Compact 14/12 kWh
 - Adapter version: v 1.2.3
 - JS-Controller version: 4.0.24
 - Node version: v18.17.0
 - Operating system: Ubuntu 22.04.3 LTS

[Spannung 30-Sekunden-Durchschnitt.csv](https://github.com/git-kick/ioBroker.e3dc-rscp/files/13129691/Spannung.30-Sekunden-Durchschnitt.csv)
[E3DC_X_RSCP.log](https://github.com/git-kick/ioBroker.e3dc-rscp/files/13129694/E3DC_X_RSCP.log)

git-kick, 2023-10-25:
Why do you consider a DCB voltage of ca. 400 V not valid? Do you know the specified voltage of your DCBs?
In any case, the adapter just stores in that object what it receives from the RSCP-interface.

The RSCP_ERR_NOT_HANDLED log-messages will disappear when you diable WB in adapter options.

The RSCP_ERR_OUT_OF_BOUNDS messages suggest that the adapter does not correctly detect the number of BATs and DCBs in your setup. For further analysis, I need

- a **debug**-log from adapter start (ca. 30 sec)
- the detailed configuration of your batteries: for every BAT: how many DCBs

Having that, I can try to adjust the adapter heuristics - problem is that there are no RSCP-Requests like "give me the set of installed BATs and DCBs". Thanks.

mxbode, 2023-11-04:
> Why do you consider a DCB voltage of ca. 400 V not valid? Do you know the specified voltage of your DCBs?
Nevermind, there was a problem with my history database.

I have installed one module with 4 batterys.

Here is the requested log file.
[E3DC_Debug.log](https://github.com/git-kick/ioBroker.e3dc-rscp/files/13256573/E3DC_Debug.log)

git-kick, 2023-11-04:
> Here is the requested log file. [E3DC_Debug.log](https://github.com/git-kick/ioBroker.e3dc-rscp/files/13256573/E3DC_Debug.log)

OK, thank you. The debug log shows that 1 BAT is found, which is correct.
Unfortunately, the debug log does not report about the details of DCB responses. I'm afraid I need a silly-log to go deeper. 10 seconds from start should be good enough. Sorry for asking twice.

Please let me also know how many e3dc-rscp.0.BAT.BAT_0.DCB_... nodes you have in you object tree - should be four of them (0..3), because you have four DCBs. Do they show any values like
[image]

Are the subtrees
- e3dc-rscp.0.BAT.BAT_0.DCB_0.DCB_CELL_TEMPERATURE
- e3dc-rscp.0.BAT.BAT_0.DCB_0.DCB_CELL_VOLTAGE

existent and do they contain objects (which are not updated)?

closed 2023-11-16 by git-kick

git-kick, 2023-12-12:
I will reopen the issue if you answer all of my questions above
https://github.com/git-kick/ioBroker.e3dc-rscp/issues/192#issuecomment-1793488849

git-kick, 2024-01-03:
I am referring to your latest [log provided above](https://github.com/git-kick/ioBroker.e3dc-rscp/issues/192#issuecomment-1852680570).

Line #357: `TAG_BAT_DCB_COUNT - type: 0x03 - UChar8 - length: 1 value: 0x04`
Your E3/DC device tells us it has 4 DCBs - like you said from the beginning.

BUT: when the adapter asks for data about DCB 0,1,2, or 3 as in
Line #1192: `TAG_BAT_REQ_DCB_ALL_CELL_TEMPERATURES - type: 0x05 - UInt16 - length: 2 value: 0` 
where the `value: 0` is the DCB index, the device returns error records like:
Line #1264: `TAG_BAT_DCB_ALL_CELL_TEMPERATURES - type: 0xFF - Error - length: 4 value: 5`

There is only one exception: INFO for DCB_0
Line #1194: `TAG_BAT_REQ_DCB_INFO - type: 0x05 - UInt16 - length: 2 value: 0 `
which correctly respons
Line #1259ff: `TAG_BAT_DCB_INFO - type: 0x0E - Container - length: 48 <Container content follows...>`

I would regard this as a misbehaviour of the E3/DC device (not the adapter): the device should deliver values for DCBs 0,1,2,3 - or report a different DCB_COUNT.

I have no solution in mind currently. Will think about it.
It is kind of strange that you are the first (of > 1.000 users) reporting such DCB count mismatch...

git-kick, 2024-01-04:
**Option 1:** ignore error messages
For the e3dc-rscp adapter, everything works fine. The out-of-range errors are just "content" which is written to the log. They do not bother the adapter's execution or invalidate the object tree in any way.

**Option 2:** switch off polling
You can tell the adapter not to ask for
- `TAG_BAT_DCB_INFO`
- `TAG_BAT_DCB_ALL_CELL_TEMPERATURES`
- `TAG_BAT_DCB_ALL_CELL_VOLATGES`

by setting the polling interval to 'N' in the adapter configuration (second tab).
But: you will not see the INFO values for DCB_0 then.

**Option 3:** ask Hager
You may try to get information or a firmware update from the E3/DC manufacturer.

**Option 4:** make the adapter work around this special case
I could add some logic to the adapter to suppress further requests after receiving an out-of-range. Indeed, there is already such code for probing some counts which are not available explicitely (e.g. PVI). But I decided not to add additional complexity for a unique case where the device's responses are inconsistent.

Please let me know if you have any other suggestions.

mxbode, 2024-01-04:
Thank you very much!

I will first ignore the messages. When time allows I will create a ticket.

Wish you a great 2024.