git-kick/ioBroker.e3dc-rscp #199: Change emergency power setting

Taffmailer, 2023-11-13:
Under certain circumstances, it makes sense to release part of the stored emergency power reserve for in-house consumption.

Currently, the tag EP_PARAM_EP_RESERVE_W
can already be changed. However, this is not accepted by the system. After a short time, the initial value is displayed again.

If changes are made directly via the system, EP_PARAM_EP_RESERVE_W is always changed in addition to the EP_PARAM_EP_RESERVE. This tag  cannot currently be changed, it is read only.

It may be possible to solve the problem if EP_PARAM_EP_RESERVE is defined as read/write.

Please check and consider as a feature request.

regards
Franc

git-kick, 2023-11-13:
@Taffmailer, thank you for your contribution, I will try to make EP_PARAM_EP_RESERVE read/write.

git-kick, 2023-11-13:
I did a first implementation of read/write for PARAM_EP_RESERVE. Works on my S10. Note that PARAM_EP_RESERVE_W changes accordingly. Because it is an energy [Wh] value (not a power [W] value), I am thinking about renaming it to PARAM_EP_RESERVE_W**H**. The tag is not listed in the official tag list, so I assume it was named using best guess.

Available for testing in branch [Issue#199](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23199)

I will do some more trial & error and implement read/write for other PARAM_* tags, if appropriate. This may take some time...

git-kick, 2023-11-16:
Implementation is done. Two objects are writable
- PARAM_EP_RESERVE
- PARAM_EP_RESERVE_ENERGY (fka PARAM_EP_RESERVE_W)

The four other PARAM_... objects are all read-only, as they reflect E3/DC status.
 
Available for testing in branch [Issue#199](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23199); will be released with v1.2.5

closed 2023-11-16 by git-kick

Taffmailer, 2023-11-18:
Great!
Thanks a lot