git-kick/ioBroker.e3dc-rscp #207: Laden per iobroker, startet nicht

bmause, 2023-11-19:
leider bekomme ich das Batterie Laden nicht zum laufen, es passiert nichts am WR.
Wenn ich zum Beispiel:
SET_POWER_MODE = 3 setzte (CHARGE)
SET_POWER_VALUE = 3000 
mit Bestätigung und ohne, keine Auswirkungen, keine Fehlermeldungen oder Protokoll Einträge.

System Daten:
Version 1.2.5 in /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v18.18.2, js-controller: 5.0.16

git-kick, 2023-11-19:
Bitte um folgende Informationen:
- ein debug-log (bitte als file attachment, nicht inline) ab Adapter-Start bis 30 Sek nach Eingabe der Werte für SET_POWER_MODE und SET_POWER_VALUE
- welche Werte zeigen SET_POWER_MODE und SET_POWER_VALUE ca. 1 Minute nach der Eingabe? 
- wie sind in den Instanzeinstellungen die beiden Werte
   - SET_POWER Wiederholintervall
   - Tupel-Schreibverzögerung

eingestellt?

bmause, 2023-11-19:
Hallo, ich hoffe das sind alle Info die du benötigst.

Die Werte bleiben gleich bei den Datenpunkten.  
z.B. 3 Charge & 2000W

SET_POWER Wiederholintervall = 0
Tupel-Schreibverzörgerung = 10

[Laden per iobroker, startet nicht · Issue #207 · git-kick:ioBroker.e3dc-rscp.pdf](https://github.com/git-kick/ioBroker.e3dc-rscp/files/13404114/Laden.per.iobroker.startet.nicht.Issue.207.git-kick.ioBroker.e3dc-rscp.pdf)
[Laden per iobroker, startet nicht · Issue #207 .txt](https://github.com/git-kick/ioBroker.e3dc-rscp/files/13404208/Laden.per.iobroker.startet.nicht.Issue.207.txt)

bmause, 2023-11-19:
ich habe die SET_Power Wiederholintervall wieder den default = 10 gestellt.
Nun lädt wieder wenn ich die werde setzte.

Was wäre den für die SET_Power Wiederholinterval der beste Werte ?

Denke somit kann der Bug geschlossen werden.  :-)

git-kick, 2023-11-19:
Die Frage ist im README.md behandelt:
https://github.com/git-kick/ioBroker.e3dc-rscp#tab-options

closed 2023-11-19 by git-kick