git-kick/ioBroker.e3dc-rscp #208: Add energy consumption day, week, month, year

RalfJL, 2024-01-19:
S10 from E3DC also stores energy consumption per day, week, month and year in kwh.
It keeps those data for a while. So it is possible to request those data for a given date.

It would be great to add this data to ioBroker at least for today and yesterday.

git-kick, 2024-01-19:
Stored values are accessible in the DB namespace. See readme.md for the objects you have to set to trigger a query.
readme-dev.md describes some trials and caveats.

closed 2024-01-20 by git-kick

RalfJL, 2024-01-21:
Missed that.
Sorry for bothering you.

RalfJL, 2024-01-21:
Sorry again,
what I meant is that it would be great to have the production and consumption energy from the past timeperido (day, week, ..) out of the box.
For most people using ioBroker I think it is impossible to go through the developer section and find anything that they can make use of. I think not many people have the programming skills to do that. Though I have written S10history it will take me hours to update the adapter the first time. 
And it will be gone with the next update.

Of course it is your decision and I will not complain again.
Thank you for making this great adapter available to others.

git-kick, 2024-01-22:
Hi Ralf,

thank you for contributing, I can see your point.
Still, I decided not to add such simplification for two reasons:

1. I do not completely understand how DB works, and it is poorly documented; so whatever I do, it will probably create confusion and more work. (This holds unless somebody has worked out all the details...)
2. My design decision is to provide a "naked" adapter, no interpretation of values delivered by E3/DC. Everything else is deliberately left to adapter users and their code (Javascript, Blockly, ...)

Additionally, I doubt that _most_ of the ioBroker users have zero lines of own code.
Hoping for your understanding.

Best, Uli