git-kick/ioBroker.e3dc-rscp #209: Unknown tag after update from E3DC

smartboart, 2024-01-31:
<html><body>

e3dc-rscp.0 | 2024-01-31 09:01:16.877 | warn | Unknown tag: tagCode=0xe2ce6fcf, len=56306, typeCode=0xf8
-- | -- | -- | --
e3dc-rscp.0 | 2024-01-31 09:01:16.868 | warn | Unknown tag: tagCode=0xe2ce6fcf, len=56306, typeCode=0xf8
e3dc-rscp.0 | 2024-01-31 09:01:16.859 | warn | Unknown tag: tagCode=0xe2ce6fcf, len=56306, typeCode=0xf8
e3dc-rscp.0 | 2024-01-31 09:01:16.850 | warn | Unknown tag: tagCode=0xe2ce6fcf, len=56306, typeCode=0xf8
e3dc-rscp.0 | 2024-01-31 09:01:16.841 | warn | Unknown tag: tagCode=0xe2ce6fcf, len=56306, typeCode=0xf8
e3dc-rscp.0 | 2024-01-31 09:01:16.832 | warn | Unknown tag: tagCode=0xe2ce6fcf, len=56306, typeCode=0xf8
e3dc-rscp.0 | 2024-01-31 09:01:14.599 | warn | Unknown tag: tagCode=0x14af33d7, len=1235, typeCode=0x1a
e3dc-rscp.0 | 2024-01-31 09:01:14.590 | warn | Unknown tag: tagCode=0x14af33d7, len=1235, typeCode=0x1a
e3dc-rscp.0 | 2024-01-31 09:01:14.582 | warn | Unknown tag: tagCode=0x14af33d7, len=1235, typeCode=0x1a
e3dc-rscp.0 | 2024-01-31 09:01:14.573 | warn | Unknown tag: tagCode=0x14af33d7, len=1235, typeCode=0x1a
e3dc-rscp.0 | 2024-01-31 09:01:14.564 | warn | Unknown tag: tagCode=0x14af33d7, len=1235, typeCode=0x1a
e3dc-rscp.0 | 2024-01-31 09:01:14.554 | warn | Unknown tag: tagCode=0x14af33d7, len=1235, typeCode=0x1a
e3dc-rscp.0 | 2024-01-31 09:01:13.296 | warn | Unknown tag: tagCode=0xc0dec319, len=24883, typeCode=0x21
e3dc-rscp.0 | 2024-01-31 09:01:13.287 | warn | Unknown tag: tagCode=0xc0dec319, len=24883, typeCode=0x21
e3dc-rscp.0 | 2024-01-31 09:01:13.278 | warn | Unknown tag: tagCode=0xc0dec319, len=24883, typeCode=0x21
e3dc-rscp.0 | 2024-01-31 09:01:13.269 | warn | Unknown tag: tagCode=0xc0dec319, len=24883, typeCode=0x21
e3dc-rscp.0 | 2024-01-31 09:01:13.260 | warn | Unknown tag: tagCode=0xc0dec319, len=24883, typeCode=0x21
e3dc-rscp.0 | 2024-01-31 09:01:13.250 | warn | Unknown tag: tagCode=0xc0dec319, len=24883, typeCode=0x21
e3dc-rscp.0 | 2024-01-31 09:01:02.598 | warn | Unknown tag: tagCode=0xb5557827, len=59353, typeCode=0x44
e3dc-rscp.0 | 2024-01-31 09:01:02.589 | warn | Unknown tag: tagCode=0xb5557827, len=59353, typeCode=0x44
e3dc-rscp.0 | 2024-01-31 09:01:02.580 | warn | Unknown tag: tagCode=0xb5557827, len=59353, typeCode=0x44
e3dc-rscp.0 | 2024-01-31 09:01:02.571 | warn | Unknown tag: tagCode=0xb5557827, len=59353, typeCode=0x44
e3dc-rscp.0 | 2024-01-31 09:01:02.562 | warn | Unknown tag: tagCode=0xb5557827, len=59353, typeCode=0x44
e3dc-rscp.0 | 2024-01-31 09:01:02.552 | warn | Unknown tag: tagCode=0xb5557827, len=59353, typeCode=0x44
e3dc-rscp.0 | 2024-01-31 09:00:30.320 | warn | Unknown tag: tagCode=0x708c2a73, len=24648, typeCode=0xf
e3dc-rscp.0 | 2024-01-31 09:00:30.299 | warn | Unknown tag: tagCode=0x708c2a73, len=24648, typeCode=0xf
e3dc-rscp.0 | 2024-01-31 09:00:30.282 | warn | Unknown tag: tagCode=0x708c2a73, len=24648, typeCode=0xf
e3dc-rscp.0 | 2024-01-31 09:00:30.272 | warn | Unknown tag: tagCode=0x708c2a73, len=24648, typeCode=0xf
e3dc-rscp.0 | 2024-01-31 09:00:30.262 | warn | Unknown tag: tagCode=0x708c2a73, len=24648, typeCode=0xf
e3dc-rscp.0 | 2024-01-31 09:00:30.251 | warn | Unknown tag: tagCode=0x708c2a73, len=24648, typeCode=0xf
e3dc-rscp.0 | 2024-01-31 08:59:52.558 | warn | Unknown tag: tagCode=0x130d39d1, len=9194, typeCode=0xb
e3dc-rscp.0 | 2024-01-31 08:59:52.548 | warn | Unknown tag: tagCode=0x130d39d1, len=9194, typeCode=0xb
e3dc-rscp.0 | 2024-01-31 08:59:52.539 | warn | Unknown tag: tagCode=0x130d39d1, len=9194, typeCode=0xb
e3dc-rscp.0 | 2024-01-31 08:59:52.530 | warn | Unknown tag: tagCode=0x130d39d1, len=9194, typeCode=0xb
e3dc-rscp.0 | 2024-01-31 08:59:52.520 | warn | Unknown tag: tagCode=0x130d39d1, len=9194, typeCode=0xb
e3dc-rscp.0 | 2024-01-31 08:59:52.511 | warn | Unknown tag: tagCode=0x130d39d1, len=9194, typeCode=0xb
e3dc-rscp.0 | 2024-01-31 08:59:27.288 | warn | Unknown tag: tagCode=0x608b061c, len=46868, typeCode=0x15
e3dc-rscp.0 | 2024-01-31 08:59:27.286 | warn | Unknown tag: tagCode=0x608b061c, len=46868, typeCode=0x15
e3dc-rscp.0 | 2024-01-31 08:59:27.283 | warn | Unknown tag: tagCode=0x608b061c, len=46868, typeCode=0x15
e3dc-rscp.0 | 2024-01-31 08:59:27.281 | warn | Unknown tag: tagCode=0x608b061c, len=46868, typeCode=0x15
e3dc-rscp.0 | 2024-01-31 08:59:27.280 | warn | Unknown tag: tagCode=0x608b061c, len=46868, typeCode=0x15
e3dc-rscp.0 | 2024-01-31 08:59:27.277 | warn | Unknown tag: tagCode=0x608b061c, len=46868, typeCode=0x15
e3dc-rscp.0 | 2024-01-31 08:58:27.429 | warn | Unknown tag: tagCode=0x39cacb99, len=19668, typeCode=0xb0
e3dc-rscp.0 | 2024-01-31 08:58:27.419 | warn | Unknown tag: tagCode=0x39cacb99, len=19668, typeCode=0xb0
e3dc-rscp.0 | 2024-01-31 08:58:27.410 | warn | Unknown tag: tagCode=0x39cacb99, len=19668, typeCode=0xb0
e3dc-rscp.0 | 2024-01-31 08:58:27.401 | warn | Unknown tag: tagCode=0x39cacb99, len=19668, typeCode=0xb0
e3dc-rscp.0 | 2024-01-31 08:58:27.392 | warn | Unknown tag: tagCode=0x39cacb99, len=19668, typeCode=0xb0
e3dc-rscp.0 | 2024-01-31 08:58:27.381 | warn | Unknown tag: tagCode=0x39cacb99, len=19668, typeCode=0xb0
e3dc-rscp.0 | 2024-01-31 08:58:16.232 | warn | Unknown tag: tagCode=0x485dfa16, len=62783, typeCode=0xb7
e3dc-rscp.0 | 2024-01-31 08:58:16.224 | warn | Unknown tag: tagCode=0x485dfa16, len=62783, typeCode=0xb7
e3dc-rscp.0 | 2024-01-31 08:58:16.216 | warn | Unknown tag: tagCode=0x485dfa16, len=62783, typeCode=0xb7
e3dc-rscp.0 | 2024-01-31 08:58:16.208 | warn | Unknown tag: tagCode=0x485dfa16, len=62783, typeCode=0xb7
e3dc-rscp.0 | 2024-01-31 08:58:16.200 | warn | Unknown tag: tagCode=0x485dfa16, len=62783, typeCode=0xb7
e3dc-rscp.0 | 2024-01-31 08:58:16.192 | warn | Unknown tag: tagCode=0x485dfa16, len=62783, typeCode=0xb7
e3dc-rscp.0 | 2024-01-31 08:57:53.742 | warn | Unknown tag: tagCode=0x9fbabddd, len=63719, typeCode=0x3c
e3dc-rscp.0 | 2024-01-31 08:57:53.717 | warn | Unknown tag: tagCode=0x9fbabddd, len=63719, typeCode=0x3c
e3dc-rscp.0 | 2024-01-31 08:57:53.696 | warn | Unknown tag: tagCode=0x9fbabddd, len=63719, typeCode=0x3c
e3dc-rscp.0 | 2024-01-31 08:57:53.686 | warn | Unknown tag: tagCode=0x9fbabddd, len=63719, typeCode=0x3c
e3dc-rscp.0 | 2024-01-31 08:57:53.674 | warn | Unknown tag: tagCode=0x9fbabddd, len=63719, typeCode=0x3c
e3dc-rscp.0 | 2024-01-31 08:57:53.662 | warn | Unknown tag: tagCode=0x9fbabddd, len=63719, typeCode=0x3c
e3dc-rscp.0 | 2024-01-31 08:57:49.204 | warn | Unknown tag: tagCode=0xf8302109, len=7662, typeCode=0xc6
e3dc-rscp.0 | 2024-01-31 08:57:49.195 | warn | Unknown tag: tagCode=0xf8302109, len=7662, typeCode=0xc6
e3dc-rscp.0 | 2024-01-31 08:57:49.187 | warn | Unknown tag: tagCode=0xf8302109, len=7662, typeCode=0xc6
e3dc-rscp.0 | 2024-01-31 08:57:49.178 | warn | Unknown tag: tagCode=0xf8302109, len=7662, typeCode=0xc6
e3dc-rscp.0 | 2024-01-31 08:57:49.169 | warn | Unknown tag: tagCode=0xf8302109, len=7662, typeCode=0xc6
e3dc-rscp.0 | 2024-01-31 08:57:49.160 | warn | Unknown tag: tagCode=0xf8302109, len=7662, typeCode=0xc6
e3dc-rscp.0 | 2024-01-31 08:57:37.418 | warn | Unknown tag: tagCode=0x67b4f5d0, len=21545, typeCode=0xae
[... 111 lines cut ...]
e3dc-rscp.0 | 2024-01-31 08:52:58.039 | warn | Unknown tag: tagCode=0xb74f7c77, len=8563, typeCode=0xa1
e3dc-rscp.0 | 2024-01-31 08:52:58.028 | warn | Unknown tag: tagCode=0xb74f7c77, len=8563, typeCode=0xa1
e3dc-rscp.0 | 2024-01-31 08:52:45.311 | warn | Unknown tag: tagCode=0x506f6814, len=27330, typeCode=0xc5
e3dc-rscp.0 | 2024-01-31 08:52:45.303 | warn | Unknown tag: tagCode=0x506f6814, len=27330, typeCode=0xc5
e3dc-rscp.0 | 2024-01-31 08:52:45.275 | warn | Unknown tag: tagCode=0x506f6814, len=27330, typeCode=0xc5
e3dc-rscp.0 | 2024-01-31 08:52:45.266 | warn | Unknown tag: tagCode=0x506f6814, len=27330, typeCode=0xc5
e3dc-rscp.0 | 2024-01-31 08:52:45.258 | warn | Unknown tag: tagCode=0x506f6814, len=27330, typeCode=0xc5
e3dc-rscp.0 | 2024-01-31 08:52:45.250 | warn | Unknown tag: tagCode=0x506f6814, len=27330, typeCode=0xc5
e3dc-rscp.0 | 2024-01-31 08:52:27.233 | warn | Unknown tag: tagCode=0x21974acc, len=13539, typeCode=0x82
e3dc-rscp.0 | 2024-01-31 08:52:27.230 | warn | Unknown tag: tagCode=0x21974acc, len=13539, typeCode=0x82
e3dc-rscp.0 | 2024-01-31 08:52:27.226 | warn | Unknown tag: tagCode=0x21974acc, len=13539, typeCode=0x82
e3dc-rscp.0 | 2024-01-31 08:52:27.223 | warn | Unknown tag: tagCode=0x21974acc, len=13539, typeCode=0x82
e3dc-rscp.0 | 2024-01-31 08:52:27.219 | warn | Unknown tag: tagCode=0x21974acc, len=13539, typeCode=0x82
[... cut ...]
e3dc-rscp.0 | 2024-01-31 08:52:24.555 | warn | Unknown tag: tagCode=0xa147091, len=10462, typeCode=0xe1
e3dc-rscp.0 | 2024-01-31 08:52:24.547 | warn | Unknown tag: tagCode=0xa147091, len=10462, typeCode=0xe1
e3dc-rscp.0 | 2024-01-31 08:52:24.539 | warn | Unknown tag: tagCode=0xa147091, len=10462, typeCode=0xe1
e3dc-rscp.0 | 2024-01-31 08:52:19.073 | warn | Unknown tag: tagCode=0x996a0c30, len=9933, typeCode=0x45
e3dc-rscp.0 | 2024-01-31 08:52:19.064 | warn | Unknown tag: tagCode=0x996a0c30, len=9933, typeCode=0x45

</body>
</html>

git-kick, 2024-01-31:
Seems like the communication was completely out-of-sync - the details of the "Unknown tag" messages do not make any sense to me, the tagCodes/typeCodes are random and len is larger than usually expected. The only "pattern" I can see is that there are always six equal events in a sequence.  

I think restarting all components could help. Otherwise, if this is reproducible, I would need documentation about the changes by E3/DC (which is not availyble, I'm afraid).

Jockelflo, 2024-01-31:
Das sind meine Fehler seit heute morgen. Es gab im PV Modul eine neue Version, so wie ich das gesehen habe

e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_GET_FS_USAGE has no assigned polling interval - assuming 'M' - please check io-package.json
-- | -- | -- | --
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_GET_FACILITY has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_GUI_TARGET has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_SW_RELEASE has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_TIME_ZONE has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_UTC_TIME has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_TIME has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_DHCP_STATUS has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_DNS has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_GATEWAY has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_MAC_ADDRESS has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.192 | warn | TAG_INFO_REQ_SUBNET_MASK has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.191 | warn | TAG_INFO_REQ_IP_ADDRESS has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.191 | warn | TAG_INFO_REQ_A35_SERIAL_NUMBER has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.191 | warn | TAG_INFO_REQ_MODULES_SW_VERSIONS has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.191 | warn | TAG_INFO_REQ_PRODUCTION_DATE has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.191 | warn | TAG_INFO_REQ_SERIAL_NUMBER has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.190 | warn | TAG_EP_REQ_EP_RESERVE has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.189 | warn | TAG_EMS_REQ_EMERGENCY_POWER_TEST_STATUS has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.189 | warn | TAG_EMS_REQ_WB_ENFORCE_POWER_ASSIGNMENT has no assigned polling interval - assuming 'M' - please check io-package.json
e3dc-rscp.0 | 2024-01-31 22:05:17.189 | warn | TAG_EMS_REQ_WB_DISCHARGE_BAT_UNTIL has no assigned polling interval - assuming 'M' - please check io-package.json

git-kick, 2024-02-01:
@Jockelflo, dein Log sieht nach einem ganz anderen Thema aus, siehe Issues #156 , #191

smartboart, 2024-02-01:
> Seems like the communication was completely out-of-sync - the details of the "Unknown tag" messages do not make any sense to me, the tagCodes/typeCodes are random and len is larger than usually [...]

Hi, thanks for feedback... I have restarted my completed iobroker system.. This solved the issue...crazy.. because i had no other issues withe broker... But it helped...

git-kick, 2024-02-01:
@smartboart, many thanks for your input. I will keep this in mind and if such trouble is reported again, then I have to think about some logic like "if the adapter throws more than x unknown-tag warnings per minute, then reset communication".

closed 2024-02-01 by git-kick

mibr85, 2024-03-08:
i have the same issue every 5 min this massage come up:
`
e3dc-rscp.0 | 2024-03-08 06:15:17.042 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 06:10:17.040 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 06:05:17.039 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 06:00:17.039 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 05:55:17.037 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 05:50:17.035 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 05:45:17.038 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
`

after restart of IOB container, problem is solved